MarlinFirmware/Marlin #4179: Changing 2nd extruder output to 2nd Z-axis stepper

5stringmonster, 2016-06-30:
How would I go about changing the 2nd extruder into the 2nd z axis control? Using MKS Gen 7 v1.4. The output farthest away from the power input at the top. Running a sunhokey prusa i3, but I've upgraded to the 1.4 and was dismayed to find only one z motor output. Don't want to run one driver circuit on two motors. Thank you.

thinkyhead, 2016-07-01:
With the latest Marlin (`RCBugFix`) you only have to enable `Z_DUAL_STEPPER_DRIVERS`, and the first unused E plug will be assigned to your second Z stepper automagically.

5stringmonster, 2016-07-01:
Can we add this to the wiki? Seems like a good configuration detail to include and expand on. I know how to code for the most part and I was stumped finding this option anywhere in the source.

thinkyhead, 2016-07-01:
>  Can we add this to the wiki?

We don't use the wiki any more. And it's a very new detail. It will end up on marlinfw.org as we find more time to work on documentation.

closed 2016-07-16 by jbrazio