MarlinFirmware/Marlin #4182: Tool offset not working anymore? (RCBugFix)

Alex9779, 2016-07-01:
I had a discussion recently about the tool offset and switching tool and got a feature introduced so a tool switch can be done with out moving but as for today's RCBugFix I can't get the machine to recognise the tool offset.
I set the offset with `M218 T1 X28 Y-18`and then a switch from T0 to T1 should move the head but nothing happens. Moves after T1 is selected still position the first tool to the given coordinates...
Another configuration error of me?

Alex9779, 2016-07-01:
Ok this is strange, sometimes the head moves when switching to T1 but only in Y direction.
I home with G28 then I set the offset with M218, head is at 0,0 with first tool, I switch with T1 and head moves in Y direction. Then if I switch back to T0 nothing happens only the Z axis seems to be adjusted because of the leveling.
After that I don't get the head to move on a tool switch though I know that somehow it is possible because I saw it during a test print where I have several switches of tools in the code but also only in Y direction.

Maybe my "strange" setup has an influence on this?
My limits are X-28 to X300 and Y0 to Y240, the second tool has an offset of X28 Y-18 when viewed from the to so it is in front and left of the first tool...

thinkyhead, 2016-07-01:
If you're using auto bed leveling it will help if you enable `DEBUG_LEVELING_FEATURE` and use `M115 S255` to enable all logging. Then we can see in `gcode_T` whether the matrix compensation is being applied incorrectly.

The other thing to check is, where on the bed are you doing the tool change? If you are near the edges, it's probable that the movement during the tool-change is being constrained by the software endstops.

As I look at the code for `gcode_T` it's pretty simple (apart from the bed leveling compensation).
```cpp
if (tmp_extruder != active_extruder) {
  float xydiff[2] = {
    hotend_offset[X_AXIS][tmp_extruder] - hotend_offset[X_AXIS][active_extruder],
    hotend_offset[Y_AXIS][tmp_extruder] - hotend_offset[Y_AXIS][active_extruder]
  };
  for (uint8_t i = X_AXIS; i <= Y_AXIS; i++) {
    current_position[i] += xydiff[i];
    position_shift[i] += xydiff[i];
    update_software_endstops((AxisEnum)i);
  }
  active_extruder = tmp_extruder;
  SYNC_PLAN_POSITION_KINEMATIC();
  prepare_move_to_destination();
}
```

Alex9779, 2016-07-02:
I updated to lasted RCBugFix again and tried what you said.
Here is the log:
```
Send: G28
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: X:0.00 Y:0.00 Z:0.00 E:0.00 Count X: 0 Y:0 Z:0
Recv: ok
[...]
Send: M218 T1 X30 Y30
Recv: echo:Hotend offsets: 0.00,0.00 30.00,30.00
Recv: ok
[...]
Send: G1 X150 Y100 F3000
Recv: ok
[...]
Send: T1
Recv: echo:Active Extruder: 1
Recv: ok
[...]
Send: T0
Recv: echo:Active Extruder: 0
Recv: ok
```

Switching extruders does not move the head as it did before...

Alex9779, 2016-07-02:
And the offset is not recognized... I switch to T1, numbers on the LCD stay the same, I do a move command to the same spot as before but nothing happens, the firmware thinks it is there.
If I do this with RC6 I switch to T1, numbers on the LCD remain, moving to the same spot as before does move the head and numbers remain...

Alex9779, 2016-07-02:
Here is my config if this helps:
<details> 
  <summary>Configuration.h </summary>
   ```
/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#ifndef CONFIGURATION_H
#define CONFIGURATION_H

/**
 *
 *  ***********************************
 *  **  ATTENTION TO ALL DEVELOPERS  **
 *  ***********************************
 *
 * You must increment this version number for every significant change such as,
 * but not limited to: ADD, DELETE RENAME OR REPURPOSE any directive/option.
 *
 * Note: Update also Version.h !
 */
#define CONFIGURATION_H_VERSION 010100

#include "boards.h"
#include "macros.h"

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
[... 1872 lines cut ...]
 * M155 B114 ; r
 * M155 B108 ; l
 * M155 B105 ; i
 * M155 B110 ; n
 * M155 S1   ; Send the current buffer
 *
 * ; Example #2
 * ; Request 6 bytes from slave device with address 0x63 (99)
 * M156 A99 B5
 *
 * ; Example #3
 * ; Example serial output of a M156 request
 * echo:i2c-reply: from:99 bytes:5 data:hello
 */

// @section i2cbus

//#define EXPERIMENTAL_I2CBUS

#include "Conditionals.h"
#include "SanityCheck.h"

#endif //CONFIGURATION_ADV_H
```
</details>

thinkyhead, 2016-07-02:
Your configuration seems fine to me.

But oops! Can you do your logging of `T0` / `T1` again? This time use `M111 S255` to turn on debugging. (I had it wrong last time. It's `M111`, not `M115`.) This will show details of the leveling correction. It should show a change in the current XY corresponding to your hotend offset values.

The last thing the tool-change is supposed to do is move back to the old position from the updated position, so the new nozzle is in the same spot as the old one.

Alex9779, 2016-07-03:
Ok did it, here is the result:
```
Send: G28
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: X:0.00 Y:0.00 Z:2.00 E:0.00 Count X: 0 Y:0 Z:6400
Recv: ok
[...]
Send: M111 S255
Recv: echo:DEBUG:ECHO,INFO,ERRORS,DRYRUN,COMMUNICATION
Recv: ok
[...]
Recv: echo:M105
[...]
Recv: echo:M105
[...]
Send: M218 T1 X30 Y30
Recv: echo:M218 T1 X30 Y30
Recv: echo:Hotend offsets: 0.00,0.00 30.00,30.00
Recv: ok
[...]
Recv: echo:M105
[...]
Recv: echo:M105
[...]
Send: G1 X150 Y100 F3000
Recv: echo:G1 X150 Y100 F3000
Recv: ok
[...]
Recv: echo:M105
[...]
Send: T1
Recv: echo:T1
Recv: echo:Active Extruder: 1
Recv: ok
[...]
Recv: echo:M105
[...]
Send: T0
Recv: echo:T0
Recv: echo:Active Extruder: 0
Recv: ok
[...]
Recv: echo:M105
[...]
Send: T1
Recv: echo:T1
Recv: echo:Active Extruder: 1
Recv: ok
Send: T0
Recv: echo:T0
Recv: echo:Active Extruder: 0
Recv: ok
```

I intentionally leveled the bed wrong so I see seem change better and I can confirm the bed moves when I switch tools but no offset is applied...

Alex9779, 2016-07-03:
This worries me a bit because that PR got merged 17 days ago... Am I the only one in the community actually testing Marlin on a real machine with multiple extruders and MBL?

Alex9779, 2016-07-03:
I think the problematic lines are these https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/Marlin_main.cpp#L6624-L6628

Alex9779, 2016-07-03:
I think these are the problematic lines:
https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/Marlin_main.cpp#L6624-L6628

```
if (mbl.active()) {
            float xpos = current_position[X_AXIS] - home_offset[X_AXIS],
                  ypos = current_position[Y_AXIS] - home_offset[Y_AXIS];
            current_position[Z_AXIS] += mbl.get_z(xpos + xydiff[X_AXIS], ypos + xydiff[Y_AXIS]) - mbl.get_z(xpos, ypos);
}
```

What is `home_offset`? 
I changed it to this:
```
if (mbl.active()) {
            float xpos = current_position[X_AXIS] += xydiff[X_AXIS],
                  ypos = current_position[Y_AXIS] +=xydiff[Y_AXIS];
            current_position[Z_AXIS] += mbl.get_z(xpos - xydiff[X_AXIS], ypos - xydiff[Y_AXIS]) - mbl.get_z(xpos, ypos);
          }
```

and this made it working.

But I have a question, why do you change the Z according to MBL? If you move the new extruder to the same spot then this does nothing, would only make sense if you don't move the nozzle with `Tx S1` but this does disable all moves even Z so the displays shows the new coordinates including the change Z value but the bed does not move...

thinkyhead, 2016-07-03:
Aha. I thought you were using Automatic Bed Leveling. That would have given us more logging. But we don't have any logging added to the Mesh Bed Leveling part of `gcode_T`.

thinkyhead, 2016-07-03:
I'm changing it to this…
```cpp
#else // !AUTO_BED_LEVELING_FEATURE

  #if ENABLED(MESH_BED_LEVELING)

    if (mbl.active()) {
      float xpos = current_position[X_AXIS] - home_offset[X_AXIS] - position_shift[X_AXIS],
            ypos = current_position[Y_AXIS] - home_offset[Y_AXIS] - position_shift[Y_AXIS];
      current_position[Z_AXIS] += mbl.get_z(xpos + xydiff[X_AXIS], ypos + xydiff[Y_AXIS]) - mbl.get_z(xpos, ypos);
    }

  #endif // MESH_BED_LEVELING

  // The newly-selected extruder XY is actually at...
  current_position[X_AXIS] += xydiff[X_AXIS];
  current_position[Y_AXIS] += xydiff[Y_AXIS];

#endif // !AUTO_BED_LEVELING_FEATURE
```
The `home_offset` indicates a shift to the coordinate system applied by `M206`. I've also added `position_shift` to the calculation. This variable indicates that the position was shifted at some point by `G92`. We have to subtract these offsets to the coordinate system in order to get back to the coordinates that originate at the unmodified `MIN_X_POS`, `MIN_Y_POS`, and `MIN_Z_POS`.

thinkyhead, 2016-07-03:
> why do you change the Z according to MBL?

Because the active nozzle is now at a different XY position (before we execute the move back to the old position). So we need to adjust Z to account for mesh variation. MBL will continue to correct Z (most likely returning it to its previous value) once we do the move from the new position back to the old position.

Alex9779, 2016-07-03:
Ok looks good there are the things I didn't understand ;)...

Still remains the question about the Z adjustment... Any thoughts on this?
IMHO the non-move-logic has to be reworked because not moving is for X and Y but not for Z... At the moment this looks like another issue with the current implementation as I described above because the machine update the coordinates but does not adjust reality. Didn't test this further, I did no additional moves and see if the Z is corrected then, so I have no idea if this is a real issue

Alex9779, 2016-07-03:
Ok strike some points of my last comment because I wrote and you answered at the same time...
But what about if not moving? Will the Z be correct on the next move?

Alex9779, 2016-07-03:
Ok I thought about this and I don't get why you calculate the positions you use to get the MBL spot differently than the change for the current positions...
Sorry if bothering but I really try to understand what's going on...

thinkyhead, 2016-07-04:
> I don't get why you calculate the positions you use to get the MBL spot differently

First, we need to get the "raw" current position that MBL can relate to. So we subtract all coordinate offsets that have been applied:
```cpp
float xpos = current_position[X_AXIS] - home_offset[X_AXIS] - position_shift[X_AXIS],
      ypos = current_position[Y_AXIS] - home_offset[Y_AXIS] - position_shift[Y_AXIS];
```
Now, we get Z at the new position…
```cpp
mbl.get_z(xpos + xydiff[X_AXIS], ypos + xydiff[Y_AXIS])
```
…and the Z at the old position…
```cpp
mbl.get_z(xpos, ypos)
```
…and we subtract the Z at the new position from Z at the old position. (Let's say the Z at the new XY position is 0.2mm higher, for our example.)
```cpp
mbl.get_z(xpos + xydiff[X_AXIS], ypos + xydiff[Y_AXIS]) - mbl.get_z(xpos, ypos);
```
We add this difference to the current Z.
```cpp
current_position[Z_AXIS] += mbl.get_z(xpos + xydiff[X_AXIS], ypos + xydiff[Y_AXIS]) - mbl.get_z(xpos, ypos);
```
Now we have Z set correctly (e.g., `current_position[Z_AXIS] + 0.2`) for its new position.

We had forgotten to update `current_position` when MBL was enabled. Now we do:
```cpp
current_position[X_AXIS] += xydiff[X_AXIS];
current_position[Y_AXIS] += xydiff[Y_AXIS];
```

Alex9779, 2016-07-04:
Sorry this explains what but not why... Why do we need to calculate that stuff to get an MBL correction value for the same spot on the bed? I still think this will never result in a move because the coordinates for the old extruder change, the coordinates for the new change but if you look at this from the bed the new nozzle is where the old was and we have only one mesh for all nozzles so the Z must be the same.

In contrast if you do not move with Tx S1. Then the new nozzle stays where is was which is a different position on the bed and thus a different correction value...

thinkyhead, 2016-07-04:
The position always changes, at first. But it **_doesn't always move back_** to the old position.

Alex9779, 2016-07-04:
Ok I think I got it, you calculate the Z difference for the new nozzle which is still in the different position.
Then you set that to the Z axis. Some lines later you set the offsets to the X and Y axes too and again later you execute the moves. Because all this happens in literally no time you would not recognize it...

But why do you use home_offset and position_shift to get that positions for which you want the correction value and not just the xydiff values as when you set the new values to X and Y axes?

Alex9779, 2016-07-04:
Oh well you are a bit too fast for me, I had a look at the PR and you changed other things too.
That stuff has something to do with the MBL stuff...
So looks good! Thanks for the quick fix and the explanations....

thinkyhead, 2016-07-04:
> why do you use home_offset and position_shift to get that positions

Because MBL only works internally with unaltered positions. If your bed is 200x200, MBL recognizes positions from 0-200. If the coordinate system was shifted by 10 in the X and you didn't un-shift it before querying MBL, you would get the wrong Z, and MBL wouldn't know at all what to do with positions like -10  or 210, which are outside of its mesh space.

Alex9779, 2016-07-04:
Ya got it now.... Thanks for the explanations again...

Alex9779, 2016-07-06:
Tested it, and seems to work now though an other issue appeared, I'll open a new one...

closed 2016-07-06 by Alex9779