MarlinFirmware/Marlin #4190: M503 does not report Bed PID Parameters (M304)

token47, 2016-07-03:
I'm about to upgrade firmware version from 1.1.0-rc6 to 1.1.0-rcbugfix of today (07/02) and while skimming through the changes I noticed that the EEPROM version number was bumped from 23 to 24.

Anyway, I went to backup current EEPROM data just in case, running a M503 which should give you commands to put all data back after upgrading. Then I noticed the heatend PID paremeters are there (M301) but the bed PID parameters are not (M304).

I was pretty sure that I was using PID on the bed because I auto-tuned it some time ago, and just checked the sources and the values are there. Is the support for bed PID not on the EEPROM yet? Should I change them on the source of the new firmware?

#define  DEFAULT_Kp 11.14
#define  DEFAULT_Ki 0.48
#define  DEFAULT_Kd 64.62

#define  DEFAULT_bedKp 356.02
#define  DEFAULT_bedKi 35.51
#define  DEFAULT_bedKd 892.46

The output of M503 shows only:

PID settings:
M301 P11.14 I0.48 D64.62 C100.00 L20

BTW: This C and L parameters are not documented on the reprap gcodes wiki page. I know it's not a bug in marlin but it would be interesting to update that also. I could do that if someone confirms this is in order and point me to some other documentation about it.

thinkyhead, 2016-07-03:
Are you sure you enabled `PIDTEMPBED` before flashing?
```cpp
#if ENABLED(PIDTEMPBED)
  CONFIG_ECHO_START;
  SERIAL_ECHOPAIR("  M304 P", thermalManager.bedKp);
  SERIAL_ECHOPAIR(" I", unscalePID_i(thermalManager.bedKi));
  SERIAL_ECHOPAIR(" D", unscalePID_d(thermalManager.bedKd));
  SERIAL_EOL;
#endif
```

thinkyhead, 2016-07-03:
> This C and L parameters are not documented on the reprap gcodes wiki page

Thanks for pointing that out! We're trying to get together all our GCode documentation, and definitely want to keep the RepRap wiki updated also.

token47, 2016-07-03:
Yes, you are right. In the last upgrade I failed to transport that config (forgot to uncoment the PIDTEMPBED. I put the parameters but not enabled them. Shame on me :-)

This can be closed. Thank you!

closed 2016-07-03 by thinkyhead