MarlinFirmware/Marlin #4191: Add menu entry in 1st level to abort and raise hotend quickly

token47, 2016-07-03:
I allway find myself thinking twice before printing without a computer (i.e. from SD card or from octoprint). The reason for that is that if anything happens it is very slow to react using the LCD and menu options.

The most common need, by far, is to stop the print and quickly raise the hotend (or just raise it if already stopped). This is very quick to do in the jog controls of the slicer but painfull to do on the LCD. If you are printing the 1st layer, the time you take to navigate the menus to move Z up will make the hotend burn my buildtak or even hurt the glass in some cases.

I think it's interesting to have options for this kind of emergencial actions in the menu 1st level, and I guess that would be of general interest too.

thinkyhead, 2016-07-03:
Thanks for the feedback! Recently we've been trying to patch up the "Stop print" code so it stops more quickly, but at the moment it's leading to a bug where the last few commands from the SD file keep looping. (Hopefully we figure this out soon!)

Meanwhile, well at least "Stop print" isn't _too far_ away. You just click once, spin the wheel quickly to the bottom, and you are now at "Stop print."

But I can see why you'd want some more comprehensive options as soon as the print is stopped, going directly to another menu with some helpful commands. I think that would be a good workflow, and we should continue to facilitate stuff like this.

In the event of "Stop print", in general it's not a bad idea to do a retract, sideways move, and Z lift (in that order, if possible), to ensure no ooze is left on the print. We also have at least the conceptual design of a resume option in mind, so you can interrupt a print at any point and pick it up later (presuming the print-head is left in place – since we cannot re-home Z without colliding). We need to make these behaviors configurable, so vendors can tailor them to their specifications.

token47, 2016-07-03:
Ok, so there is a greater goal in progress. That's very good. I guess what you described, together with babysteps make a lot of recovery possible from the LCD / SD prints.

I always thought the first level of the menus were a bit underused. The options on the extremes are very quick to reach, and the ones next (second and one before last) are also very easy as you do a quick turn and back just one. These are good candidates for the most used options that are normally buried under the menus (of at least leave the options where they are now and put shortcuts to them on the outside -- maybe even configurable on compilation time).

If I could list the options that I would bring to front (shortcuts), they would be:

- Prepare -> Homing
- Prepare -> Move Axis -> Move 1mm -> Move Z
- Preheat ABS -> Bed
- Control -> Motion -> Z offset
- Control -> Save to EEPROM

As you can see, not a lot of options, and the ones I use 99% of the time. The Move Z up by 1mm is the winner. Always a pain to enter 4 submenus to do that.

ntoff, 2016-07-03:
>In the event of "Stop print", in general it's not a bad idea to do a retract, sideways move, and Z lift (in that order, if possible), to ensure no ooze is left on the print.

HELL. NO. If the nozzle is still at bed height and touching the bed. Then you've got it digging a great chunk out of any kind of tape or PEI sheet and possibly colliding with the clips holding any glass sheet on or worse. If it's homed below the build surface like mine is, you could have it trying to break the end of the nozzle off.

IMHO *Stop print* should just stop whatever is currently going on and sit there and wait for further instructions, treating it like a non emergency situation. Then there should be *emergency stop*  which stops dead, no movement, all heaters off and refuses to accept incoming commands until the printer is reset.

thinkyhead, 2016-07-03:
> IMHO Stop print should just stop whatever is currently going on and sit there

Sorry, I meant to type "Pause print" above. Anyway, for either Pause or Stop, IMHO it should be left up to individual users and vendors to decide if they want to augment how their machines respond. So I suggest that we make it configurable, but keep the default behavior simple.

> If the nozzle is still at bed height and touching the bed…

…then the machine could account for this and respond differently than if the nozzle is higher up. Of course, in the "Pause print" sideways move it would not just move blindly towards the edge, but towards the center in the X and/or Y axes. I've experimented with this a bit, and a quick sideways move combined with retraction just before lifting the nozzle pretty much eliminates all blobs.

ntoff, 2016-07-04:
And if your Z height is misconfigured then any kind of movement would destroy the PEI sheet or in the case of my printer, potentially break the nozzle off. Just because it works for you doesn't mean it'll work for everyone else. If I hit stop or pause I wouldn't expect the machine to just go off and do its own thing, what the hell kind of thinking is that? It's homing-gate all over again and I'm thoroughly glad I've switched to repetier firmware.

thinkyhead, 2016-07-04:
> And if your Z height is misconfigured then any kind of movement would destroy the PEI sheet

When, where, how?

>  It's homing-gate all over again

I'm just a geek talking to another geek. Exchanging ideas. Don't get so uptight.

token47, 2016-07-04:
Hehe.

I see another discussion derived from my suggestion, which is great. But my initial suggestion still stands: I often have to stop and raise quickly (not to burn my buildtak) and I would love to have and option at hand to do just that. I guess other people would have a use to that too.

It does not need to be the only "stop" action available, nor is it meant to pause and continue later, it would just be abort and raise, clearly named as such so you only use it if you have a case for it. It could be configurable, of course.

thinkyhead, 2016-07-04:
Sure enough, @token47. With new feature requests, my tendency is to consider them in the context of existing features, if possible. We want to keep the number of configurable options from exploding too much. But also, I'm open to re-arranging the UI if it improves the user experience or clarifies the workflow.

So it sounds like, at minimum, it would help out if the "Pause Print" and "Stop Print" commands were closer to the top of the Main Menu. And, once you pause or stop a print, it would help if you immediately arrived at a menu with "Park Nozzle" (actual verbiage to be decided) at the top, "Resume Print" and "Abort Print" at the bottom, and maybe some other options.

token47, 2016-07-05:
That seems sufficient. My only issue is that "raise Z" is so buried down 4 levels of menus that at the time I get there it may be too late. The "stop" itself is not so much of a problem because most of the time I can navigate to find it wherever it is, the important timing is between stopping and rising (although in a few cases, stopping fast is also desirable).

Your proposal seems to be better as it covers more use cases.

jbrazio, 2016-07-23:
Marking as duplicate of #3407

closed 2016-07-23 by jbrazio

token47, 2016-07-24:
jbrazio, I'm not sure the two issues (#3407 and #4191) are duplicate. One talks about lifting the nose on a thermal runaway firmware freeze, the other (mine) talks about putting an entry on the menu to stop the print and raise the nose because navigating through the menus to lift the nose is very complicated (you have to do a dozen clicks) and in the mean time you and up with a burn in the plate.

I'll leave coments on the other issue to emphasize this.

jbrazio, 2016-07-24:
Both require the same implementation concept, one requires an additional menu entry.