MarlinFirmware/Marlin #4193: LCD display black dots and simbol

gfunas, 2016-07-03:
So i tried to update new firmware  to my geeetech prusa i3c.
Got strange sound coming from LCD buzzer, one high frequency and LCD showing this
`#define LANGUAGE_INCLUDE GENERATE_LANGUAGE_INCLUDE(en)
#define DISPLAY_CHARSET_HD44780_JAPAN        // this is the most common hardware
//#define DISPLAY_CHARSET_HD44780_WESTERN
//#define DISPLAY_CHARSET_HD44780_CYRILLIC`
Tried all tree charset. 

> https://s31.postimg.org/9xfuh9b2j/IMG_20160703_111905.jpg

thinkyhead, 2016-07-03:
Please try the latest `RCBugFix` and see if it does any better. These settings have changed in the new version, so you will need to update them, and possibly some others as well…
```cpp
#define LANGUAGE_INCLUDE GENERATE_LANGUAGE_INCLUDE(en)
#define DISPLAY_CHARSET_HD44780_JAPAN        // this is the most common hardware
```
… are now …
```
#define LCD_LANGUAGE en
#define DISPLAY_CHARSET_HD44780 JAPANESE
```

thinkyhead, 2016-07-21:
Closed for lack of response.

closed 2016-07-21 by thinkyhead