MarlinFirmware/Marlin #4194: Missing steps causing print to lose coordinates

alitai, 2016-07-03:
Thank you for the great work - I apologize if this was covered before (or is a configuration issue).

Setup:  I am trying RC6 on a Mini Kossel with a Megatronics 2.0. I copied a working configuration settings from an older firmware into a new configuration.h and advanced configuration files. I also have a few pins disabled in the pins.h file.

Problem: Whatever settings I use the print proceeds to a consistent point - at which it seems the physical coordinates for x0 y0 & z0 change. It would continue printing at an offset of a few cm, and at a slant. Slowing the print by reducing the max feed rates changes the point where it happens - but it still happens.

I don't know if this is a bug or a wrong configuration. Reverting to an older firmware - the prints work fine. 

Thanks,

Assaf

thinkyhead, 2016-07-03:
Please test with [the latest `RCBugFix` code](https://github.com/MarlinFirmware/Marlin/tree/RCBugFix) to see if we already addressed your issue. If you still see problems, then we will take a look at your configuration and make sure everything there is copacetic.

alitai, 2016-07-04:
So I tested printing on RCBugFix version - and it worked well. Closed - Thank you.

closed 2016-07-04 by alitai