MarlinFirmware/Marlin #4197: Documentation for M302 outdated in reprap wiki

token47, 2016-07-03:
The M302 "Allow cold extrusion" documentation on the reprap gcodes wiki page is wrong for Marlin. It states that P0 and P1 are used to turn on/off when in fact the only option available is S and it sets the minimum temperature (with 0 being the default).

I could try to edit this small issues myself but can't seem to find a way to create a user there.

thinkyhead, 2016-07-04:
Hmm… Perhaps we should support either `S` or `P` going forward.

thinkyhead, 2016-07-04:
I've modified the wiki page to indicate that RepRapFirmware is now an outlier in using "P" for this parameter.

closed 2016-07-04 by thinkyhead

token47, 2016-07-04:
Sorry to be picky but it seems the update on the M302 command on reprap wiki is not accurate. Marlin uses S<temp> where you can set the temperature, not a boolean flag 0 or 1. In fact, if you use 0 (which is the default) you set the temp so low that the effect is the same as "enabling" the feature. But setting it to 1 will not "disable", you would have to set it to, say, "170".

Maybe you are right on the sugestion of supporting both S and P, each one with it's own semantics. Then you could use "P" option to just get back to the temperature the firmware had as minimum, without having to know which one it is.

thinkyhead, 2016-07-04:
Ah, I see. I wasn't careful in my edit there! I will fix it up and see about implementing the alternative semantic `P` option.