MarlinFirmware/Marlin #4198: M190  and M109 may lead to zombie state

miblooming, 2016-07-03:
After I send "M190 S60" to Marlin from Printrun, I cannot cancel the order util the temperature of hotbed   reach 60.If I don't want to wait, I have to press the RESET button on RAMPS. 
If someone want to contral printer with APP in mobile phone, after "M190" or "M109" sended, he/shi cannot  pause the printing task at once.Sometimes It's troublesome。
So I suggest: Marlin can deal new  "M190" commond when dealing "M190". The same as "M109"

From looming

jbrazio, 2016-07-03:
`M190: Wait for bed temperature to reach target temp`
If you don't want to wait.. `M140: Set Bed Temperature`

thinkyhead, 2016-07-03:
Commands that go into a loop, such as `M190`, `M109`, `G29`, etc., have the effect of blocking the command queue, so no further commands can be received and processed until the command is completed.

We have a request (#3611) to add a new (un-blockable) GCode to break out of some of these blocking loops, but they require adding a more low-level check for the given command. And there is still no guarantee that a command can be received and processed while the queue is blocked, because we have to respect a full serial character buffer also. See that thread for the issues raised.

For now, until we can look at some redesign of the foundations of our command processing (preserving compatibility with host software), this is how it will have to work. I realize that `M140` and `M104` are not a very viable solution when printing from SD card. Sorry we don't have any better options at this time.

maukcc, 2016-07-05:
A patch for now is to send it a temp command that is lower then the actual temp from LCD( I dont know if this works from host), faster is the reset switch :)

thinkyhead, 2016-07-05:
@maukcc That works because commands sent from the LCD bypass the command queue.

thinkyhead, 2016-07-07:
With #4226 and #3611 merged we now have a method to break out of these loops.

closed 2016-07-07 by thinkyhead