MarlinFirmware/Marlin #4206: ABL on Delta starts at x0 y0 (instead of at edge of bed) and slants

alitai, 2016-07-04:
Hi - Still working on getting the BugFix version to work. Downloaded a new RCBugFix branch today and  am getting an issue with the ABL functionality. The probing starts in the center (X0 Y0) and moves to the rear right quadrant, and does so at a slant.

I am using the Zmin pin and the nozzle as a probe (with a Trinket Based FSR detector), so offsets are all '0'.

Thanks,

Assaf

// @section bedlevel

#define AUTO_BED_LEVELING_FEATURE // Delete the comment to enable (remove // at the start of the line)
//#define DEBUG_LEVELING_FEATURE

#if ENABLED(AUTO_BED_LEVELING_FEATURE)

  // There are 2 different ways to specify probing locations:
  //
  // - "grid" mode
  //   Probe several points in a rectangular grid.
  //   You specify the rectangle and the density of sample points.
  //   This mode is preferred because there are more measurements.
  //
  // - "3-point" mode
  //   Probe 3 arbitrary points on the bed (that aren't collinear)
  //   You specify the XY coordinates of all 3 points.

  // Enable this to sample the bed in a grid (least squares solution).
  // Note: this feature generates 10KB extra code size.
  #define AUTO_BED_LEVELING_GRID  // Deltas only support grid mode.

  #if ENABLED(AUTO_BED_LEVELING_GRID)

    // Set the rectangle in which to probe
    #define DELTA_PROBEABLE_RADIUS (DELTA_PRINTABLE_RADIUS - 10)
    #define LEFT_PROBE_BED_POSITION -(DELTA_PROBEABLE_RADIUS)
    #define RIGHT_PROBE_BED_POSITION DELTA_PROBEABLE_RADIUS
    #define FRONT_PROBE_BED_POSITION -(DELTA_PROBEABLE_RADIUS)
    #define BACK_PROBE_BED_POSITION DELTA_PROBEABLE_RADIUS

alitai, 2016-07-04:
// Enable Z_MIN_PROBE_ENDSTOP to use _both_ a Z Probe and a Z-min-endstop on the same machine.
// With this option the Z_MIN_PROBE_PIN will only be used for probing, never for homing.
//
// *** PLEASE READ ALL INSTRUCTIONS BELOW FOR SAFETY! ***
//
// To continue using the Z-min-endstop for homing, be sure to disable Z_SAFE_HOMING.
// Example: To park the head outside the bed area when homing with G28.
//
// To use a separate Z probe, your board must define a Z_MIN_PROBE_PIN.
//
// For a servo-based Z probe, you must set up servo support below, including
// NUM_SERVOS, Z_ENDSTOP_SERVO_NR and Z_SERVO_ANGLES.
//
// - RAMPS 1.3/1.4 boards may be able to use the 5V, GND, and Aux4->D32 pin.
// - Use 5V for powered (usu. inductive) sensors.
// - Otherwise connect:
//   - normally-closed switches to GND and D32.
//   - normally-open switches to 5V and D32.
//
// Normally-closed switches are advised and are the default.
//
// The Z_MIN_PROBE_PIN sets the Arduino pin to use. (See your board's pins file.)
// Since the RAMPS Aux4->D32 pin maps directly to the Arduino D32 pin, D32 is the
// default pin for all RAMPS-based boards. Some other boards map differently.
// To set or change the pin for your board, edit the appropriate pins_XXXXX.h file.
//
// WARNING:
// Setting the wrong pin may have unexpected and potentially disastrous consequences.
// Use with caution and do your homework.
//
//#define Z_MIN_PROBE_ENDSTOP

// Enable Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN to use the Z_MIN_PIN for your Z_MIN_PROBE.
// The Z_MIN_PIN will then be used for both Z-homing and probing.
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

// To use a probe you must enable one of the two options above!

// This option disables the use of the Z_MIN_PROBE_PIN
// To enable the Z probe pin but disable its use, uncomment the line below. This only affects a
// Z probe switch if you have a separate Z min endstop also and have activated Z_MIN_PROBE_ENDSTOP above.
// If you're using the Z MIN endstop connector for your Z probe, this has no effect.
//#define DISABLE_Z_MIN_PROBE_ENDSTOP

// Enable Z Probe Repeatability test to see how accurate your probe is
//#define Z_MIN_PROBE_REPEATABILITY_TEST

//
// Probe Raise options provide clearance for the probe to deploy, stow, and travel.
//
#define Z_RAISE_PROBE_DEPLOY_STOW 50 // Raise to make room for the probe to deploy / stow
#define Z_RAISE_BETWEEN_PROBINGS 5  // Raise between probing points.

//
// For M851 give a range for adjusting the Z probe offset
//
#define Z_PROBE_OFFSET_RANGE_MIN -20
#define Z_PROBE_OFFSET_RANGE_MAX 20

thinkyhead, 2016-07-04:
Hi @alitai You should post your entire `Configuration.h` so we can see your movement limits, endstop configuration, etc.

alitai, 2016-07-05:
/**
 * Marlin RCBugfix
 * 
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#ifndef CONFIGURATION_H
#define CONFIGURATION_H

/**
 *
 *  ***********************************
 *  **  ATTENTION TO ALL DEVELOPERS  **
 *  ***********************************
 *
 * You must increment this version number for every significant change such as,
 * but not limited to: ADD, DELETE RENAME OR REPURPOSE any directive/option.
 *
 * Note: Update also Version.h !
 */
#define CONFIGURATION_H_VERSION 010100

#include "boards.h"
#include "macros.h"

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
[... 1217 lines cut ...]
 * Note may require analog pins to be defined for different motherboards
 **********************************************************************/
// Uncomment below to enable
//#define FILAMENT_WIDTH_SENSOR

#define DEFAULT_NOMINAL_FILAMENT_DIA 1.75  //Enter the diameter (in mm) of the filament generally used (3.0 mm or 1.75 mm) - this is then used in the slicer software.  Used for sensor reading validation

#if ENABLED(FILAMENT_WIDTH_SENSOR)
  #define FILAMENT_SENSOR_EXTRUDER_NUM 0   //The number of the extruder that has the filament sensor (0,1,2)
  #define MEASUREMENT_DELAY_CM        14   //measurement delay in cm.  This is the distance from filament sensor to middle of barrel

  #define MEASURED_UPPER_LIMIT         3.30  //upper limit factor used for sensor reading validation in mm
  #define MEASURED_LOWER_LIMIT         1.90  //lower limit factor for sensor reading validation in mm
  #define MAX_MEASUREMENT_DELAY       20     //delay buffer size in bytes (1 byte = 1cm)- limits maximum measurement delay allowable (must be larger than MEASUREMENT_DELAY_CM  and lower number saves RAM)

  #define DEFAULT_MEASURED_FILAMENT_DIA  DEFAULT_NOMINAL_FILAMENT_DIA  //set measured to nominal initially

  //When using an LCD, uncomment the line below to display the Filament sensor data on the last line instead of status.  Status will appear for 5 sec.
  //#define FILAMENT_LCD_DISPLAY
#endif

#include "Configuration_adv.h"
#include "thermistortables.h"

#endif //CONFIGURATION_H

alitai, 2016-07-06:
I've been running the points as they appear in the log - they are indeed spread evenly on the bed. 

I don't know why the actual G29 shifts so they run in a single quadrant. And at a slant...

Blue-Marlin, 2016-07-06:
For a test set your `Z_RAISE_BETWEEN_PROBINGS` to a very high value. 0,0 is usually not a point to probe but the place to go before the first probing. If you touch the bed here the tower coordinate system is messed up.

Blue-Marlin, 2016-07-06:
Alternately your `MANUAL_Z_HOME_POS` could be too high.

alitai, 2016-07-06:
@Blue-Marlin  - Thank you - I changed Z_RAISE_BETWEEN_PROBINGS from 5 to 25. While jumping higher between probes - It still touches the x0 y0 z0 point, and then moves to the next point in the negative (-) X direction. 

Indeed it looks as if the coordinate system is messed up - but when I printed some items (without G29) and they printed well - and centered. Can the coordinate system be messed up only for ABL?

alitai, 2016-07-06:
@Blue-Marlin  - If the MANUAL_Z_HOME_POS  would be too high then issuing a G0 Y0 X0 Z0 would touch the bed? It doesn't. I calibrated the printer so the nozzle is paper thickness from the glass bed.

Blue-Marlin, 2016-07-06:
Not for ABL. Because ABL touces 0,0 unplanned.
However - the code in this are changed last night. Please try the newest RCBugFix.

alitai, 2016-07-06:
I did install the latest today (about 2 hours ago). I'll see if there were any new commits since I last downloaded.

Interesting about having a different coordinates for ABL. How does one go about zeroing XY&Z for ABL?

thinkyhead, 2016-07-06:
Try setting `AUTO_BED_LEVELING_GRID_POINTS` to a more sane value like 3. There's no need to sample 81 points on the bed!

thinkyhead, 2016-07-06:
> Can the coordinate system be messed up only for ABL?

ABL takes a given idealized XYZ from your GCode and modifies the XYZ to produce a corrected position based on the tilt of the bed. Since ABL uses only a single tilt across the whole bed, it's important that the sampled points be accurate and that there be very little irregularity in the bed when you probe. Too much irregularity will give strange values.

Blue-Marlin, 2016-07-06:
@thinkyhead 
He has a delta. Here ABL is more like Mesh-Leveling. No matrix!

thinkyhead, 2016-07-06:
Ah… I'm not too familiar with the difference in Delta leveling.

alitai, 2016-07-07:
@thinkyhead The  setting is at 9 points   #define AUTO_BED_LEVELING_GRID_POINTS 9. It is similar to what I had before.

@Blue-Marlin ABL worked well on RC6 (the print itself didn't, though) and on a previous Marlin firmware I have been using for more than a year. The Bed is Glass, is rather flat, and the endstops are finely adjusted using a screws so I have a rather well calibrated system (if I say so myself - a 50mm test jig comes out at 50.08mm and symmetric). I use Hiwin rails which make the 1515 rather stable.

Since ABL worked on RC6 - I wonder what had changed since RC6?

alitai, 2016-07-07:
So I made a few movies of running ABL on the Bugfix version (where it fails) and the RC6 version (where it succeeds). 

An coincidental observation: Something during G29 is screwing up the Delta Coordinates - I stopped shooting the Bugfix movie right as it hit the GT2 belt. So to move it away from the tower I used the motor controls and discovered that the movement plane was completely off - a Y move resulted in a nearly 45 degree slant on Z. After using M28, the move was perfectly flat parallel to the bed plane.

thinkyhead, 2016-07-07:
Would you mind posting the G29 log from 1.1.0-RC6 so I can compare it to the log from RCBugFix? Meanwhile I will look more closely at the code. We did a huge amount of cleanup on bed leveling, and somewhere in the process we clearly overlooked something.

alitai, 2016-07-07:
@thinkyhead Thank you - I will do it in an hour or so when I get back.

alitai, 2016-07-07:
Here is the Log of the G29 from RC6:

22:27:56.143 : start
22:27:56.143 : echo:Marlin 1.1.0-RC6
22:27:56.143 : echo: Last Updated: 2016-04-24 12:00 | Author: (none, default config)
22:27:56.143 : Compiled: Jul  7 2016
22:27:56.143 : echo: Free Memory: 4285  PlannerBufferBytes: 1232
22:27:56.143 : echo:Hardcoded Default Settings Loaded
22:27:56.143 : echo:Steps per unit:
22:27:56.143 : echo:  M92 X200.00 Y200.00 Z200.00 E865.00
22:27:56.143 : echo:Maximum feedrates (mm/s):
22:27:56.143 : echo:  M203 X200.00 Y200.00 Z200.00 E25.00
22:27:56.143 : echo:Maximum Acceleration (mm/s2):
22:27:56.143 : echo:  M201 X9000 Y9000 Z9000 E10000
22:27:56.143 : echo:Accelerations: P=printing, R=retract and T=travel
22:27:56.143 : echo:  M204 P3000.00 R3000.00 T3000.00
22:27:56.143 : echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
22:27:56.143 : echo:  M205 S0.00 T0.00 B20000 X20.00 Z20.00 E20.00
22:27:56.143 : echo:Home offset (mm):
22:27:56.143 : echo:  M206 X0.00 Y0.00 Z0.00
22:27:56.143 : echo:Endstop adjustment (mm):
22:27:56.143 : echo:  M666 X0.00 Y0.00 Z0.00
22:27:56.143 : echo:Delta settings: L=diagonal_rod, R=radius, S=segments_per_second, ABC=diagonal_rod_trim_tower_[123]
22:27:56.143 : echo:  M665 L215.00 R105.30 S50.00 A0.00 B0.00 C0.00
22:27:56.143 : echo:Material heatup parameters:
22:27:56.143 : echo:  M145 S0 H180 B70 F255
22:27:56.143 : echo:  M145 S1 H240 B100 F255
22:27:56.143 : echo:PID settings:
22:27:56.143 : echo:  M301 P34.03 I4.36 D66.34 C100.00 L20
22:27:56.143 : echo:  M304 P679.07 I68.17 D1691.15
22:27:56.143 : echo:Filament settings: Disabled
22:27:56.143 : echo:  M200 D1.75
22:27:56.143 : echo:  M200 D0
22:27:56.143 : echo:Z-Probe Offset (mm):
22:27:56.143 : echo:  M851 Z0.00
22:27:56.328 : N1 M110*34
22:27:56.328 : N2 M115*36
22:27:56.328 : N3 M105*36
22:27:56.328 : N4 M114*35
22:27:56.343 : N5 M111 S6*98
22:27:56.345 : N6 T0*60
22:27:56.345 : N7 M20*22
22:27:56.345 : N8 M80*19
22:27:56.345 : N9 M105*46
22:27:58.479 : ok
22:27:58.531 : FIRMWARE_NAME:Marlin 1.1.0-RC6 (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Mini Kossel EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
22:27:58.531 : ok
22:27:58.531 : N10 M220 S100*80
22:27:58.531 : N11 M221 S100*80
22:27:58.534 : ok T:25.4 /0.0 B:26.3 /0.0 B@:0 @:0
22:27:58.534 : X:0.00 Y:0.00 Z:0.00 E:0.00 Count X: 37490 Y:37490 Z:37490
22:27:58.534 : ok
22:27:58.534 : N12 M111 S6*84
22:27:58.534 : echo:DEBUG:INFO,ERRORS
22:27:58.534 : ok
22:27:58.534 : echo:Active Extruder: 0
22:27:58.534 : ok
22:27:58.534 : ok
22:27:58.534 : ok
22:27:58.534 : N13 T0*8
22:27:58.535 : ok T:25.4 /0.0 B:26.3 /0.0 B@:0 @:0
22:27:58.547 : ok
22:27:58.547 : ok
22:27:58.547 : echo:DEBUG:INFO,ERRORS
22:27:58.547 : ok
22:27:58.547 : echo:Active Extruder: 0
22:27:58.547 : ok
22:27:59.392 : N14 M105*18
22:27:59.412 : ok T:25.5 /0.0 B:26.3 /0.0 B@:0 @:0
22:28:02.454 : N15 M105*19
[... 1072 lines cut ...]
22:31:12.346 : ok T:25.5 /0.0 B:26.3 /0.0 B@:0 @:0
22:31:15.389 : N77 M105*23
22:31:15.401 : ok T:25.5 /0.0 B:26.3 /0.0 B@:0 @:0
22:31:18.456 : N78 M105*24
22:31:18.474 : ok T:25.5 /0.0 B:26.2 /0.0 B@:0 @:0
22:31:21.514 : N79 M105*25
22:31:21.529 : ok T:25.5 /0.0 B:26.3 /0.0 B@:0 @:0
22:31:24.580 : N80 M105*31
22:31:24.585 : ok T:25.5 /0.0 B:26.3 /0.0 B@:0 @:0
22:31:27.641 : N81 M105*30
22:31:27.657 : ok T:25.5 /0.0 B:26.2 /0.0 B@:0 @:0
22:31:30.706 : N82 M105*29
22:31:30.714 : ok T:25.5 /0.0 B:26.3 /0.0 B@:0 @:0
22:31:33.766 : N83 M105*28
22:31:33.786 : ok T:25.5 /0.0 B:26.4 /0.0 B@:0 @:0
22:31:36.830 : N84 M105*27
22:31:36.842 : ok T:25.5 /0.0 B:26.2 /0.0 B@:0 @:0
22:31:39.891 : N85 M105*26
22:31:39.897 : ok T:25.5 /0.0 B:26.4 /0.0 B@:0 @:0
22:31:42.955 : N86 M105*25
22:31:42.969 : ok T:25.5 /0.0 B:26.4 /0.0 B@:0 @:0
22:31:46.013 : N87 M105*24
22:31:46.025 : ok T:25.5 /0.0 B:26.2 /0.0 B@:0 @:0
22:31:49.075 : N88 M105*23
22:31:49.081 : ok T:25.5 /0.0 B:26.3 /0.0 B@:0 @:0

thinkyhead, 2016-07-07:
From the video it looks like a weird assortment of probe points. Are they all within a single quadrant of the bed, or does it probe elsewhere?

Are you using the same `Configuration.h` file for `RCBugFix` as you are using for `RC6`, or did you modify it a bit?

thinkyhead, 2016-07-07:
Can you try enabling `BED_CENTER_AT_0_0` and disabling `Z_SAFE_HOMING` to see if it has any effect? (I doubt it will change anything, but I want to eliminate these factors, in case they are involved.)

thinkyhead, 2016-07-07:
So far in my examination of the G29 code nothing is jumping out as an obvious bug. I think it must be someplace in-between, maybe. If you do a normal `G28` before anything else and then send movement commands, does the delta go to all the right positions?

And when you do the `G29` leveling procedure does the LCD show the correct X, Y, Z coordinates?

This is making me think it would be nice to have an LCD option to cancel `G29` leveling in-progress.

thinkyhead, 2016-07-08:
@alitai I've created a branch for testing where I can add more logging to figure this out. I already did a few tweaks to it, so give it a try and see if it works any differently:

https://github.com/thinkyhead/Marlin/tree/th_delta_debug_homing_leveling

alitai, 2016-07-08:
@thinkyhead 
1. It seems to work in one quadrant only. I can't let it finish as it hits one of the towers. 
2. I always do a G28 prior to any G29 (just to be safe! My concern is hitting the bed prior to probing).
3. The movie above [Normal movement after G28: https://vimeo.com/173734397](https://vimeo.com/173734397) - shows that it works very well without a G29 - moving parallel to the bed in the Y direction. Just finished printing a sink stopper from flexible PLA and it printed very well.

alitai, 2016-07-08:
I actually tried to use the same Configuration.h. But the one I use now is different - there are quite a few differences between the RC6 and the Bugfix files - Hitachi character set was changed from Japan to Japanese; there were additional Z probe Raise parameters in RC6 that don't exist in the Bugfix, etc.

alitai, 2016-07-08:
@thinkyhead So I looked at the LCD - Indeed there is a mismatch between the LCD and where the effector/hotend is. For example, it starts off at Y=-49 (which should be close to the edge of the bed) - but the effector is at Y=0. 

So the first probe should be X=31 Y=-49 but it ends up being X=0 Y=0.

alitai, 2016-07-08:
Here is the log with the new debug build:

Printer reset detected - initalizing
start
echo:Marlin 1.1.0-RCBugFix
echo: Last Updated: 2016-04-27 12:00 | Author: (none, default config)
Compiled: Jul  8 2016
echo: Free Memory: 4303  PlannerBufferBytes: 1232
echo:Hardcoded Default Settings Loaded
echo:Steps per unit:
echo:  M92 X200.00 Y200.00 Z200.00 E865.00
echo:Maximum feedrates (mm/s):
echo:  M203 X200.00 Y200.00 Z200.00 E25.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X9000 Y9000 Z9000 E10000
echo:Accelerations: P=printing, R=retract and T=travel
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X20.00 Z20.00 E20.00
echo:Home offset (mm)
echo:  M206 X0.00 Y0.00 Z0.00
echo:Endstop adjustment (mm):
echo:  M666 X0.00 Y0.00 Z0.00
echo:Delta settings: L=diagonal_rod, R=radius, S=segments_per_second, ABC=diagonal_rod_trim_tower_[123]
echo:  M665 L215.00 R105.85 S200.00 A0.00 B0.00 C0.00
echo:Material heatup parameters:
echo:  M145 S0 H180 B70 F255
echo:  M145 S1 H240 B100 F255
echo:PID settings:
echo:  M301 P34.03 I4.36 D66.34 C100.00 L20
echo:  M304 P679.07 I68.17 D1691.15
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Z-Probe Offset (mm):
echo:  M851 Z0.00
N1 M110*34
N2 M115*36
N3 M105*36
N4 M114*35
N5 M111 S6*98
N6 T0*60
N7 M20*22
N8 M80*19
N9 M105*46
N10 M220 S100*80
FIRMWARE_NAME:Marlin 1.1.0-RCBugFix (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Mini Kossel EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
N11 M221 S100*80
N12 M111 S6*84
X:0.00 Y:0.00 Z:0.00 E:0.00 Count X: 37428 Y:37428 Z:37428
echo:DEBUG:INFO,ERRORS
echo:Active Extruder: 0
N13 T0*8
echo:DEBUG:INFO,ERRORS
echo:Active Extruder: 0
N14 M105*18
N15 M105*19
N16 M105*16
N17 M111 S32*102
echo:DEBUG:LEVELING
N18 M105*30
N19 M105*31
N20 G28*33
>>> gcode_G28
reset_bed_level
setup_for_endstop_or_probe_move > current_position: (0.00, 0.00, 0.00)
sync_plan_position > current_position: (0.00, 0.00, 0.00)
>>> homeaxis(0)
sync_plan_position > current_position: (0.00, 731.10, 731.10)
sync_plan_position > current_position: (0.00, 731.10, 731.10)
[... 510 lines cut ...]
<<< probe_pt
>>> probe_pt(15.00, 15.00, 1)
> current_position: (31.00, 15.00, 10.00)
do_probe_raise(10.00)
> do_blocking_move_to_xy(15.00, 15.00)
do_blocking_move_to: (15.00, 15.00, 10.00)
> set_probe_deployed > current_position: (15.00, 15.00, 10.00)
deploy: 1run_z_probe (DELTA) 1 > current_position: (15.00, 15.00, 10.00)
prepare_move_to_destination_raw > destination: (15.00, 15.00, -10.00)
run_z_probe (DELTA) 2 > current_position: (15.00, 15.00, -10.00)
sync_plan_position_delta > current_position: (15.00, 15.00, -10.00)
> do_probe_raise
do_probe_raise(10.00)
do_blocking_move_to: (15.00, 15.00, 10.00)
prepare_move_to_destination_raw > destination: (15.00, 15.00, 10.00)
<<< probe_pt
>>> probe_pt(-1.00, 15.00, 1)
> current_position: (15.00, 15.00, 10.00)
do_probe_raise(10.00)
> do_blocking_move_to_xy(-1.00, 15.00)
do_blocking_move_to: (-1.00, 15.00, 10.00)
> set_probe_deployed > current_position: (-1.00, 15.00, 10.00)
deploy: 1run_z_probe (DELTA) 1 > current_position: (-1.00, 15.00, 10.00)
prepare_move_to_destination_raw > destination: (-1.00, 15.00, -10.00)
Printer reset detected - initalizing

alitai, 2016-07-08:
FWIW, the debug version does print well [Printing a Flexible Sink Stopper](https://vimeo.com/173874500)

alitai, 2016-07-08:
So I wanted to try to isolate the problem - configuration or code. I took the RC6 configuration.h and configuration_adv.h and modified them to work on the Bugfix debug version that @thinkyhead sent.

That did not solve the problem - the probing started (incorrectly) at X0 Y0. So I guess it isn't a configuration issue.

Below are the changes that had to be done to the older (RC6) configuration.h and confiuration_adv.h to get them to compile successfully: 

Configuration.h

#define CONFIGURATION_H_VERSION 010100
#define DISPLAY_CHARSET_HD44780 JAPANESE 
#define FIX_MOUNTED_PROBE
#define Z_RAISE_PROBE_DEPLOY_STOW 50 // Raise to make room for the probe to deploy / stow
#define LCD_LANGUAGE en
#define XY_PROBE_SPEED 4000

configuration_adv.h
#define CONFIGURATION_ADV_H_VERSION 010100

alitai, 2016-07-08:
I also tried enabling #define BED_CENTER_AT_0_0 and disabling //#define Z_SAFE_HOMING. Didn't help.

thinkyhead, 2016-07-09:
I was thinking, if the probing was limited to a single quadrant, then maybe it was related to software endstops. But then, you would be limited in all your movements, regardless of bed leveling.

This is indeed quite a puzzle. I could use additional assistance from someone with a delta. I think @Roxy-3D has one…? I'm really stuck without a delta of my own to test with.

Roxy-3D, 2016-07-09:
I can help in a couple of days.   But right now, my Delta is missing one of its stepper motors.   I was trying to convert my FolgerTech i3-2020 to use 3mm filament and it was skipping steps.  So I wanted a higher torque stepper motor to see if that would let me use 3mm filament.    I canabolized my Delta to get a motor so I could keep going.   (Unfortunately, the higher torque motor didn't fix the skipped steps.)    In a few days I'll have the motor back where it belongs.

alitai, 2016-07-09:
I was skipping steps with RC6 (quite a lot actually) - but bugfix version solved that.

I am stumped too. The lift in Z seems to imply the coordinate translations work as well. The LCD shows the proper coordinates. It is as if an offset is introduced in the stepper counters...

Roxy-3D, 2016-07-09:
> I was skipping steps with RC6 (quite a lot actually) - but bugfix version solved that.

Has anybody else been talking about stepper motors losing steps in RC-6?   I know in my case it was just the extruder motor.  And I also know the stepper motor wasn't strong enough to push the heavier filament through the nozzle.    But for sure, we don't want this to be caused by the firmware!

thinkyhead, 2016-07-10:
@alitai Can you give a test of the probing commands outside of `G29` to see that they do the right thing?
```
 * G30 - Single Z probe, probes bed at current XY location.
 * M401 - Lower Z probe if present
 * M402 - Raise Z probe if present
```
Meanwhile I will try to figure out someplace where added logging might help.

AnHardt, 2016-07-10:
My best guess, for the moment, is the blocking_move_to(first probe point) from home position. In a not subdivided delta move its obvious - all axis move in negative direction - non of the endstops can be triggered. Due to the subdividing one of the axis might go a tiny bit upwards. Because the endstops are still triggered at home position the first of the subdivided moves is stopped, the coordinate system messed up.
For a test we could ask for adding an `endstop_adj` of a few mm to all axes to free the endstops before an other move can happen. Or we could simply try 
G28
G91
G0 Z-10
G90
G29

thinkyhead, 2016-07-10:
@AnHardt Another change we made recently was - for deltas that had a specific (large) Z before probing value, the effector would move down to that coordinate. But now we don't move down, only up, so we might have lost that initial move. Or, with delta, I might have made an exception and lowered to that point. I'll have to peek at the code to remind myself. But, perhaps that initial lowering is helpful.

alitai, 2016-07-10:
> @alitai Can you give a test of the probing commands outside of G29 to see that they do the right thing?

 * G30 - Single Z probe, probes bed at current XY location.
 * M401 - Lower Z probe if present
 * M402 - Raise Z probe if present
Meanwhile I will try to figure out someplace where added logging might help reveal something.<

So I probed at a few points outside the quadrant the ABL is stuck in such as x-35 and y-35 and it worked well. The nozzle touched the bed at the proper location - and returned properly.

alitai, 2016-07-10:
> For

 a test we could ask for adding an endstop_adj of a few mm to all axes to free the endstops before another move can happen. Or we could simply try:

G28
G91
G0 Z-10
G90
G29<

I tried this - It doesn't solve the problem - what should I be looking for?

AnHardt, 2016-07-10:
I hoped for a straight correct move from [0,0,z_max-x] to the first probe point in G29 with x bigger than a few 1/10mm. :-(
Did we already talk about `ENDSTOPS_ONLY_FOR_HOMING` (Configuration_adv.h)? Please alter it and test again.

alitai, 2016-07-10:
Tried quoting out - 
//#define ENDSTOPS_ONLY_FOR_HOMING // If defined the endstops will only be used for homing

And it made no difference. :-(

AnHardt, 2016-07-10:
Ok. Let's see what's going on.
https://github.com/AnHardt/Marlin/pull/56
Not as beauty as thinkyhead could have done it, but hopefully i can see what when happens.

thinkyhead, 2016-07-13:
@alitai Did you have a chance to test the debug code @AnHardt has provided?

alitai, 2016-07-13:
@thinkyhead @AnHardt I do apologize. I am rather new on GitHub and didn't know that the AnHardt#56 was a build for me to try. I was actually looking at this thread daily for the next set of instructions. I will test this later today.

alitai, 2016-07-13:
I am not sure how to use this. I tried showing the code for commit AnHardt#56 as full code and downloading it. Added it to the RCBugFix and to the AnHardt code tab download - and in both cases got a "conflicting decleration 'volatile boot_wait_for_heatup'" compiler error.

In the main AnHardt code tab - I don't see the report_current_position() calls that the #56 has...

alitai, 2016-07-13:
Tried quoting out the lines that compiler was unhappy with and made it even less happy...

I guess the question is how do I find and download the code that goes with the #56 commit?

thinkyhead, 2016-07-13:
At the top, notice that the name of the branch that AnHardt#56 comes from is [rep-pos-delta-G29](https://github.com/AnHardt/Marlin/tree/rep-pos-delta-G29) (https://github.com/AnHardt/Marlin/tree/rep-pos-delta-G29).

alitai, 2016-07-13:
@thinkyhead - Thank you - downloaded it and it worked. Posted the log and configuration.h into the AnHardt#56 message.

BTW - On a side note, the configuration.h version may need to change as CONFIGURATION_H_VERSION 010100 no longer compiles with the code. Missing some declarations.

alitai, 2016-07-13:
@anhardt figured it out - the effector is trying to move to the initial point while still homed. So the carriages hit the endpoints. A move to point X:31.00 Y:-49.00 Z:243.70 E:0.00 Count X: 86168 Y:86248 Z:85995 remains at X:0 Y:0....

In the past, the effector used to go down prior to probing. Is that settable in configuration.h? 

There is a workaround for now - running:
G28
G0 Z100
G29

works, and completes a full G29 cycle.

Thank you @AnHardt, @thinkyhead, @Roxy-3D and @Blue-Marlin for all your help - you guys are awesome.

alitai, 2016-07-13:
Should I close this issue - or should we close it only after the issue is resolved in code (or configuration)?

alitai, 2016-07-14:
@anhardt - I apologize again - You had asked about this before - "For a test we could ask for adding an  endstop_adj  of a few mm to all axes to free the endstops before another move can happen. Or we could simply try:

G28
G91
G0 Z-10
G90
G29"

Only now do I see what you meant - when the effector tries to get to the coordinates it does hit the endstops. you were on to the problem 4 days ago...

thinkyhead, 2016-07-14:
> the effector used to go down prior to probing

Yes, this was recently (unintentionally) changed. It was assumed that the initial height was purely for probe deployment, and that movement to the first XY point would occur afterward. Clearly on a DELTA the initial "raise" needs to be applied even if it is lower.

thinkyhead, 2016-07-14:
#4303 is almost the right idea, but....

Background: I believe the upper part of a delta's total build volume is actually a cone, so the full volume looks like a pointy rocket. Marlin currently supports making full use of that limited upper-region of the build area, as long as you don't try to move any of the carriages beyond their upper limit. So you can print a taller Chrysler Building than a cylinder.

That said (and apropos of the corporate engineering lab) I think it's good to place safety margins. But… I think enforcing a fixed height based on where the nozzle can reach the outer edge is probably too strict.

Compare to software endstops on a nice square Cartesian. They are easy to enforce, but ultimately meaningless on a delta. Instead we can constrain based on the radius (no `sqrt` is required). We can account for the upper conical region, which starts at the full radius at the safe-distance figured in #4303, and descends linearly to a single point at the Z_MAX position).

jbrazio, 2016-07-23:
@thinkyhead since #4303 is merged, can we close this ?

closed 2016-07-23 by thinkyhead

alitai, 2016-08-10:
RC7 works really well on my Mini Kossel - Thank you all for the excellent work!