MarlinFirmware/Marlin #4208: Raising twice before leveling bed after Z_RAISE_PROBE_DEPLOY_STOW change

token47, 2016-07-05:
The behaviour of how much (and when) to rise before homing and bed leveling has changed in the last few days (I think it was on the same change as Z_RAISE_PROBE_DEPLOY_STOW was renamed), and I'm not sure it was intentional because it seems worse now (or maybe there's a good reason and I don't see it).

Behaviour before the change: It would raise until minimum Z height, home X and Y, and if that height was enough for deploying the probe it would just deploy, else it would raise a bit more to make clearance for the probe.

Behaviour now: it will raise until minimum Z height, home X and Y, and then raise again the value of Z_RAISE_PROBE_DEPLOY_STOW regardless of the current height being already enough to deploy the probe. So it will always raise twice, which is tedious and slow. I'm tempted to put 0 (zero) on the Z_RAISE_PROBE_DEPLOY_STOW value since the homing height is enough to deploy the probe but I'm not sure about the consequences.

It seems that Z_RAISE_PROBE_DEPLOY_STOW is now relative to the homing minimum height and before it was abolute to zero.

I checked the comments on the configuration file and they still seem to imply old behaviour.

AnHardt, 2016-07-05:
@thinkyhead 

I don't know if we changed the order here recently. 
A relative to `current_position` move in `deploy_z_probe()` does not make much sense when we set `current_position[axis] = 0;` before.

In homexis()
```diff
-    // Set the axis position as setup for the move
-    current_position[axis] = 0;
-    sync_plan_position();

    // Homing Z towards the bed? Deploy the Z probe or endstop.
    #if HAS_BED_PROBE
      if (axis == Z_AXIS && axis_home_dir < 0) {
        #if ENABLED(DEBUG_LEVELING_FEATURE)
          if (DEBUGGING(LEVELING)) SERIAL_ECHOPGM("> ");
        #endif
        deploy_z_probe();
      }
    #endif

+    // Set the axis position as setup for the move
+    current_position[axis] = 0;
+    sync_plan_position();

   // Set a flag for Z motor locking
    #if ENABLED(Z_DUAL_ENDSTOPS)
      if (axis == Z_AXIS) stepper.set_homing_flag(true);
    #endif

    // Move towards the endstop until an endstop is triggered
    destination[axis] = 1.5 * max_length(axis) * axis_home_dir;
    feedrate = homing_feedrate[axis];
    line_to_destination();
    stepper.synchronize();
```

lrpirlet, 2016-07-05:
@@AnHardt <https://github.com/AnHardt> , do we realy need to set the position to ZERO at all ??? I was to test, when I have time, if the suppression of those lines was ok… 

Even if X, Y or Z axis is at  max_length(axis) we move it by 150% of that max_length toward the end-stop, so it should end up to the end-stop anyway… I was looking for the possible side effects (and had find none yet , but I may be wrong cause I have not tested it)

token47, 2016-07-05:
Here is my configuration. I cleaned it up to leave only the parts I find relevant to this matter. If I missed anything please ask.

The config before the issue is RC-BugFix from 2016-06-09 11:10pm. The config after the issue is RC-BugFix from 2016-07-02 4:05pm.

Before:

```
-#define Z_RAISE_BEFORE_PROBING 15   // Raise before probe deploy (e.g., the first probe).
-#define Z_RAISE_AFTER_PROBING 15    // Raise before probe stow (e.g., the last probe).
+#define Z_RAISE_BEFORE_PROBING 10   // Raise before probe deploy (e.g., the first probe).
+#define Z_RAISE_AFTER_PROBING 10    // Raise before probe stow (e.g., the last probe).

-//#define MIN_Z_HEIGHT_FOR_HOMING 4 // (in mm) Minimal z height before homing (G28) for Z clearance above the bed, clamps, ...
+#define MIN_Z_HEIGHT_FOR_HOMING 15 // (in mm) Minimal z height before homing (G28) for Z clearance above the bed, clamps, ...

-//#define AUTO_BED_LEVELING_FEATURE // Delete the comment to enable (remove // at the start of the line)
+#define AUTO_BED_LEVELING_FEATURE // Delete the comment to enable (remove // at the start of the line)

-  #define X_PROBE_OFFSET_FROM_EXTRUDER 10  // X offset: -left  +right  [of the nozzle]
-  #define Y_PROBE_OFFSET_FROM_EXTRUDER 10  // Y offset: -front +behind [the nozzle]
-  #define Z_PROBE_OFFSET_FROM_EXTRUDER 0   // Z offset: -below +above  [the nozzle]
+  #define X_PROBE_OFFSET_FROM_EXTRUDER +39      // X offset: -left  +right  [of the nozzle]
+  #define Y_PROBE_OFFSET_FROM_EXTRUDER +20      // Y offset: -front +behind [the nozzle]
+  #define Z_PROBE_OFFSET_FROM_EXTRUDER -5.20    // Z offset: -below +above  [the nozzle]

-  #define XY_TRAVEL_SPEED 8000         // X and Y axis travel speed between probes, in mm/min.
+  #define XY_TRAVEL_SPEED 10000 //8000         // X and Y axis travel speed between probes, in mm/min.

-  #define Z_RAISE_BETWEEN_PROBINGS 5  // How much the Z axis will be raised when traveling from between next probing points.
+  #define Z_RAISE_BETWEEN_PROBINGS 10  // How much the Z axis will be raised when traveling from between next probing points.

-  //#define FIX_MOUNTED_PROBE
+  #define FIX_MOUNTED_PROBE

-//#define Z_SAFE_HOMING
+#define Z_SAFE_HOMING

-//#define Z_DUAL_STEPPER_DRIVERS
+#define Z_DUAL_STEPPER_DRIVERS

-//#define QUICK_HOME  //if this is defined, if both x and y are to be homed, a diagonal move will be performed initially.
+#define QUICK_HOME  //if this is defined, if both x and y are to be homed, a diagonal move will be performed initially.
```

After:

```
-//#define Z_ENDSTOP_SERVO_NR 0
-//#define Z_SERVO_ANGLES {70,0} // Z Servo Deploy and Stow angles
+#define Z_ENDSTOP_SERVO_NR 0
+#define Z_SERVO_ANGLES {100,65} // Z Servo Deploy and Stow angles

-#define X_PROBE_OFFSET_FROM_EXTRUDER 10  // X offset: -left  +right  [of the nozzle]
-#define Y_PROBE_OFFSET_FROM_EXTRUDER 10  // Y offset: -front +behind [the nozzle]
-#define Z_PROBE_OFFSET_FROM_EXTRUDER 0   // Z offset: -below +above  [the nozzle]
+#define X_PROBE_OFFSET_FROM_EXTRUDER +43 // X offset: -left  +right  [of the nozzle]
+#define Y_PROBE_OFFSET_FROM_EXTRUDER +30 // Y offset: -front +behind [the nozzle]
+#define Z_PROBE_OFFSET_FROM_EXTRUDER -1.00 // Z offset: -below +above  [the nozzle]

-#define XY_PROBE_SPEED 8000
+#define XY_PROBE_SPEED 10000

-//#define Z_MIN_PROBE_REPEATABILITY_TEST
+#define Z_MIN_PROBE_REPEATABILITY_TEST

-#define Z_RAISE_PROBE_DEPLOY_STOW 15 // Raise to make room for the probe to deploy / stow
+#define Z_RAISE_PROBE_DEPLOY_STOW 5 // Raise to make room for the probe to deploy / stow
 #define Z_RAISE_BETWEEN_PROBINGS 5  // Raise between probing points.

-//#define MIN_Z_HEIGHT_FOR_HOMING 4 // (in mm) Minimal z height before homing (G28) for Z clearance above the bed, clamps, ...
+#define MIN_Z_HEIGHT_FOR_HOMING 10 // (in mm) Minimal z height before homing (G28) for Z clearance above the bed, clamps, ...

-//#define AUTO_BED_LEVELING_FEATURE // Delete the comment to enable (remove // at the start of the line)
+#define AUTO_BED_LEVELING_FEATURE // Delete the comment to enable (remove // at the start of the line)

-//#define Z_SAFE_HOMING
+#define Z_SAFE_HOMING

-//#define NUM_SERVOS 3 // Servo index starts with 0 for M280 command
+#define NUM_SERVOS 1 // Servo index starts with 0 for M280 command

-//#define DEACTIVATE_SERVOS_AFTER_MOVE
+#define DEACTIVATE_SERVOS_AFTER_MOVE

-//#define Z_DUAL_STEPPER_DRIVERS
+#define Z_DUAL_STEPPER_DRIVERS

-//#define QUICK_HOME  //if this is defined, if both x and y are to be homed, a diagonal move will be performed initially.
+#define QUICK_HOME  //if this is defined, if both x and y are to be homed, a diagonal move will be performed initially.

```

Note: I changed from a capacitive fixed probe to a servo based optical probe between these versions, but I got the chance to test the new version with the capacitive still and the results were the same. As you can see I had a "Z_RAISE_BEFORE_PROBING" of 10 even when using the capacitive probe and didn't notice it because the MIN_Z_HEIGHT_FOR_HOMING was 15 so it was always higher and this didn't make any difference. At the moment I changed to the newer version, it started to raise 10mm more between homing X+Y and homing Z.

Do you want me to make any specific test or see anything wrong in my config?

thinkyhead, 2016-07-05:
> do we really need to set the position to ZERO at all?

Probably not. But it will do no harm either. It's always possible that someone could do a `G92 Z9999` before `G28`, which would confuse it. Setting it to zero here will at least work around that kind of possibility.

thinkyhead, 2016-07-05:
> Do you want me to make any specific test or see anything wrong in my config?

The patch from above has been merged now. So using the very latest code, try enabling `DEBUG_LEVELING_FEATURE` and then use `M111 S32` to enable extra logging. Then when you execute a `G28` or `G29` we can see what Marlin is doing with your settings.

token47, 2016-07-05:
Well, I just downloaded the RCBugFix of hald an hour ago, and can't reproduce the problem. It is working as it was before (as expected). Even if I put "10" both on MIN_Z_HEIGHT_FOR_HOMING and Z_RAISE_BEFORE_PROBING, it will only raise once, before homing X/Y. After that, it just deploys the probe (at it seems to already be high enough for that) and homes Z.

If I understand correct, the patch by AnHardt did the trick? (I checked and it is applied). I'm testing more to make sure it's ok.

token47, 2016-07-05:
Humm, something is strange. I thought it was raising too much, then I mesured it. Right now the relevante parameters are:

```
#define Z_ENDSTOP_SERVO_NR 0
#define Z_SERVO_ANGLES {100,65}
#define X_PROBE_OFFSET_FROM_EXTRUDER +43
#define Y_PROBE_OFFSET_FROM_EXTRUDER +30
#define Z_PROBE_OFFSET_FROM_EXTRUDER -1.00
#define XY_PROBE_SPEED 10000
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define Z_MIN_PROBE_REPEATABILITY_TEST
#define Z_RAISE_PROBE_DEPLOY_STOW 6
#define Z_RAISE_BETWEEN_PROBINGS 4
#define MIN_Z_HEIGHT_FOR_HOMING 6
#define AUTO_BED_LEVELING_FEATURE
#define AUTO_BED_LEVELING_GRID
#define Z_SAFE_HOMING
#define NUM_SERVOS 1
#define DEACTIVATE_SERVOS_AFTER_MOVE
```

On the very first move it raises about 9mm (mesured with a ruler, not exact) on just one go. I would expect it to raise just 6mm.

I'm running with the DEBUG now.

token47, 2016-07-05:
Where does the debug information go? I enabled it (DEBUG_LEVELING_FEATURE), confirmed I'm compiling the right tree but can't see any change in the output of the communication log. I'm trying a G28 and a G29.

thinkyhead, 2016-07-05:
You need to use `M111 S32` to enable logging of homing and leveling.

token47, 2016-07-05:
Hehe, excuse the newbie in me. Now I got it:

```
Attempting connection at \\.\COM3...
RECEIVED: start
  Connected to machine!
RECEIVED: echo: External Reset
SENT: T0
RECEIVED: Marlin 1.1.0-RCBugFix
RECEIVED: echo: Last Updated: 2016-04-27 12:00 | Author: (Token, my config)
Last Updated: 2016-04-27 12:00 | Author: (Token, my config)
RECEIVED: Compiled: Jul  5 2016
RECEIVED: echo: Free Memory: 3256  PlannerBufferBytes: 1232
Free Memory: 3256  PlannerBufferBytes: 1232
RECEIVED: echo:V24 stored settings retrieved (398 bytes)
V24 stored settings retrieved (398 bytes)
RECEIVED: echo:Steps per unit:
Steps per unit:
RECEIVED: echo:  M92 X80.00 Y80.00 Z4000.00 E620.00
M92 X80.00 Y80.00 Z4000.00 E620.00
RECEIVED: echo:Maximum feedrates (mm/s):
Maximum feedrates (mm/s):
RECEIVED: echo:  M203 X300.00 Y300.00 Z5.00 E25.00
M203 X300.00 Y300.00 Z5.00 E25.00
RECEIVED: echo:Maximum Acceleration (mm/s2):
Maximum Acceleration (mm/s2):
RECEIVED: echo:  M201 X3000 Y3000 Z100 E10000
M201 X3000 Y3000 Z100 E10000
RECEIVED: echo:Accelerations: P=printing, R=retract and T=travel
Accelerations: P=printing, R=retract and T=travel
RECEIVED: echo:  M204 P3000.00 R3000.00 T3000.00
M204 P3000.00 R3000.00 T3000.00
RECEIVED: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
RECEIVED: echo:  M205 S0.00 T0.00 B20000 X15.00 Z0.40 E5.00
M205 S0.00 T0.00 B20000 X15.00 Z0.40 E5.00
RECEIVED: echo:Home offset (mm)
Home offset (mm)
RECEIVED: echo:  M206 X0.00 Y0.00 Z0.00
M206 X0.00 Y0.00 Z0.00
RECEIVED: echo:Material heatup parameters:
Material heatup parameters:
RECEIVED: echo:  M145 S0 H205 B60 F0
M145 S0 H205 B60 F0
RECEIVED: echo:  M145 S1 H235 B100 F0
M145 S1 H235 B100 F0
RECEIVED: echo:PID settings:
PID settings:
RECEIVED: echo:  M301 P11.14 I0.48 D64.62 C100.00 L20
M301 P11.14 I0.48 D64.62 C100.00 L20
RECEIVED: echo:  M304 P356.02 I35.51 D892.46
M304 P356.02 I35.51 D892.46
RECEIVED: echo:Filament settings: Disabled
Filament settings: Disabled
RECEIVED: echo:  M200 D3.00
M200 D3.00
RECEIVED: echo:  M200 D0
M200 D0
RECEIVED: echo:Z-Probe Offset (mm):
Z-Probe Offset (mm):
RECEIVED: echo:  M851 Z-2.80
M851 Z-2.80
SENT: M105
RECEIVED: echo:SD card ok
RECEIVED: echo:Active Extruder: 0
Active Extruder: 0
RECEIVED: ok
RECEIVED: echo:SD card ok
RECEIVED: ok T:19.4 /0.0 B:22.2 /0.0 B@:0 @:0
SENT: M105
[... 90 lines cut ...]
RECEIVED: do_probe_raise(6.00)
RECEIVED: do_blocking_move_to: (54.00, 67.00, 8.80)
RECEIVED: echo:busy: processing
busy: processing
SENT: M105
RECEIVED: <<< homeaxis(2)
RECEIVED: <<< Z_SAFE_HOMING
RECEIVED: > (home_all_axis || homeZ) > final > current_position: (54.00, 67.00, 8.80)
RECEIVED: sync_plan_position > current_position: (54.00, 67.00, 8.80)
RECEIVED: clean_up_after_endstop_or_probe_move > current_position: (54.00, 67.00, 8.80)
RECEIVED: <<< gcode_G28
RECEIVED: X:54.00 Y:67.00 Z:8.80 E:0.00 Count X: 4320 Y:5360 Z:35200
RECEIVED: ok
RECEIVED: ok T:19.5 /0.0 B:22.1 /0.0 B@:0 @:0
RECEIVED: ok T:19.5 /0.0 B:22.1 /0.0 B@:0 @:0
RECEIVED: ok T:19.5 /0.0 B:22.1 /0.0 B@:0 @:0
RECEIVED: ok T:19.5 /0.0 B:22.1 /0.0 B@:0 @:0
RECEIVED: ok T:19.5 /0.0 B:22.1 /0.0 B@:0 @:0
RECEIVED: ok T:19.5 /0.0 B:22.1 /0.0 B@:0 @:0
RECEIVED: ok T:19.5 /0.0 B:22.1 /0.0 B@:0 @:0
RECEIVED: ok T:19.5 /0.0 B:22.1 /0.0 B@:0 @:0
SENT: M105
RECEIVED: ok T:19.6 /0.0 B:22.3 /0.0 B@:0 @:0

```

token47, 2016-07-05:
Hummm, 2.80 is my Z  Offset (stored on the EEPROM). I guess it's taking it into account.

thinkyhead, 2016-07-05:
Yes. Your EEPROM values override your default (`Configuration.h`) values.

token47, 2016-07-06:
Does it makes sense to add it to the value to raise? I think it's right to check if the MIN_Z_HEIGHT_FOR_HOMING is at least => the Z offset or bad things will happen, but if I configure a value greater I'm not expecting it to be increased further because I already considered that offset.

thinkyhead, 2016-07-06:
It may be that `MIN_Z_HEIGHT_FOR_HOMING` has not been considered in recent changes. I will check to see that it is being applied properly. In general it is not a raise to a certain position. It is a "blind" raise to perform at the start of `G28`.

thinkyhead, 2016-07-06:
`MIN_Z_HEIGHT_FOR_HOMING` is just the initial raise to do before homing, meant to get the nozzle off the bed and above the bed clips before homing XY. That is its only purpose. So, you should get an initial raise by that amount as you start `G28`. Then it will home X and Y. Then before it tries to home Z with a probe it will raise up as much as needed (`Z_RAISE_PROBE_DEPLOY_STOW`) to deploy the probe.

token47, 2016-07-06:
Yes, but that is not the case. Both are set to 6. It's only going up bu 8.8 because (and I'm just guessing here) it's adding 2.8 to the value of 6 I provided. And it must be taking this 2.8 from my Z offset (on the EEPROM), but I see no point to add that when the initial value is big enough. It could check and not let it be smaller (sanity check at compile time), but once being equal or greater it could just raise 6 (what I configured). Right?

thinkyhead, 2016-07-06:
>  It's only going up by 8.8 because (and I'm just guessing here) it's adding 2.8 to the value of 6 I provided.

The use of the probe raise in the homing function is incorrect, so that will change. It should raise to `home_offset[Z_AXIS] + 6` and should not include the probe Z offset.

token47, 2016-07-06:
Thinking about it for a moment, I can see the rationale behind this change. The value to raise before deploying the probe should not be less than Z offset or else the probe will/can crash. But the value of Z offset can be changed on the fly on the LCD display so it's not possible to just make a sanity check on compilation time. And it's clunky to try to display an error later (or worse, not obey a homing command because of some transient value set at runtime, after booting). So it seems easier to just add and do the homing, it will not hurt to raise a little more and things get easier.

If this is the desired behaviour, I guess it's ok, but the documentation should be clarified to state that that value will still be increased by the z offset value, and why.

thinkyhead, 2016-07-06:
All the Z raises have recently been much more generalized. They now indicate, more accurately, the clearance between the probe and the bed. Or if the nozzle is lower than the probe, the clearance between the nozzle and the bed. So `do_probe_raise(10)` makes sure there are always 10mm between the bed and the lowest part of your extruder/probe assembly.

closed 2016-07-07 by thinkyhead