MarlinFirmware/Marlin #4210: Is the Genuino 101 a viable successor?

paulusjacobus, 2016-07-05:
The Genuino 101 is now out. It has a duo core processor, 32 Mhz 196krom, 80 k sram memory, 4 Pwm and has the uno form factor.  It is based on the intel curie processor from intel. I saw the price $29 in the usa. Here in Australia it is about $68.   @Roxy-3D you will be pleased to see that it has a real RTOS on board. 
So from a first glance at the arduino.cc site it seems a good contender. Since it has the uno form factor, you should be able use the cnc shield that supports 4 steppers. For powering the extruder and bed you probably need to build a custom shield or some one could design a ramps 1.5? Anyone who has done some more analysis on it applicability?

Blue-Marlin, 2016-07-05:
If it has UNO for factor we'll probably miss lots of pins.

WZ9V, 2016-07-05:
Considering all the other 32-bit players are aiming way higher I think anything less than a Due/RADDS type combo is doomed to fail unless you are just trying to be the cheapest 32-bit alternative.

jbrazio, 2016-07-05:
I believe our current issue is not finding a 32bit ARM Arduino (blink blink Due) but to find a good interface between the uC board and all the required components.. as discussed previously we need a interface shield which allows us to connect all 5V/12V parts (fans, LCD, whatever) and interface that with the 3.3V logic of the ARM. The second important job of the shield is to deal with the reverse logic of the IO pins..

This is key as most people only need a new Due and a new shield, all the rest is the same.

Now, the need is there, the Arduino uC is there.. we just need the new shield. :-)

paulusjacobus, 2016-07-05:
@jbrazio, The genuino 101 has the nice option to work with 5v inputs/outputs despite the processor working with 3.3v.
@Blue-Marlin yes, the write read operation came to my mind as well. The arduino family allows faster io and not sure if these fast tricks are available to the curie processor.

Drakelive, 2016-07-05:
> At first glance it seems strange a new 32-Bit processor is only running at
> 32MHz. But I guess that is because it is meant to be very low power because
> this is something Intel wants people to wear on their cloths or sports
> equipment. It is kind of cool that it has a 6-Axis gyroscope built into it,
> but I think I would trade that for higher processing speed.
>
> At this address I have a question:
> https://www.arduino.cc/en/Guide/Arduino101#toc2
>
> Although the 101 shares the same peripherals and connectors as the Arduino
> Uno, the Uno's ATmega328 has been replaced by Intel® Curie™, meaning that
> the board now has two cores, an x86 (Quark) and an ARC (Argonaut RISC
> Core).
>
> So the Genuino 101 has two cores in it, and they are different? That would
> be complicated to make use of.

Hi

Genuino 101  support is just a waste of time.
I think we should works to support Arduino DUE and STM32 ( Nucleo boards? )
which allow a considerable increase in power.
We should add a Hardware Abstraction Layer (HAL) for each platform.
@Magokimbra has already implemented in your GitHub the HAL for AVR and
SAM3X8E (Arduino DUE)
We start from this to include in Marlin.

Drk

jbrazio, 2016-07-05:
Like I said.. this thread is not bringing nothing new into the discussion. ;-)

WZ9V, 2016-07-05:
The Due has the RADDS board and bobc has been doing some work to finish his original RAMPS-FD design check over on RepRap forums.  If his 2.2 design is ready perhaps we could gather enough interest to have someone make a small run.

WZ9V, 2016-07-05:
BTW, They went M4 processor for movement and ESP2866 for web interface on the new Duet design.  I think Smoothie2 is doing something similar.

paulusjacobus, 2016-07-05:
Ok seems we have an answer. Genuino 101 is not viable because of better alternatives like DUE

closed 2016-07-05 by paulusjacobus