MarlinFirmware/Marlin #4213: Extruder runs off bed on RCBugFix

meros, 2016-07-05:
For RCBugFix, using 1236a3b3ab08d83147b9192fbdf850b45b11d07a I get normals prints, but when I rebased my configs on latest RCBugFix (d4fda424b7810082d685c581cb1068d84f976677) the extruder runs off the bed shortly after print.

thinkyhead, 2016-07-05:
That first link makes sense - July 2, 2016. The second link goes to a commit from April 24. Are your links reversed?

What about the current `RCBugFix` as of today? How is that behaving?

Are you printing from USB or SD? If printing over USB, what `BAUD` rate are you using?

meros, 2016-07-05:
Sorry, messed up my git hashes :)

I rebased my config on RCBugFix today (see updated comment). That didn't work. The one I am printing with now, thats working, is the first link (a0b00d2).

meros, 2016-07-05:
I will try the very latest and come back to you (I thought I did, but I see now there are more commits on top)

meros, 2016-07-16:
Sorry for late feedback, I retried with latest as of today, and there seem to be no issue anymore

jbrazio, 2016-07-16:
Great ! Thanks for the feedback.

closed 2016-07-16 by jbrazio