MarlinFirmware/Marlin #4216: What  is  the  relation  between   e-stepper movement  and  x&y  stepper  movement ?

lizlovesspurs, 2016-07-06:
I am a student from China and I am learning Marlin by myself recently. I can not fully understand the source code. I want to know if the e-stepper movement and x&y stepper movement are independent of each other. I also want to know if e-stepper movement is exactly according to E value in G-code. Because plastic can not be melt fast by the extruder

when x and y steppers are moving. I hope developers can help me. Thank you very much!

thinkyhead, 2016-07-06:
With the current Marlin code you must enable the `LIN_ADVANCE` feature to have E movement managed separately from XYZ movement. Without this feature enabled, the E axis moves in direct proportion to the other axes.

Roxy-3D, 2016-07-06:
> I also want to know if e-stepper movement is exactly according to E value in G-code. Because plastic can not be melt fast by the extruder

If you have a Prusa i3 available , put the bed somewhere in the middle of its print envelope.   (I know what I'm telling you to do here is safe on a Prusa i3.)   Gently press harder and harder on the rigid part of the bed.   You will feel the stepper motor deflect more and more as you press harder and harder.  Eventually you will cause the motor to skip.   But up until that point, the motor will act as a spring and be trying to get to the right position.   The extruder motor will behave the same way putting more pressure on the filament until it is close to where it is told to be.

lizlovesspurs, 2016-07-06:
@thinkyhead  Thank   you  for   your  answer!  As  for   the   proportion  you  mentioned ,  could   you  please  tell  me  where  the  proportion  or   some  formulas  or  codes  about  the  relation between  E  axis  movement  and  other  axis  movement  is   in  Marlin  source code. I  want  to  know  some  details  about  it. If  you  can  tell  me  about  which  file   I  can  find  details  in  Marlin, I   will  appreciate  it

lizlovesspurs, 2016-07-06:
@Roxy-3D     Thank  you   for  your   explanation!  I  have  a  makerbot  machine  and   I  can  understand  your  description. It  is  very  vivid!  I  also  have  a  thought  that  the  more  speed  e-stepper  has ,  the  more   pressure is on  the  filament   by  the  extruder.  Is  my  thought  right? Otherwise,  which  factors  affect  the  pressure  on  the  filament?

Blue-Marlin, 2016-07-06:
The heart of Marlin and the proportionality is the 4 dimensional 'Bresenham' algorithm in the stepper ISR in stepper.cpp. The 'blocks' are set up in planner.cpp

Blue-Marlin, 2016-07-06:
For more information about the pressure search for 'ADVANCE' and 'LIN_ADVANCE' here and in the pull requests.

Roxy-3D, 2016-07-06:
> I also have a thought that the more speed e-stepper has , the more pressure is on the filament by the extruder. Is my thought right?

The more speed the extruder stepper motor can handle, the quicker it can increase (and decrease) the pressure.   But the pressure can only be increased so much and the nozzle will lose steps.  There are limitations.    Please take Blue-Marlin's suggestion to study the Bresenham algorithm and various implementations of it.   That will help you to understand all of the issues.

closed 2016-07-16 by jbrazio