MarlinFirmware/Marlin #4219: LCD Smart Controller works but no menu

russellchan123, 2016-07-06:
I have a full discount smart controller that displays the home screen with the nozzle temperature and the axis coordinates. It even displays the Marlin logo upon startup. However when I try to scroll and click nothing happens. There is no menu. It just stays on the same screen. I have added U8glib.h to my libraries and went through the configuration.h files very thoroughly. If anyone knows a solution I would greatly appreciate it. Thanks.

Blue-Marlin, 2016-07-06:
Please append your Configuration.h and if edited, your used pins.h file.

thinkyhead, 2016-07-06:
- What version of Marlin are you using?

russellchan123, 2016-07-06:
I am using Marlin 1.1.0-RC6. My LCD code is below: 
#define LANGUAGE_INCLUDE GENERATE_LANGUAGE_INCLUDE(en)
// LCD CHARACTER SET
// Choose ONE of the following charset options. This selection depends on
// your physical hardware, so it must match your character-based LCD.
//
// Note: This option is NOT applicable to graphical displays.
//
// To find out what type of display you have:
//  - Compile and upload with the language (above) set to 'test'
//  - Click the controller to view the LCD menu
//
// The LCD will display two lines from the upper half of the character set.
//
// See also https://github.com/MarlinFirmware/Marlin/wiki/LCD-Language
//
#define DISPLAY_CHARSET_HD44780_JAPAN        // this is the most common hardware
//#define DISPLAY_CHARSET_HD44780_WESTERN
//#define DISPLAY_CHARSET_HD44780_CYRILLIC
//
// LCD TYPE
//
// You may choose ULTRA_LCD if you have character based LCD with 16x2, 16x4, 20x2,
// 20x4 char/lines or DOGLCD for the full graphics display with 128x64 pixels
// (ST7565R family). (This option will be set automatically for certain displays.)
//
// IMPORTANT NOTE: The U8glib library is required for Full Graphic Display!
//                 https://github.com/olikraus/U8glib_Arduino
//
//#define ULTRA_LCD   // Character based
#define DOGLCD      // Full graphics display
//
// SD CARD
//
// SD Card support is disabled by default. If your controller has an SD slot,
// you must uncomment the following option or it won't work.
//
#define SDSUPPORT
//
// SD CARD: SPI SPEED
//
// Uncomment ONE of the following items to use a slower SPI transfer
// speed. This is usually required if you're getting volume init errors.
//
//#define SPI_SPEED SPI_HALF_SPEED
//#define SPI_SPEED SPI_QUARTER_SPEED
//#define SPI_SPEED SPI_EIGHTH_SPEED
//
// SD CARD: ENABLE CRC
//
// Use CRC checks and retries on the SD communication.
//
//#define SD_CHECK_AND_RETRY
//
// ENCODER SETTINGS
//
// This option overrides the default number of encoder pulses needed to
// produce one step. Should be increased for high-resolution encoders.
//
#define ENCODER_PULSES_PER_STEP 1

//
// Use this option to override the number of step signals required to
// move between next/prev menu items.
//
//#define ENCODER_STEPS_PER_MENU_ITEM 5

//
// This option reverses the encoder direction for navigating LCD menus.
// By default CLOCKWISE == DOWN. With this enabled CLOCKWISE == UP.
[... 144 lines cut ...]
// For more info: https://github.com/lincomatic/LiquidTWI2
//
// Note: The PANELOLU2 encoder click input can either be directly connected to
// a pin (if BTN_ENC defined to != -1) or read through I2C (when BTN_ENC == -1).
//
//#define LCD_I2C_PANELOLU2

//
// Panucatt VIKI LCD with status LEDs,
// integrated click & L/R/U/D buttons, separate encoder inputs.
//
//#define LCD_I2C_VIKI

//
// SSD1306 OLED full graphics generic display
//
#define U8GLIB_SSD1306

//
// CONTROLLER TYPE: Shift register panels
//
// 2 wire Non-latching LCD SR from https://goo.gl/aJJ4sH
// LCD configuration: http://reprap.org/wiki/SAV_3D_LCD
//
//#define SAV_3DLCD

russellchan123, 2016-07-06:
I also tried to enable the buzzer but it doesn't make any noise. If nothing is wrong with my code, perhaps I should try an older version of Marlin or it is a hardware problem?

thinkyhead, 2016-07-06:
Try disabling `U8GLIB_SSD1306` and see if that makes a difference.

Also, use the latest `RCBugFix` code. We might already have fixed something affecting you.

russellchan123, 2016-07-07:
I downloaded and uploaded the RCBugFix code and get the same thing. Neither the buzzer or the scroller works but I still get the marlin logo and the home screen. Perhaps I will try an older version of marlin

thinkyhead, 2016-07-07:
@russellchan123 Did you also disable the `U8GLIB_SSD1306` option when testing?

Could you post a photograph of your LCD controller board? (The bare controller without a case.)

russellchan123, 2016-07-08:
[image]

I also disabled  U8GLIB_SSD1306. 
Just to make sure, all I should need to do is add the marlin and u8glib to the Aurduino libraries folder. I then open up Marlin, go under Configuration.h and comment in the code I pasted above. I also just need to type in #include <U8glib.h> to reference the library right? That seems to be all the steps that I have found online and I suppose it is partially working otherwise I wouldn't get a home screen on my LCD.

Blue-Marlin, 2016-07-08:
"ERR: MINTEMP" kills the machine immediately. No input is possible then. Ether connect a thermistor or use a fake thermistor (998 999).

thinkyhead, 2016-07-08:
> I also just need to type in `#include` to reference the library right?

That will be done for you automatically when you select a graphical display.

> I suppose it is partially working

It's definitely getting close if you have a status display. I wonder if there could be timing issues involved. Recently some board-specific delay code was added to `ultralcd_st7920_u8glib_rrd.h`. So you might try adding these settings to your `Configuration.h` and try different delays:
```cpp
// If you want you can define your own set of delays in Configuration.h
//#define ST7920_DELAY_1 DELAY_0_NOP
//#define ST7920_DELAY_2 DELAY_0_NOP
//#define ST7920_DELAY_3 DELAY_0_NOP
```
You can replace `DELAY_0_NOP` with `DELAY_1_NOP`, `DELAY_2_NOP`, etc. See `ultralcd_st7920_u8glib_rrd.h` for the combinations currently used for different boards. @Blue-Marlin might know more about whether these delays will help.

Blue-Marlin, 2016-07-08:
MINTEMP ERROR -> NO CHANCE TO GET INTO THE MENUE. Look at the picture.

thinkyhead, 2016-07-09:
Translation: Your display is working! If it doesn't work _in situ_ then there could be some kind of static interference.

russellchan123, 2016-07-09:
Thank you so much everything works fine. I just had to plug in the thermistor. I just assumed everything would work without it. Again, thank you.

Blue-Marlin, 2016-07-09:
Do we need an kill-screen? A error message in the baseline seems not to alerting enough.

Roxy-3D, 2016-07-09:
> Do we need an kill-screen? A error message in the baseline seems not to alerting enough.

I presume you mean something that takes over the entire display?   That might be a good idea.   Something else I would like to see is a small area of the screen that developers can take over and put up short messages.

Blue-Marlin, 2016-07-09:
How about "the status line". Simply do it.

msamblanet, 2016-07-09:
I just encountered this also - something as simple as changing the text to "Err: MINTEMP - Halted" might be sufficient (if it fits) - I saw the mintemp error but did not realize it halted the board.  This would seem to be a common mistake for those of us taking our boards away from the printer to update firmwares and such.

thinkyhead, 2016-07-10:
Is it worth including the last (failing) temperature value on the screen?

Blue-Marlin, 2016-07-10:
In almost all cases for a MINTEMP error this will fail during boot - no previous value.
The kill-scren can display other kill messages, not all are temperature related.
If you really want - squeeze the temperature into the status line.

russellchan123, 2016-07-11:
So using my LCD screen I can move the axis. However, the axis only goes one way. For example, when I go from 0 to +10mm it moves but if I try to move it back from +10 to 0mm nothing happens . Do any of you know what could be causing this issue? Perhaps it is in the code somewhere?

lavato, 2016-07-12:
This is the same issue I raised months ago #3257.

I think people mostly experience this when they are building a new printer or testing new HW.

What would have helped me at the time is if there was a Warning message next to e.g. a **Board** section in the Configuration.h informing users about:

* the need to connect thermistors or 
* suggest to set a thermistor type to 999 so that they can test their  HW without having to connect anything.

Blue-Marlin, 2016-07-12:
If you'd write "I see a MINTEMP error" we'd immediately know what to do. If you tell us "Can't get into the menu" there are some dozen possible sources for the problem.

Roxy-3D, 2016-07-12:
> I also agree with above Blue-Marlin's comment that the current message to new users is not alerting enough, it wasn't to me at the time.

https://www.youtube.com/watch?v=REvmhBO99I4

thinkyhead, 2016-07-12:
> if there was a Warning message next to e.g. Board section in the Configuration.h

And/or in the LCD Controllers section. This is a good (although non-obvious) use case for those dummy thermistors. Perhaps it even makes sense to set 999 as the default thermistor (only in the base configuration) instead of 1…?

Blue-Marlin, 2016-07-12:
> Perhaps it even makes sense to set 999 as the default thermistor (only in the base configuration) instead of 1…?

Sorry. No. That's not save. If you set an target temperature the heater will heat infinite if the thermo protections are off.

Blue-Marlin, 2016-07-12:
Let's see if the 'kill-screens' will help.

thinkyhead, 2016-07-13:
> if I try to move it back from +10 to 0mm nothing happens

This might occur if you have not homed with `G28` yet, but you have Max endstops enabled, so you are only allowed to move in one direction. This would occur in an older version of Marlin by failing to enable `DISABLE_MAX_ENDSTOPS`. With the newest Marlin it can occur due to misconfiguration of endstops.

jbrazio, 2016-07-18:
Can we close this issue ?

russellchan123, 2016-07-18:
Yes this issue is all solved.

closed 2016-07-18 by jbrazio