MarlinFirmware/Marlin #4220: Bed leveling and dual X carriage

oysteinkrog, 2016-07-06:
Hi,

I'm having some trouble with Z offsets on my new dual X carriage prototype printer, and I'm wondering if bed leveling (G29) is supposed to work in combination with DUAL_X_CARRIAGE in AUTO_PARK mode?
From reading the source code at:
https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/Marlin_main.cpp#L6364
It seems like the bed leveling offsets are not applied at all when using DUAL_X_CARRIAGE.
This makes sense to me when using DUPLICATION_MODE, but both FULL_CONTROL and AUTO_PARK mode should have the offsets applied I think.

oysteinkrog, 2016-07-06:
Not sure what happened there, must have miss-clicked.

thinkyhead, 2016-07-07:
@Blue-Marlin asked the same question when working on that code…
```cpp
// No extra case for AUTO_BED_LEVELING_FEATURE in DUAL_X_CARRIAGE. Does that mean they don't work together?
```
I believe more work needs to be done to properly support ABL with `DUAL_X_CARRIAGE`. Without any XYZ correction in the tool-change command results will be unpredictable.

oysteinkrog, 2016-07-10:
How is auto bed leveling normally applied?
I'm not sure if dual carriage in auto park mode should really need any special consideration, as when you change the tool the hotend offset in X/Y does not actually change, so the Z value compensation should be the same?

thinkyhead, 2016-07-10:
@oysteinkrog Well, when you have ABL enabled, any override of the `current_position` needs to account for the difference in Z at the new point, even though the next thing that may occur is a move of the new tool to the same point the other tool was at. Since we added the `S1` option to the tool-change command, it's permitted to leave the newly-selected tool at its current position on the tool-change. So we need to have the code that corrects the position in there anyway.

oysteinkrog, 2016-07-12:
So there needs to be several special cases, but do you agree that the AUTO_PARK case of tool-switch should actually be OK as it is right now. This case currently just moves the new tool to the current position, but it does also apply the tool offset. The current position does not change though I think so it should be ok?

thinkyhead, 2016-07-12:
Actually, I don't know what the behavior of `AUTO_PARK` is supposed to be (as this is not an area of the system where I have yet focused) and I've never really seen a dual x-carriage machine up close. I will have to look at some videos that illustrate this feature in action to understand where you're coming from. In any instance where there's more than one nozzle, we always have to account for bed leveling. Without knowing what is "special" about `AUTO_PARK` mode, it seems to be this should be the case in that mode as much as any other.

oysteinkrog, 2016-07-13:
Auto park is special because the firmware handles "parking" the tool when switching to a new one.
E.g. If T0 is at X100 Y100 with tool offsets set correctly, T1 will park the first carriage (at e.g. X-20) and then move the second carriage from park position to the exact same physical position as the first carriage was before switching tools. Since there is no change in physical position and also no change in firmware position it seems to me that this specific case of dual X carrriage might be OK as it is, though the other modes are likely broken.

thinkyhead, 2016-07-14:
> Since there is no change in physical position

It's no different from the other cases. First we establish that the nozzle is at some new position (parked, or whatever). We must do bed leveling correction then and there to establish this position. It doesn't matter that the nozzle eventually ends up at the same position. Before we can even do the move (while bed leveling correction is enabled, mind you) we must perform the necessary correction to get the "true" position, otherwise the "move to the same location" won't end up at the same location. This is especially true for nozzles that have a different Z offset, because with some forms of leveling correction, a difference in Z also shifts the XY.

jbrazio, 2016-08-01:
@oysteinkrog can we close this ticket ?

oysteinkrog, 2016-08-01:
I am not sure, unfortunately my printer is currently in-between upgrades,
so I can't test.
Have there been changes to this code?

thinkyhead, 2016-08-03:
I haven't done any adjustments to bed leveling plus AUTO_PARK mode, nor examined the code very closely. So this is one area that still needs more attention.

nicksears, 2016-08-09:
Do you know what RC you're using? I was using RC4 and had issues because I had to hack some things up. I need auto bed leveling with multiple extruders, but I change the z position of extruders often, so I try and get it close enough and probe to perfect it within ~10um with a z-offset.

In RC4 I commented out the if statements that only allow a z-offset with dual carriage, but now in RC7 it looks like it's handled differently. Since SWITCHING_EXTRUDER was added in RC7, I've resigned to use that, but make the steppers act like normal (not a single motor for both extruders with one reversed. I thought we were going to make the z-offset available universally, no matter what setup?

thinkyhead, 2016-08-09:
> I've resigned to use that, but make the steppers act like normal

@oysteinkrog We don't recommend using a hack if there's a real issue needing to be fixed. Surely you don't want to have to hack Marlin from now till the end of time. We'd rather that users report such issues and help us fix them.

nicksears, 2016-08-09:
Yes, sorry for my hastiness, I just need a fix now and it actually helped me understand all the components of the code and how they work together. I think the fix is simple and mentioned in your comment in #2649, define a z offset and make it zero by default. then we need to get rid of the conditionals for M218 in marlin_main. Finally, we need an else statement to apply the z offset if it is non-zero. Trying that now, I'm sure there are things I'm no thinking of.

nicksears, 2016-08-10:
Ok, I've finished making a solid solution. Basically, add a z offset commented by default and conditional for the toolchange to catch when there's a z offset, but not dual carriage or switching extruders.

I've made changes to an RC version in Github desktop, but I'll admit I'm brand new to this aspect, so I'm not sure if I know what I'm doing.

as far as the actual changes:
config.h, add:
  #ifndef HOTEND_OFFSET_Z
    #define HOTEND_OFFSET_Z {0.00, 0.00}  // (in mm) for each extruder, offset of the hotend on the Z axis, commented by default, uncomment to enable z offset for special cases other than DUAL_X_CARRIAGE or SWITCHING_EXTRUDER
  #endif

in configuration_adv.h make new conditional after "#endif //DUAL_X_CARRIAGE":
#if defined(HOTEND_OFFSET_Z)// works for all cases where hotend_offset_z is defined DUAL_X_CARRIAGE, SWITCHING_EXTRUDER, or if it is manually defined
  #define TOOLCHANGE_ZLIFT 0.5 // the distance to raise Z axis when switching extruder with different height
#endif

I think we should unify the TOOLCHANGE_PARK_ZLIFT and z_raise used in the SWITCHING EXTRUDER to a more generic TOOLCHANGE_ZLIFT, but I don't know what all changes will be necessary to make that hapen. I've defined a new TOOLCHANGE_ZLIFT for now.

Then finally, in marlin_main.cpp, in the M218 section, change two conditionals:
#if ENABLED(DUAL_X_CARRIAGE) || ENABLED(SWITCHING_EXTRUDER)
to
#if defined(HOTEND_OFFSET_Z)

This should enable M218 for any configuration that needs it. We can either manually define it, or it is automatically defined for SWITCHING_EXTRUDER. I think we should also add it by default to the DUAL_X_CARRIAGE, but if it's not defined somewhere already...I'm not sure how that works.

The final addition in marlin_main.cpp is the main change. After the SWITCHING_EXTRUDER code in the tool_change method, before the #endif, we insert:

          #else  // !DUAL_X_CARRIAGE && !SWITCHING_EXTRUDER

          #if defined(HOTEND_OFFSET_Z)
            float x_diff = hotend_offset[X_AXIS][active_extruder] - hotend_offset[X_AXIS][tmp_extruder],
            y_diff = hotend_offset[Y_AXIS][active_extruder] - hotend_offset[Y_AXIS][tmp_extruder],
            z_diff = hotend_offset[Z_AXIS][active_extruder] - hotend_offset[Z_AXIS][tmp_extruder],
            z_raise = TOOLCHANGE_ZLIFT + (z_diff > 0.0 ? z_diff : 0.0);
            
            #if ENABLED(DEBUG_LEVELING_FEATURE)
              if (DEBUGGING(LEVELING)) {
                SERIAL_ECHOPAIR("Raise to ", current_position[Z_AXIS] + z_raise); SERIAL_EOL;
                SERIAL_ECHOPAIR("MoveX to ", current_position[X_AXIS] + x_diff); SERIAL_EOL;
                SERIAL_ECHOPAIR("MoveY to ", current_position[Y_AXIS] + y_diff); SERIAL_EOL;
                SERIAL_ECHOPAIR("Lower to ", current_position[Z_AXIS] + z_diff); SERIAL_EOL;
              }
            #endif

            // Switch to head offset in z: 1) raise 2) offset XY for new extruder 3) lower to correct z
            for (uint8_t i = 0; i < 3; i++)
              planner.buffer_line(
                current_position[X_AXIS] + (i == 0 ? 0 : x_diff),
                current_position[Y_AXIS] + (i == 0 ? 0 : y_diff),
                current_position[Z_AXIS] + (i == 2 ? z_diff : z_raise),
                current_position[E_AXIS],
                planner.max_feedrate_mm_s[i == 1 ? X_AXIS : Z_AXIS],
                active_extruder
              );
            stepper.synchronize();
          
          active_extruder = tmp_extruder;

          #if ENABLED(DEBUG_LEVELING_FEATURE)
            if (DEBUGGING(LEVELING)) DEBUG_POS("New Extruder", current_position);
          #endif
        #endif
		
And then there's an #endif that was already following that...

boelle, 2016-08-11:
So is this a bug? and if so is it still present in RC7?

thinkyhead, 2016-08-12:
@nicksears Git is fun. Are you using Linux, macOS, or otherwise have a *nix-style command-line on your box? Command-line git can help with setting up the proper environment and submitting pull requests.

Anyway, I will make a new branch from `RCBugFix` and create a PR with your (general) changes. I will go about it thusly…
```bash
git fetch upstream
git checkout RCBugFix
git merge upstream/RCBugFix
git checkout -b rc_nicksears_dualx
```
…make my changes, commit them, and then…
```bash
git add *
git commit -m "Allow bed leveling with more Dual X modes"
git push --set-upstream origin rc_nicksears_dualx
```
…ready to make a pull request!

boelle, 2016-08-14:
any news in this one?

nicksears, 2016-08-15:
I'm hving issues with G29 and the height being some random number ~1mm too high after G29 ala issues: 4577, 4593, 4612, etc. I swear I had it working at one point recently, but I'm tearing my hair out trying to figure out what I've changed. I've reverted all of my changes, but until I figure out what I did I'd hate to do more harm than good...

nicksears, 2016-08-15:
Ok I've gone back to RC4  to get bed leveling to work. I really wish we could keep the last few RCs around to try and determine exactly when/what changes might have broken things. I had to change a bunch of syntax to work back on RC4, but it looks exactly the same, but with the correct Z after g29. I just don't get where G29 got broken, does anyone have it working on RC7? (not mesh).

Does it have something to do with the application of auto bed leveling on line 6819 where it no longer has a z component in RC7? (compare to line ~5890 in RC4).

thinkyhead, 2016-08-19:
> I really wish we could keep the last few RCs around

https://github.com/MarlinFirmware/Marlin/blob/1.1.0-RC6/Marlin/Marlin_main.cpp
https://github.com/MarlinFirmware/Marlin/blob/1.1.0-RC5/Marlin/Marlin_main.cpp
https://github.com/MarlinFirmware/Marlin/blob/1.1.0-RC4/Marlin/Marlin_main.cpp
https://github.com/MarlinFirmware/Marlin/blob/1.1.0-RC3/Marlin/Marlin_main.cpp
https://github.com/MarlinFirmware/Marlin/blob/1.1.0-RC2/Marlin/Marlin_main.cpp

thinkyhead, 2016-08-19:
@nicksears After the above PR is merged give RCBugFix another test. It makes some adjustments to leveling that could possibly have some effect on this issue.

boelle, 2016-08-24:
@nicksears had time to retest this one?

boelle, 2016-08-26:
instead of paper use steel feeler blades

these stays at the same thickness and can be used over and over

paper will change thickness as soon you use it. 

i use the 0.1mm feeler blade and if i want it to be more precise i use the 0.05mm blade

boelle, 2016-08-28:
so i guess we can close thi one as @thinkyhead  branch solves the problem

boelle, 2016-09-03:
@thinkyhead @jbrazio @Roxy-3D what do you think?

boelle, 2016-09-07:
@thinkyhead @jbrazio @Roxy-3D think we can close this one... 14 days since anyone had written anything in here

closed 2016-09-07 by thinkyhead

oysteinkrog, 2017-02-16:
I've finally rebuild my machine for proper dual X carriage (not a hacked together version).
I'm currently calibrating tool offsets etc, but I seem to still get this problem...
I.e. When setting T1 offset such that both nozzles are at the correct hight, I still get the wrong Z offset on T1 when printing after a G29.

thinkyhead, 2017-02-18:
Let's start a new issue to investigate that.

Ziyayildizz3, 2019-01-22:
Hello there. Is it possible to set z offset for dual x from the printer display? How about we make an addition?

InsanityAutomation, 2019-01-22:
You can - see the idex menu under configuration.

Ziyayildizz3, 2019-01-22:
I want to set the Z offset x2 in "#define HOTEND_OFFSET_Z {0.0, 0.28}" from the printer display. I didn't quite understand what you said.

InsanityAutomation, 2019-01-22:
Look in that menu with a current marlin 2.0 build and you'll see it there.