MarlinFirmware/Marlin #4221: Feedback on BQ Hephestos2

thopiekar, 2016-07-06:
Today I took the example configuration for my printer, built and uploaded the firmware.
But apparently I'm facing the following problems:

* I can't set the nozzle offset. When using menu -> Prepare -> Set home offsets it just says "Err: Too far!" and when I use alternatively menu -> Control -> Motion -> Z Offset to "-1.13" (Which will have then the correct distance to the bed when Z=0.0) it doesn't use it when running some G-Code.
Could you explain in short the difference between both settings?

* Additionally, Marlin uses a different home, then BQ. When I'm not wrong "home" is the position, where the printer waits before it is starting a print until the nozzle is heated up.
The BQ firmware uses the same "home" like in my marlin build, but be correct the "true" home is at
```X4 Y297 Z15```

* Moving the axis isn't save and it is possible to crash on the side, where no endstop is available. If you move to the direction, where endstops are the printer doesn't crash (luckily).

Thanks!

thopiekar, 2016-07-06:
Ah! I had to save my settings into memory to make the Z Offset working!

thopiekar, 2016-07-06:
Starting my first print at the moment. My Hephestos2 is quieter than before!

Blue-Marlin, 2016-07-06:
Homing is not a goto, it's a process where the printer trys to find out where the nozzle is. The end-point is undefined - or better saied, depends on a dozen of configurable settings. 

>Moving the axis isn't save and it is possible to crash on the side, where no endstop is available. If you move to the direction, where endstops are the printer doesn't crash (luckily).
That is what the software endstops are for.

thopiekar, 2016-07-06:
OK. The default behaviour of the stock firmware was to home before being able to move the axis.

thopiekar, 2016-07-06:
Rechecked and it is still possible to crash after running "Auto home".
The maximum is at X=208mm.

jbrazio, 2016-07-06:
I am the author of the config and I own one bqh2, congrats for jumping out of the fence ! :-)

The home position is actually the same as the stock bq firmware, what bq does is move the print head into a "rest position" when idle which we don't do on Marlin currently.

When you say you can crash the head. is it in the X axis ? I believe I have software endstops preventing crashes on both axis, only the Y I let the nozzle go outside the bed a few mm so I can "purge and cut the filament" when straying printing. This was introduced on the latest RCBugfix, which version are you using ?

thopiekar, 2016-07-06:
Great to hear that there is someone, who owns the same printer.

Ok, before jumping into "vanilla" marlin, I tried to replace BQ's start-
and end-gcodes work regular gcodes. That's how I found this. OK, then
didn't know that there is a difference.

When moving the axis in Y it is possible to move the bed to -3.00 where I
think it should stop at 0.00. At least the signal of the physical end stop
should be able to do a emergency stop if the bed is still moving to the
wrong direction.

Just cloned the RC branch today. Was it a bad idea? Is there a version you
can recommend and/or works perfectly?
Yes, it is still crashing the ends of x and z axis. There is no problem
when printing, as Cura cares here about that, but I'm currently helping out
at Ultimaker and it is still possible that a bug gets into the master
branch of CuraEngine and kills my printer 😅
Oh, that "feature is even very useful.

jbrazio, 2016-07-06:
On github clone the RCBugfix branch, I believe the limit issue you report was fixed.

thopiekar, 2016-07-06:
Thanks. Will give this branch a try 😉

thopiekar, 2016-07-07:
Yes, the limit issue is gone :+1: 

How is it possible to change the filament now? Heat up the nozzle and turn the extruder backwards? :sweat_smile:

thopiekar, 2016-07-07:
As there are no problems left, I'll close this now :wink:

thinkyhead, 2016-07-07:
> How is it possible to change the filament now?

Try `M600` after enabling…
```cpp
#define FILAMENT_CHANGE_FEATURE
```

thopiekar, 2016-07-13:
Tried it now and it works! Is there a way to bring this command into the LCD menu?
I see this define in Configuration_adv.h and this filament changing seems to be Ultimaker-only at the moment.

Additionally, to cool down the nozzle BQ uses the second fan to speed up the process. Is it possible to do it here, too?

thopiekar, 2016-07-13:
Finally: Is there a way to keep the motors on?
BQ just lets you move the axes when the printer knows its position for sure.
Moreover, when doing multiple prints you won't need to home the position everytime.

jbrazio, 2016-07-13:
> BQ just lets you move the axes when the printer knows its position for sure.

This is also valid for Marlin.

Have a look in `Configuration_adv.h` at:
```cpp
// Time can be set by M18 and M84.
#define DEFAULT_STEPPER_DEACTIVE_TIME 120
```