MarlinFirmware/Marlin #4230: minimum step/mm value

solo220, 2016-07-06:
I am using a Marlin in my printer with the x,y,z setting equal to 50pulse/mm , but i discovered that the fimware generate pulses in multiple of 10 and in my case a layer with 0.1 mm move which is equal to 5 pulses will not generate any movement in the printer .
Is this because of my acceleration/jerk settings or this is how marlin works ?
thank you.

thinkyhead, 2016-07-06:
> i discovered that the firmware generate pulses in multiple of 10

Where did you hear that? It's not the case.

solo220, 2016-07-07:
**@thinkyhead**
 thank you for your reoly ,
my acceleration settings are low and these are my settings:
```
#define DEFAULT_AXIS_STEPS_PER_UNIT   {52.63,52.63,50,100}  
#define DEFAULT_MAX_FEEDRATE          {300, 300,7, 25}    
#define DEFAULT_MAX_ACCELERATION      {100,100,10,100}  
#define DEFAULT_ACCELERATION          100    
#define DEFAULT_RETRACT_ACCELERATION  50   
#define DEFAULT_TRAVEL_ACCELERATION   200    
#define DEFAULT_XYJERK                10.0    
#define DEFAULT_ZJERK                 1.0    
#define DEFAULT_EJERK                 1.0  

```
the issue is that all my printer motors move normaly but very small details that appear in the gcode file simulation disappear in the print so in order to diagnose the problem i moved my motors using steps of 0.1 mm and the motors moved but after two successive 0.1 mm steps and not at each 0.1 mm step.
then i doubled the step/mm in the settings and the motors started to move at each 0.1mm but not at each 0.01mm step.

epatel, 2016-07-07:
@solo220 I wonder if this an effect of too little current to the steppers and the micro stepping? Did it work in a previous version and changed to this, or are you setting up a new printer?

solo220, 2016-07-07:
@epatel 
I will try to change the steppers current and will check the pulses output using an oscilloscope .

Blue-Marlin, 2016-07-07:
Try to lower `dropsegments` in `Configuration_adv.h`

solo220, 2016-07-07:
@Blue-Marlin 
I think this is where my problem  , thank you
i will test and reply .

solo220, 2016-07-07:
Isuue solved thank you every one.

closed 2016-07-07 by thinkyhead

rszasz, 2017-11-30:
This seems to be  MIN_STEPS_PER_SEGMENT  now

thinkyhead, 2017-12-01:
And it's required.
```cpp
#elif !defined(MIN_STEPS_PER_SEGMENT)
  #error Please replace "const int dropsegments" with "#define MIN_STEPS_PER_SEGMENT"
 (and increase by 1) in Configuration_adv.h.
```