MarlinFirmware/Marlin #4237: Marlin 1.1.0-RC7

thinkyhead, 2016-07-08:
This is a placeholder for the Change Log for 1.1.0-RC7 where we can itemize all changes since RC6.

Here's a convenient [link to all PRs since RC6](https://github.com/MarlinFirmware/Marlin/pulls?q=is%3Apr+is%3Amerged+merged%3A%3E2016-04-28&utf8=%E2%9C%93
)…

Roxy-3D, 2016-07-08:
I presume the RC tag will point at this very soon?   And RCBugFix will initially point at the same thing the RC tag points at?

thinkyhead, 2016-07-08:
Yep. But in one or two (or three) days. There are a couple of annoying bugs still hanging around.

petrzjunior, 2016-07-08:
What about the `Stop print` bug? Is anyone going to fix it? Schould be done before release.

Roxy-3D, 2016-07-08:
> What about the Stop print bug? 

Almost for sure...  ThinkyHead is looking at that.

> Is anyone going to fix it? 

Almost for sure, somebody is going to get it fixed.   There is a better than 50%-50% chance it will be ThinkyHead that finds and fixes it.  But there are other people that fix SD Memory card issues so one of those people may find it before he does.

> Should be done before release.

Agreed.  But we are only talking about about starting a new Release Candidate and not actually promoting a Release Candidate to 'Golden' status.

Blue-Marlin, 2016-07-08:
It won't be bad if you'd follow what is going on here.
The loop after stop when printing from SD issue is fixed.
The shift position after stop problem is minimized for Cartesian printers as far physics allows. (You can't simply stop a full speed move without losing steps) 
The forward calculation for DELTAs is ready but not applied in quick stop now.

The question nobody raised up to now is: "Does it make sense to continue after a emergency stop at all. Wouldn't it be better to use 'pause' if you intend to continue."

thinkyhead, 2016-07-08:
@petrzjunior The "stop print" bug was fixed a couple of days ago.

jstefanop, 2016-07-08:
So RC7 wont be the final 1.1 release?

thinkyhead, 2016-07-08:
> So RC7 wont be the final 1.1 release?

@jstefanop That will depend on how many bugs are found and how serious they are. If it appears very stable, we may release it as-it-is. If it only has some minor things, we may patch it up and make that 1.1.0. But if it has any serious or complex issues, we would consider doing another interim release to work out those issues. This release, as all releases, will generate an upsurge in issue reports, and we will need to sort out which are configuration issues versus real bugs.

Personally, I expect no serious issues. But we haven't tested on every piece of hardware. To help with that, in a couple of weeks I'm hosting a Marlin test lab with the ["PDX 3D Printing Lab" meetup group here in Portland](http://www.meetup.com/Portland-3D-Printing-Lab/events/220173407/). We're getting together about 20-25 users and their printers for a one-day test lab where we will go through configuration, installation, calibration, and testing. I will be fixing-up any extant issues as we find them. More precise details on the lab will be available soon, as we work them out.

oxivanisher, 2016-07-13:
I like that. Will surely help test the RC7 on my delta. :+1:

jackshi618, 2016-07-15:
When publishing RC7?

Blue-Marlin, 2016-07-15:
When ready! Still to much to do.

jackshi618, 2016-07-15:
Waiting for a long time.

adamfilip, 2016-07-15:
Feel free to help out to make things faster..

jbrazio, 2016-07-15:
I have to take some people as jokers otherwise the swearing would be bad from my side. :-)

thinkyhead, 2016-07-17:
Sorry for the long wait @jackshi618 — I also keep expecting the release to be ready "any minute now" and then we discover some subtle bug like #4310 that is challenging to solve, or some minor code cleanup (in this instance with redundant bed-leveling functions) turns into a bigger piece of work requiring additional regression testing.

Of course, when investigating bugs we have to wait for feedback from testers, and sometimes this back-and-forth takes days. During those waiting periods we tend to work on code cleanup, trying to reduce the build size, writing documentation, discussing the roadmap, polishing the configuration files, and trying to address longstanding weak-points.

The good news is that we have only a few un-fixed issues remaining, with testers and active debugging ongoing. As soon as we deal with these last remaining items and all our testers give us a green light, RC7 will be unleashed on the world. It should be any day now…!

Roxy-3D, 2016-07-17:
> Sorry for the long wait @jackshi618 — I also keep expecting the release to be ready "any minute now" and then we discover some subtle bug like #4310 that is challenging to solve, or some minor code [...]

I have a lot of these issues fixed in my new code.    This will all be fixed (all ready) in the Development branch that gets released at the same time as you go Golden!     Let's freeze the probing and auto bed leveling parts of the Release Candidate.

thinkyhead, 2016-07-17:
> Let's freeze the probing and auto bed leveling parts of the Release Candidate.

Let's fix the bugs. Then freeze.

Roxy-3D, 2016-07-17:
Well...   The thing is very few people are seeing problems in the bed leveling and probing right now.   And those few that do have some quirks can probably get by just fine with the new bed leveling system.     They can help test it too!

thinkyhead, 2016-07-17:
> very few people are seeing problems

Very few people are testing.

@Roxy-3D Are you planning to include your new bed leveling system in RC7 then?

Roxy-3D, 2016-07-17:
I really feel that would be a mistake.   It is far too invasive.   I've been trying very hard to do everything clean and correct.   But you know how even 10 or 15 lines of new code can cause problems.   This is literally 1000 lines of new code when you add in the Mesh Validation Tool.   (Don't worry about code size.  Much of it can be turned off.  And all of the original G29 code (all 3 flavors) is gone.  Plus, I'm going to offer to ability to do the iterative Least Squares Fit instead of the big QR_SOLVE which saves 10Kb.)

But with that said...     I just finished my first print with it 30 minutes ago.   Everything seems to be working correctly.   But I'm not done with the Mesh Validation Tool, and once that is working how I want I was going to use it's results to finish up the Mesh Editor.   (I had to stop working on the Mesh Editor because I didn't have the information I needed to fine tune the Mesh.)    Both of those things are straight forward now that the system is working.   And even better, both of those things don't need to hook into the main body of Marlin so it is much faster to code.

I was thinking it made sense to release at the same time you promote the RC to 'Golden'.   But if it is going to take 4 or 6 weeks to get to that point, how bad would it be to put bug fixes and code clean up into both branches?  I guess what I'm asking is would it be a bad thing to release the Development Branch ahead of the Golden Release ?

thinkyhead, 2016-07-17:
@Roxy-3D I suggest we release 1.1.0-RC7 with the current cleaned-up (soon bug-free) leveling and homing code. Then when it has been certified as working by the community, we release RC7 as our final 1.1.0. (With no pre-release "golden" anything.) The new bed leveling code would be placed in another branch at that time, and we would include it in Marlin 1.1.1-RC1.

jbrazio, 2016-07-17:
We get out of one RC cycle to enter another one ?.. I frown a bit at that Scott.
(I know you spoken about a branch.. but a branch with the master one having heavy development would mean a mess to merge back them together aka the MarlinDev syndrome). When we enter 1.1.0 and start developing 1.1.1 we should be out of RC mode until we feel we want to release 1.1.1, only then we enter RC mode and freeze.

For 1.1.0 I agree with @Roxy-3D no UBL, this would be our main feature for 1.1.1.

thinkyhead, 2016-07-17:
> I frown a bit at that Scott.

Sorry, it would be my understanding that before we release things, we first make sure they get some beta testing. Should we conceal that and just do it "in-house"?

Roxy-3D, 2016-07-17:
> Should we conceal that and just do it "in-house"?

We don't have a full test suite of machine types available.  We need users to help verify correct operation.   

What most companies do as part of the release process is they stop making any changes that don't add to 'stability'.   A lot of time, it is a judgment call.  (For example, a very minor bug that can be worked around and is very difficult to fix would get pushed out to be addressed in a Development cycle)    A super serious bug that only requires a + sign being changed to a - sign would probably get the OK to be changed.  (I realize you both know this.  I'm just trying to communicate my thinking.)

And that is 1/2 of my thinking on why we should 'try' to freeze the Bed Leveling and Probing sections of the code.    Those areas of the code work.   There are some minor issues but even so, there are usually good work arounds.  Changes in those areas are like opening a can of worms.   You open the lid, and pretty soon you have worms crawling all over the table.    (The other 1/2 of my thinking is I can't keep up with all of the changes.  First I have to figure out what you did, and then I have to change everything I did to match it.   And then debug all of those changes in my code.)

Can we at least agree to  'try' to avoid making serious changes to the Auto Bed Leveling and Probing code?   I think that would help us stabilize things.   And it would certainly make it easier for my to get my code finished.

thinkyhead, 2016-07-17:
> we should 'try' to freeze the Bed Leveling and Probing sections of the code. Those areas of the code work.

From my point of view both homing and leveling are currently broken. Delta issues where it only probes one corner. `G28` homing losing track of the current position. . . . These need to be fixed before unleashing this thing.

thinkyhead, 2016-07-17:
> I can't keep up with all of the changes.

This is what Richard ran into by trying to get too far ahead of everyday bug-fixing and cleanup. It's great if you have standalone code that you wrote yourself, which eschews all the old cruft. But if you rely on hooking into the existing broken functions and methodologies, copying or building upon what already exists, then you're bound to get thrown by the bug fixes and cleanup. The amount of reduction and simplification that has been made is pretty impressive, getting down to the essentials. After https://github.com/AnHardt/Marlin/pull/58 there shouldn't be too much more, but it would be good to do a comprehensive review and make sure everything is still holding together, and see what more might be got rid of at this point.

Roxy-3D, 2016-07-17:
> From my point of view both homing and leveling are currently broken. Delta issues where it only probes one corner. G28 homing losing track of the current position. . . . These need to be fixed before unleashing this thing.

Ooops!  I forgot about that.   You are correct that needs to be fixed!

jbrazio, 2016-07-17:
I can see Roxy's standpoint, specially when she is playing behind the scenes getting something ready to be presented and we keep changing code that will be replaced in the end.. But until we have UBL on the wild.. MBL and ABL will need to be patched a brought down to some standard degree of "cleanup".

I personally would advise to branch out UBL and Roxxane to publish the code she is working on, I bet some people would love to weight in in a **constructive way**. This not only would be good for Marlin dev as we could be reviewing and suggesting possible improvements sooner but it would also be good for her as she would finally learn how to use git from the cmd line. :-P

GreatGrizzly, 2016-07-20:
This seems like a lot of this is semantics, release candidate this, and golden that. You all have done an incredible job with features and bug fixes, more then enough to be considered a milestone for the project.

Marlin is made for a very small specific market. This market, by its nature, has consumers that are more tech savvy than most people. This means that bugs are less of an impediment to progress because the same consumers can work around them. Bugs will **never** go away, ever, ever, ever.  

Pick a milestone and stick to it. From my 3rd person perspective, you all have exceeded a milestone.

thinkyhead, 2016-07-21:
> Bugs will never go away, ever, ever, ever.

@GreatGrizzly We have known, confirmed, glaring bugs, and we _are_ going to fix them before we do another release. We could use help in testing.

> Pick a milestone and stick to it.

We didn't get hung up on "milestones" for 1.1.0 because we never really set any except "make it not broken." I'm sure we will make use of them more formally in future, because we have a more concrete set of goals.

Roxy-3D, 2016-07-21:
> I personally would advise to branch out UBL and Roxanne to publish the code she is working on, I bet some people would love to weigh in in a constructive way. This not only would be good for Marlin dev as we could be reviewing and suggesting possible improvements sooner but it would also be good for her as she would finally learn how to use git from the cmd line. :-P

@jbrazio @thinkyhead   I finished the Mesh Validation Command today.  That was the one that was killing me.   I now have the information (and a way to verify) the changes made by the Mesh Editor.   I'm guessing I will be ready to create a fork in a week.     Are you both OK with me 'releasing' it prior to us getting all of the bugs fixed in the Release Candidate?

thinkyhead, 2016-07-21:
Release away. We'll do our best to keep it up-to-date with `git rebase` as needed.

Roxy-3D, 2016-07-21:
I don't think I can do what I want to do with the Git Web Interface.   I want to fork the version of RC-6 that existed on 07-04-2016 (on July 4th).   Can you create a branch in the MarlinFirmware/Marlin domain that is derived from that?   I will get all of my stuff put into that branch.   If this can be done, it probably would make sense to give it a reasonable 'Unified Bed Leveling --- Development Branch' name.

(Alternatively, I could fork today's version of RC-6 and just erase any changes that have happened since July 4th, but my guess is the first idea is cleaner)   

I need to finish up the Mesh Editor, but that is not going to be difficult.   All of the support routines are already in place for other reasons.   The only other big thing outstanding is Grid Based Leveling, but I know how to do that.   The messiest part of doing that is I want to make it selectable whether the original QR_SOLVE is used or the 10KB smaller iterative version is used.   I'm committed to getting the Grid Based Leveling going, but the honest truth is, it isn't needed.   When it comes to Auto Bed Leveling, the High Resolution Mesh approach really is the Cat's meow!  @epatel pioneered an important new technology with Mesh Leveling! 

Oh!   And right now Delta's have never been tested.  (My Delta is still apart but getting it going on Delta's shouldn't be too hard once I have a machine to test on.)  It probably makes sense to do a staged release with Cartesian being done first.   They are so much easier to debug.

jbrazio, 2016-07-21:
@Roxy-3D here you go:
https://github.com/MarlinFirmware/Marlin/commits/devel-ubl

The last commit on that branch is:
> commit e48502866b23909eca7d21d8f0d4d1e33402135e
> Merge: d70197f 840e13f
> Author: Scott Lahteine <thinkyhead@users.noreply.github.com>
> Date:   Mon Jul 4 01:19:31 2016 -0700

You should update your personal fork of Marlin and use PR to publish to this new branch, please do not commit directly to it !

Roxy-3D, 2016-07-21:
> You should update your personal fork of Marlin and use PR to publish to this new branch, please do not commit directly to it !

Oh?   That was my plan.  I was going to cut & paste every file into the new branch.   Supposedly, that is the equivalent of doing a PR.   Can we make a one time exception just so I can get things loaded easily?

thinkyhead, 2016-07-21:
After you make a fork and copy it to your local machine, you can use the `git` command line (!) to create the branch you need from any point in time. For example, July 4 at noon. I believe that your local timezone applies.
```
git checkout 'RCBugFix@{2016-07-04 12:00:00}'
```

Roxy-3D, 2016-07-22:
> Can I fork this July 4th branch (to make a personal fork of Marlin) and do direct commits to that without screwing things up? If so, maybe we can merge the two branches back together again?

Will this work to get everything into the system?   Does it screw up anything important (like the audit trails or who knows what) if I do this?

thinkyhead, 2016-07-22:
If you've made a fork on Github you can simply create a new branch from one of the existing branches using the website. Fetch your new branch using the Github Desktop app. Then you can just edit the code and use the Github Desktop app to commit your changes and "sync" the changes up to your fork. At that point I can use your branch as the basis for a new branch here.

thinkyhead, 2016-07-22:
Gosh. Fix that OS so you can use the command line. It is pretty vital. And there's no reason for humble command-line git to have any crazy effect on your system.

thinkyhead, 2016-07-22:
> fork this July 4th branch

You don't fork branches. You fork repositories. The branches in your repository are separate from those in the main repository, and you can modify them all you want without any effect on the source repository.

Roxy-3D, 2016-07-22:
Thank you for the the comments!   I'll re-read them in the morning (twice).   I agree I need to learn how to use Git.   But that can wait a few weeks.  Right now, I'm focused on getting the new development branch ready to launch.

@thinkyhead @jbrazio   You both have Prusa i3 type machines, right?   Do you both have Ultra 20x4 LCD Panels?   I'm hoping the answer to both questions is a "Yes!"

jbrazio, 2016-07-22:
@Roxy-3D I have both LCD, a full graphical one and a 20x4.
Why the question ?

Roxy-3D, 2016-07-22:
I've tried to be clean and correct how I did things.   But first, I'm almost positive the code won't work on a Delta just because they are such a pain in the butt to get to move right.  Almost for sure I've over looked something.   It won't be hard to get the code running on a Delta, but it will have to be done by somebody that has a Delta.   (My Delta is not running at the moment.   If nobody beats me too it, I'll get my Delta put back together and get the code running there next.   Realistically...  That is several weeks from now.)

The same thoughts apply to the 20x4 LCD Panel.   I've tried to cut into the LCD Code in a precise and surgical manner.   But I only have a 20x4 LCD Panel to test against.  Almost for sure I've missed something.   So what I was hoping was you both had Prusa i3 like machines with 20x4 LCD Panels.   If you make the minimum number of changes to the Configuration.h file, such as bed size and homing direction and inversion of stepper directions....   (That kind of stuff)  It is going to work for you.  And you will be able to see it run.     

If you can see it work, then it will be less difficult to know what is supposed to happen on a machine type where it doesn't (quite yet) work.

thinkyhead, 2016-07-22:
> … have Ultra 20x4 LCD Panels?

I do, but currently the graphical display is connected because I was focusing on code for that.

thinkyhead, 2016-07-22:
> need to learn how to use Git. But that can wait

@Roxy-3D Seriously, don't wait. You can't do any of what you want to do without command-line Git. You can't copy a branch from `upstream` without it. You can't even create a reference to `upstream` (this repository) without it. You can't create a branch from a specific date without it. You can't get into the flow and produce decent pull requests without it.

To get this ball rolling, please zip your copy of Marlin and send it to me. I will create a `UBL` branch in this repository (or my fork), which you can later copy to your own fork. Is it still correct that the best starting-point is the last commit from July 4? In other words, that is where your UBL code started from?

I will start with the given date, then drop in your updated files and try to parcel them out into commits by theme. After that I will attempt to rebase these commits on the latest code. With any luck it won't be too difficult. This might take me a couple of hours. I would rather you spent those 2 hours to learn some basic Git, then I would have time to do other things!

- Git tutorial videos (Windows): https://www.youtube.com/watch?v=j1oFazXrzN4
- A great written tutorial: https://www.atlassian.com/git/tutorials/using-branches/git-branch
- A great interactive tutorial: http://learngitbranching.js.org/

Roxy-3D, 2016-07-22:
>> … have Ultra 20x4 LCD Panels?
>I do, but currently the graphical display is connected because I was focusing on code for that.  If you only modify ultralcd.cpp and not the ultralcd_impl_*.h files then you should be fine.

Yes, I've only needed to make a few tweaks in ultralcd.cpp.  Right now I'm fighting with it to get the Mesh Editor fully working.   But my changes are very simple and only in a few places.   So who knows?  Maybe it will work with the full graphics controller?    I guess we will know soon enough.

> @Roxy-3D Seriously, don't wait. You can't do any of what you want to do without command-line Git. You can't copy a branch from upstream without it. You can't even create a reference to upstream [...]
Yes.   I know I'm handicapped without it.   And I do want to learn it and become proficient at it.  But Joao made me the July 4th branch I needed and I've forked that over in my repository.  As soon as I get the Mesh Editor fully running and checked out, I'll put all of my changes there.   At that point, the Grid Based Leveling won't be in the code yet, but I know how to do that.   It shouldn't take long to get that in place.

> To get this ball rolling, please zip your copy of Marlin and send it to me. I will create a UBL branch in this repository (or my fork), which you can later copy to your own fork. Is it still correct [...]

If you want, I can do that.  But I think I have everything I need to get my changes into a fork of the July 4th branch.   I was tracking your changes pretty closely, but at July 4th, I started just focusing on my code.   So, currently there are some changes you have made since July 4th that I do not have.

> I will start with the given date, then drop in your updated files and try to parcel them out into commits by theme. After that I will attempt to rebase these commits on the latest code. With any luck it won't be too difficult. This might take me a couple of hours. I would rather you spent those hours to learn some basic Git, then I would have time to do other things!

The problem is the changes are all over the place.   Without all of the changes, nothing is going to work.   So the question I have is:   Does it make sense to break the package apart into smaller commits ?    

I'll try to watch those videos as soon as I get the Mesh Editor running right.

thinkyhead, 2016-07-23:
> Does it make sense to break the package apart

My approach —when the code has diverged too much to do a simple rebase— is to go over the changes by hand using the Github diff view (on the site or in the app) and re-apply the changes where they make sense in the newest code. That way if new methodologies have been developed, the code will be brought up to date. I try to do them in some logical order and create new commits that feature each type of change.

Roxy-3D, 2016-07-23:
OK....    I don't know enough to argue if I wanted to (and I dont!!!).    But from how I see the world, none of these changes have any value unless the whole set of changes is present.  So, it is kind of a 'Take it or leave it' thing.    

But I'll get the whole set into that branch and transfer ownership so you can decide the 'right' way to get that branch started.

thinkyhead, 2016-07-23:
> it is kind of a 'Take it or leave it' thing

I look forward to taking it all in.

Roxy-3D, 2016-07-24:
I crossed all of the UBL code over to a branch of  devel-ubl, but it looks like all of that effort was for nothing.  devel-ubl branches off of RC instead of RCBugFix.   Aaaaurrggghhhh !

@thinkyhead @jbrazio 

I've emailed you both a .ZIP file of the UBL code working with RCBugFix as of July 4th, 2016.   Can you take a look and make a recommendation on how how we should proceed?   I'm hoping you can just get  the few changes that have happened in the last 10 days into it, and get a suitable development branch setup.    The file extension needs to be changed from .ABC back to .ZIP   (Google won't let .ZIP files be sent)

thinkyhead, 2016-07-24:
If it started from July 4, it should be easy to extract the changes and make a branch here. I will try the following:
```
git remote add roxy git@github.com:Roxy-3D/Marlin.git
git fetch roxy
git checkout 'RCBugFix@{2016-07-04 12:00:00}' -b july_four_plus_ubl
git diff july_four_plus_ubl roxy/devel-ubl | git apply
git push --set-upstream upstream july_four_plus_ubl
```

Roxy-3D, 2016-07-24:
No...    The problem is, you cant grab the one I did on github.    I forked the devel-ubl branch into my account and did the work to get my changes into it and working.    Only later I found out the devel-ubl branch was started from RC-6 (as it existed on July 4th).    It did not start from RCBugFix as it existed on July 4th.

(And in fact...   looking at the files in july_four_plus_ubl there are no G25.cpp, G26.cpp or G29.cpp files.  (Those files are definitely in the .ZIP file.)

Can you take another look at things with this extra information available to you?   One option is for me to fork the current RCBugFix and try to work as fast as possible to get my stuff moved to it.   That will probably take me several days of effort.    I don't want to do anything without clear and concise guidance that it won't be wasted effort.

That is why I emailed the .ZIP file to you.

thinkyhead, 2016-07-24:
@Roxy-3D I dropped your files into the [july_four_plus_ubl](https://github.com/MarlinFirmware/Marlin/tree/july_four_plus_ubl) branch, went through and turned the differences into commits. You can see the 4 commits I created for it [here](https://github.com/MarlinFirmware/Marlin/commits/july_four_plus_ubl).

Roxy-3D, 2016-07-24:
OK THANKS!!!!!!     I'm very confused because I looked at the files in july_four_plus_ubl and my new files were missing when I wrote that message.   Did I catch a view of it while things were in process?

Anyway...   Just looking at the files and the commits, that looks viable!    I'm guessing those extra branches can go away now?    And I need confirmation:   Should I fork the devel-ubl branch to continue my work?   As soon as I hear back...   I'll fork and verify everything is still working.   And then I'll be in a position to do Pull Requests to get stuff back to what ever branch you tell me to fork.

thinkyhead, 2016-07-24:
> Should I fork the devel-ubl branch…?

You can do that now, if you wish. I've finished patching up those commits. After you've got it working the way you like, later on we can isolate the changes and then apply the more up-to-date equivalent changes to the latest RCBugFix code.

Roxy-3D, 2016-07-24:
Thanks Again!    Can you get the Readme file close to what I have?   The contents I have in it are aimed at helping people understand what the current status of UBL is and what they should expect.

I'll re-fork devel-ubl and continue forward!

thinkyhead, 2016-07-24:
> Can you get the Readme file close to what I have?

The README.md in your ZIP file is no different from the README.md file from July 5. I'm sure you can do the update, as needed.

Roxy-3D, 2016-07-24:
OK.  I brute forced the Readme to be correct.   But the fish logo is missing because the Documentation directory is not there.    I don't dare to do that.

jbrazio, 2016-07-24:
I fixed the image for you @Roxy-3D

Roxy-3D, 2016-07-24:
Thank You!   We can't do anything without the Marlin being there!   :)

[13 more replies]

closed 2016-12-11 by thinkyhead