MarlinFirmware/Marlin #4238: Mash bed leveling probe does not position correctly

lavato, 2016-07-08:
I have a following scenario:

1) **HW**
HotEnd 1 @ (-30), HotEnd 2  @ (0), PRBOBE  @ (30), 

2) **Config**
```
#define FIX_MOUNTED_PROBE
#define X_PROBE_OFFSET_FROM_EXTRUDER 30
#define Y_PROBE_OFFSET_FROM_EXTRUDER 0
#define Z_PROBE_OFFSET_FROM_EXTRUDER -1.82

#define MESH_BED_LEVELING  
#if ENABLED(MESH_BED_LEVELING)
  #define MESH_INSET 0
  #define MESH_NUM_X_POINTS 3
  #define MESH_NUM_Y_POINTS 3
  #define MESH_HOME_SEARCH_Z 4

  #define MANUAL_BED_LEVELING  
  #if ENABLED(MANUAL_BED_LEVELING)
    #define MBL_Z_STEP 0.025
  #endif  

#endif 
```

3) **Issue**
When doing manual Bed Level, sampling is moved by 30mm across the X axes as it appear it uses the position of the HotEnd1 as a probe location,  despite `X_PROBE_OFFSET_FROM_EXTRUDER`    being set to 30.
As a result the last sampling point is outside the bed area.

Blue-Marlin, 2016-07-08:
Neither MESH nor MANUAL levelling can handle can handle a probe!. Don't define one. One of your nozzles is the probe you use for the paper-test.

lavato, 2016-07-08:
:( ok.

Are there plans for this feature to support a probe?

Roxy-3D, 2016-07-08:
> Are there plans for this feature to support a probe?
Yes.  And much more.

> Can this be marked as a feature request please?
There is no need for a feature request.  The code is almost done.

lavato, 2016-07-08:
Great news for most I would say.

Is this planned for RC6/7 or a later releases?

closed 2016-07-08 by lavato

Roxy-3D, 2016-07-08:
Right now it is planned for when RC-7 goes final and turns Golden.