MarlinFirmware/Marlin #4247: HotEnd temperature gets set only after G28 and/or G29 is completed

lavato, 2016-07-09:
The above was different to all previous releases.

This extends the overall time.

Blue-Marlin, 2016-07-09:
Rubbish!
Don't generalize. Talk about your special case.

lavato, 2016-07-09:
@Blue-Marlin 
I think you should be more temperated in a way you respond to people. 

This is a high profile project and we are all trying to help to get Marlin work better, so if you expect people to respond to you politely, you should try to do so yourself. 

**As requested, here are some more details:**

In this release, RCBugFix (c23dd7e), HotEnd temperature is set after G29 completes its cycle, in RC6 for example, hot end temperature was set just before G29, so heater starts heating during the G29 cycle.

**My Slic3r start code**:

G28 ; home all axes
G29 ;  bed level
G1 Z5 F5000 ; lift nozzle

Blue-Marlin, 2016-07-09:
How about a complete sequence where you can see what is going on?

```
M105
M104 S200
M105
G28
M105
G29
M105
M109 S200
M105
```
Where is the problem?
```
21:17:56.034 : N57 M105*21
21:17:56.053 : ok T:174.1 /0.0 B:38.0 /0.0 B@:0 @:0
21:17:58.689 : N58 M105*26
21:17:58.706 : ok T:173.1 /0.0 B:38.0 /0.0 B@:0 @:0
21:17:58.707 : N59 M104 S200*91
21:17:58.711 : ok N59 P15 B3
21:17:58.711 : N60 M105*17
21:17:58.728 : ok T:173.1 /200.0 B:38.0 /0.0 B@:0 @:0
21:17:58.729 : N61 G28*36
21:18:13.186 : X:100.00 Y:50.00 Z:15.60 E:0.00 Count X: 9412 Y:4706 Z:62400
21:18:13.186 : ok N61 P15 B3
21:18:13.186 : N62 M105*19
21:18:13.202 : ok T:175.7 /200.0 B:37.7 /0.0 B@:0 @:127
21:18:13.202 : N63 M105*18
21:18:13.217 : ok T:175.7 /200.0 B:37.7 /0.0 B@:0 @:127
21:18:13.217 : N64 G29*32
21:18:13.222 : G29 Auto Bed Leveling
21:19:04.169 : Eqn coefficients: a: -0.00620185 b: -0.00018357 d: 1.22371411
21:19:04.174 : Bed Level Correction Matrix:
21:19:04.174 : +0.999981 +0.000000 -0.006202
21:19:04.174 : -0.000001 +1.000000 -0.000184
21:19:04.178 : +0.006202 +0.000184 +0.999981
21:19:04.178 : X:189.90 Y:138.99 Z:16.79 E:0.00 Count X: 17882 Y:13082 Z:62363
21:19:04.181 : ok N64 P15 B3
21:19:04.181 : N65 M105*20
21:19:04.198 : ok T:199.1 /200.0 B:36.9 /0.0 B@:0 @:35
21:19:04.198 : N66 M105*23
21:19:04.213 : ok T:199.1 /200.0 B:36.9 /0.0 B@:0 @:35
21:19:04.213 : N67 M109 S200*91
21:19:04.230 : T:199.1 /200.0 B:36.9 /0.0 B@:0 @:35 W:?
21:19:05.229 : T:199.1 /200.0 B:36.9 /0.0 B@:0 @:37 W:9
21:19:06.229 : T:199.3 /200.0 B:36.9 /0.0 B@:0 @:33 W:8
21:19:07.228 : T:199.3 /200.0 B:36.8 /0.0 B@:0 @:35 W:6
21:19:08.231 : T:199.3 /200.0 B:36.8 /0.0 B@:0 @:35 W:5
21:19:09.231 : T:199.1 /200.0 B:36.8 /0.0 B@:0 @:41 W:4
21:19:10.231 : T:199.2 /200.0 B:36.8 /0.0 B@:0 @:38 W:3
21:19:11.230 : T:199.3 /200.0 B:36.8 /0.0 B@:0 @:37 W:2
21:19:12.229 : T:199.2 /200.0 B:36.8 /0.0 B@:0 @:40 W:1
21:19:13.229 : T:199.4 /200.0 B:36.8 /0.0 B@:0 @:36 W:0
21:19:14.212 : ok N67 P15 B3
21:19:14.212 : N68 M105*25
21:19:14.228 : ok T:199.4 /200.0 B:36.8 /0.0 B@:0 @:37
```

Where do you see the heater is not started?
Did you change your start script?
There is no temperature output during G28, G29. But that has not changed.

lavato, 2016-07-09:
OK, I spent more time on this today.

Yes, the issue is actually with Slic3r, which had M109 in it's **start-code**, which was not visible to me as it was pushed down in a non visible part of the text box.

Slic3r does not generate ANY M109 codes at the start if it finds temperature related gcodes in its start script, therefore  to me, the behavior looked as if the Marlin has changed (as I updated yeterday), since normally M109 codes are generated first, and therefore get executed just before G28/G29. 

This same issue with Slic3r also caused my second extruder not to print, as it never generated the M109 codes for the second extruder. This should however be detected by Marlin as it does the air print.

closed 2016-07-09 by lavato

thinkyhead, 2016-07-10:
> This should however be detected by Marlin as it does the air print.

Marlin outputs a lot of messages, but continues to "air print" when you try to extrude below temperature. This is historic behavior, and hasn't been altered yet.

jbrazio, 2016-07-11:
And in fact "air-printing" this is sometimes useful as a debug feature.