MarlinFirmware/Marlin #4248: At times buzzer sounds continues to Beep

lavato, 2016-07-09:
At times buzzer sounds continues to Beep, the only way to stop it is to press a button.

thinkyhead, 2016-07-10:
I suggest this change:
```diff
-      else if (millis() >= this->state.timestamp) this->reset();
+      else if (ELAPSED(millis(), this->state.timestamp)) this->reset();
```

thinkyhead, 2016-07-10:
I note that `lcd_buzz` is still defined, but never used.

thinkyhead, 2016-07-10:
Another thing that may help in `ultralcd.cpp`…?
```diff
    #if ENABLED(LCD_USE_I2C_BUZZER)
      lcd.buzz(LCD_FEEDBACK_FREQUENCY_DURATION_MS, LCD_FEEDBACK_FREQUENCY_HZ);
    #elif PIN_EXISTS(BEEPER)
      buzzer.tone(LCD_FEEDBACK_FREQUENCY_DURATION_MS, LCD_FEEDBACK_FREQUENCY_HZ);
+     buzzer.tick();
    #endif
```

thinkyhead, 2016-07-10:
Alternatively… the delay in `lcd_quick_feedback` could be done as a loop which calls `buzzer.tick` every one or two ms.

jbrazio, 2016-07-10:
@lavato is it related with the SD card menu item(s) ?

lavato, 2016-07-10:
@jbrazio 
I would say yes.
Also, It appears that the sound will stop on its own if you leave it long enough.
I have a feeling that when the CPU is busy it just cannot stop the sound.

@thinkyhead 
I'll try replacing lines whit your suggested changes to see and get back.

lavato, 2016-07-10:
Solution in issue #4249  also "fixed" my buzzer problem.

However, I still think there is an issue with the  way how buzzer code stops the sound, as any delay or "hang" appears to cause this issue.

As I said above, I will try thinkyhead's  code before closing this.

lavato, 2016-07-10:
@thinkyhead 

For me this worked with the "bad" SD card. Thank you!
```

-      else if (millis() >= this->state.timestamp) this->reset();
+      else if (ELAPSED(millis(), this->state.timestamp)) this->reset();
#define PENDING(NOW,SOON) ((long)(NOW-(SOON))<0)
#define ELAPSED(NOW,SOON) (!PENDING(NOW,SOON))
```

BTW: How do you paste colored code into the comment?

thinkyhead, 2016-07-11:
> BTW: How do you paste colored code into the comment?

Right after the three backticks you can append a file-extension for the language whose highlighting you want.
```
Start a block with ```cpp for C++ highlighting
Start a block with ```diff or ```patch for diff highlighting
et-cetera!
```

closed 2016-07-11 by thinkyhead

lavato, 2016-07-11:
@thinkyhead Cheers for that.