MarlinFirmware/Marlin #4249: Selecting a file using the menu is extremely slow

lavato, 2016-07-09:
When printing from SD, scrolling trough file list is extremely slow.
This has been an annoyance for a while so decided to raise it. 

I also believe the Buzzer issue I raised  in a separate issue can be reproduced here.

Any other menu area is quick and responsive.

I am using RAMPS1.4 with the Full Discounted LCD option.

Blue-Marlin, 2016-07-09:
How many files do you have in the directory?

lavato, 2016-07-09:
If more than 3-4 files, it start getting very slow.

lavato, 2016-07-09:
...to paint a better picture.
* When I turn the encoder while on the file list section, the selector only updates the screen about 2-4 second later.
* When I click on a file to print, it reacts about 2 seconds later.

Blue-Marlin, 2016-07-09:
Did you play with 
```
//
// SD CARD: SPI SPEED
//
// Uncomment ONE of the following items to use a slower SPI transfer
// speed. This is usually required if you're getting volume init errors.
//
//#define SPI_SPEED SPI_HALF_SPEED
//#define SPI_SPEED SPI_QUARTER_SPEED
//#define SPI_SPEED SPI_EIGHTH_SPEED
```
?

lavato, 2016-07-09:
I haven't - I will try tomorrow.

RicardoGA, 2016-07-10:
@lavato this happened to me once and after do a low level format to the card the problem was solved

lavato, 2016-07-10:
@RicardoGA - What did you use to low level format the SD card?

RicardoGA, 2016-07-10:
@lavato i used HDD Low Level Format Tool,  here's the link

http://hddguru.com/software/HDD-LLF-Low-Level-Format-Tool/

lavato, 2016-07-10:
@RicardoGA 

OMG that worked, thank you!!

thinkyhead, 2016-07-11:
@MarlinFirmware/documentation-maintainers We should mention this potential event in our SD Card section. Which I guess means adding an SD Card section…

closed 2016-07-11 by thinkyhead