MarlinFirmware/Marlin #4260: [improvement] [delta] [Bedleveling] [grid] Add leveling probe at [0,0] for easier analysis

kdupke, 2016-07-10:
During my own delta build I figured out that my delta radius was not proper set because the nozzle dragged into the bed (with the other marlin fork without a nozzle lift during probe).

To figure out if I have a concave or convex setting (expecting the bed to be flat because glass), I tuned delta radius to get the nozzle z measurement delta around on the bed close to zero compared to what I measure on [0,0]. To get this relative measurement it would be good to have the first or last measurement at [0,0].

This also can show me if the bed is plan mounted or if it is uneven placed on the FSR mounts.

Attached you see the graphical display when the data is used for a bubble diagram over X/Y.Easily to see that with R=177 the nozzle is too high at the outside and the bed is higher on X=0,Y=-100, then at X=0,Y=100.

[Oh, and this is using the difference based on the closest to [0,0] point, which is [-2,-2] here.

[image]
[image]
[image]
[image]

I expect that with a better leveled bed and more accurate radius I might not need autocalibration or it might still provide better print results. If this is wrong and a just enough flat bed is OK then the effort I do might be not needed....

regards,
kai

kdupke, 2016-07-10:
P.S.: 1.1.0-RC6

jbrazio, 2016-07-11:
Exactly what is the question ?

Roxy-3D, 2016-07-11:
> Exactly what is the question ?

I'm also not clear on what you asking for help.  But 'Thank You' for taking the time to provide all of the detail.    Incidentally, on my branch where I'm doing the Unified Bed Leveling work, I have several things to help the Delta crowd get their printers calibrated better.   First, with the high resolution Mesh being used by the system, it is possible the calibration of Delta printers becomes less important.  (That remains to be seen!)

But separate from that fact there are two other improvements.   The M665 command is expanded to have radius trim for each of the towers.     And there is also a Delta physical bed leveling tool (G27) to help adjust the bed and get data to feed into the M665 command:

~~~~
#if ENABLED(DELTA)
  /**
   * M665: Set delta configurations
   *
   *    L = diagonal rod
   *    R = delta radius
   *    S = segments per second
   *    A = Alpha (Tower 1) diagonal rod trim
   *    B = Beta (Tower 2) diagonal rod trim
   *    C = Gamma (Tower 3) diagonal rod trim
   *    I = Alpha (Tower 1) radius trim
   *    J = Beta (Tower 2) radius trim
   *    K = Gamma (Tower 3) radius trim
   */
  inline void gcode_M665() {
    if (code_seen('L')) delta_diagonal_rod = code_value_linear_units();
    if (code_seen('R')) delta_radius = code_value_linear_units();
    if (code_seen('S')) delta_segments_per_second = code_value_float();
    if (code_seen('A')) delta_diagonal_rod_trim_tower_1 = code_value_linear_units();
    if (code_seen('B')) delta_diagonal_rod_trim_tower_2 = code_value_linear_units();
    if (code_seen('C')) delta_diagonal_rod_trim_tower_3 = code_value_linear_units();
    if (code_seen('I')) delta_radius_trim_tower_1 = code_value_linear_units();
    if (code_seen('J')) delta_radius_trim_tower_2 = code_value_linear_units();
    if (code_seen('K')) delta_radius_trim_tower_3 = code_value_linear_units();
    recalc_delta_settings(delta_radius, delta_diagonal_rod);
  }
~~~~

And the header block from the G27 Delta Physical Bed Leveling tool:

~~~~
// G27 3-Point Supported Bed Adjustment
//
// This function greatly helps in physically leveling a bed supported at its perimeter by 3 points.
// There are several modes of operation.  The function defaults to C-ircular if no other probing option
// is specified.  Suggested usage is to use the C option ignoring the outer circle numbers.  Then use
// the L option to get the DELTA_RADIUS closer and then fine tune the C option's outer circle numbers.
//
// C -	Circular Probing Operation.  This is primarily useful for measuring the height at the 3 bed support
// 	points at uniform distances from the center of the bed.  It will probe 3 points
// 	along the perimeter of a circle.  The operation will default to 3 circles with the outer most circle at
//	DELTA_PROBEABLE_RADIUS if no radius is specified.   The positioning of the 3 bed support points is
//	assumed to be halfway between each tower unless an angular offset is given using the T option.  A number
//	can follow the C command and will specify the number of circles to probe.  If no number is provided the
//	command will default to 3 circles with the outer most circle at the defaulted or specified radius.
//
//	The outer circle numbers are useful because they most closely measure the height of the bed
//	at the mounting points.  However, those numbers should not be as trusted as the inner numbers because
//	Delta printers have positioning inaccuracies that compound as they move further from center.   The
//	inner circles are more useful for accurate leveling.
//
// L -	Radial lines from the far side of the bed through the origin to the Towers.  This is useful to see the
//	slope of the bed straight away from the towers.  But it is especially valuable when trying to get the
//	DELTA_RADIUS number set accurately.  The command assumes the back tower is straight back.
//
// D -	Displace Probe.  Aligns lines of sampled points to the 1st nozzle instead of to the probe offset.
// 	This will displace the Z-Probe from the radial lines such that the nozzle tracks the radial line.  This
// 	will result in the diagonal rods being straight out from the tower that is having a radial line probed to
// 	it.
//
// O -	Set the origin on the bed if not at the bed's center. [CURRENTLY NOT IMPLEMENTED.  WAITING FOR FEEDBACK]
//
// T -	Set the angular offset for bed mount points and radial lines to the towers.  If the bed support points
// 	are not positioned halfway between the towers you will need to consider running the C and L commands
// 	seperately with different offsets to get the desired numbers.  T represents the Theta of the back tower.
// 	Theta defaults to 90 degrees using an X / Y grid on the printer's bed.
//
// R -	Set the radius for the probing operations.  DELTA_PROBEABLE_RADIUS is used as the maximum radius if
// 	no radius is specified.
//
// Example commands:	G27 C 			Probe 3 bed mounting points at DELTA_PROBEABLE_RADIUS
//			G27 C 4 T 30 R 80	Probe 3 bed mounting points at a radius of 80mm and
//						angular displacement of 30 degrees with 4 circles
//			G26 C 2  R 65		Probe 3 bed mounting points at a radius of 65mm with 2 circles
//			G27 L 			Probe 3 radial lines going to each tower using a radius of
//						DELTA_PROBEABLE_RADIUS with 5 points per line
//			G27 L 7			Probe 3 radial lines going to each tower using a radius of
//						DELTA_PROBEABLE_RADIUS with 7 points per line
//			G27 L 7  R 65		Probe 3 radial lines going to each tower using a radius of
//						65mm and 7 points per line
//
~~~~

kdupke, 2016-07-11:
jbrazio:
Title Add leveling probe at [0,0] for easier analysis

Make sure the bed level probes when using G29 always probe the coordinate [0,0] (preferred) at the beginning of a probe series.

Roxy-3D:
Does this mean I should use a different branch to test? Or with the different branch I would not need to do bed leveling or estimate delta radius at all?

Roxy-3D, 2016-07-11:
@kdupke   It was mostly a "For your information".  However, re-reading your posts,  your pictures don't really make sense to me.  The deformations are 'square'.   With Deltas, usually you have problems forming a pattern along radius lines going either to a tower, or going to a location between two towers.  You don't have that.

If you are in a hurry to add the delta radius trim it is very straight forward to do.  I can send you the files that do it.   But if you are willing to wait a week or two,  The G27 Delta Physical Bed Leveling tool should be ready to release also.   (Right now, it needs to be tweaked because of all the probe changes that have happened, and I'm busy working on the Unified Bed Leveling, so it will take a few weeks until that is fully baked again.)

kdupke, 2016-07-11:
@Roxy-3D that is because the measure grid is square and not circular, and the bed might be plan but isn't in X/Y pane (means the mounting points are not all on the same level), and with a concave/convex due to a wrong radius the outer delta is different than the inner delta. Which is the reason I want to adjust the radius accordingly. 

Please send me the link/patch, I have some other duties so I might not be able to immediately test it, but it will be on my top list of things.

Roxy-3D, 2016-07-11:
You have the code (up above) for the altered M665 command.   Add that to your Marlin_main.cpp file.  Then add these declarations to Marlin_main.cpp also:
~~~~
  float delta_radius = DELTA_RADIUS;
  float delta_radius_trim_tower_1, delta_radius_trim_tower_2, delta_radius_trim_tower_3;
  float delta_tower1_x = -SIN_60 * (delta_radius + DELTA_RADIUS_TRIM_TOWER_1); // front left tower
  float delta_tower1_y = -COS_60 * (delta_radius + DELTA_RADIUS_TRIM_TOWER_1);
  float delta_tower2_x =  SIN_60 * (delta_radius + DELTA_RADIUS_TRIM_TOWER_2); // front right tower
  float delta_tower2_y = -COS_60 * (delta_radius + DELTA_RADIUS_TRIM_TOWER_2);
  float delta_tower3_x = 0;                                                    // back middle tower
  float delta_tower3_y = (delta_radius + DELTA_RADIUS_TRIM_TOWER_3);
~~~~

and 

~~~~
  void recalc_delta_settings(float radius, float diagonal_rod) {
    delta_tower1_x = -SIN_60 * (radius + delta_radius_trim_tower_1);  // front left tower
    delta_tower1_y = -COS_60 * (radius + delta_radius_trim_tower_1);
    delta_tower2_x =  SIN_60 * (radius + delta_radius_trim_tower_2);  // front right tower
    delta_tower2_y = -COS_60 * (radius + delta_radius_trim_tower_2);
    delta_tower3_x = 0.0;                                             // back middle tower
    delta_tower3_y = (radius + DELTA_RADIUS_TRIM_TOWER_3);
    delta_diagonal_rod_2_tower_1 = sq(diagonal_rod + delta_diagonal_rod_trim_tower_1);
    delta_diagonal_rod_2_tower_2 = sq(diagonal_rod + delta_diagonal_rod_trim_tower_2);
    delta_diagonal_rod_2_tower_3 = sq(diagonal_rod + delta_diagonal_rod_trim_tower_3);
  }
~~~~

Alter Marlin.h to be like this:
~~~~

  extern float delta_diagonal_rod;
  extern float delta_segments_per_second;
  extern float delta_diagonal_rod_trim_tower_1;
  extern float delta_diagonal_rod_trim_tower_2;
  extern float delta_diagonal_rod_trim_tower_3;
  extern float delta_radius_trim_tower_1;
  extern float delta_radius_trim_tower_2;
  extern float delta_radius_trim_tower_3;
~~~~

and lastly, alter Configuration_store.cpp to be like this in three places:
~~~~
EEPROM_WRITE_VAR(i, delta_diagonal_rod_trim_tower_2); // 1 float
EEPROM_WRITE_VAR(i, delta_diagonal_rod_trim_tower_3); // 1 float
EEPROM_WRITE_VAR(i, delta_radius_trim_tower_1); // 1 float
EEPROM_WRITE_VAR(i, delta_radius_trim_tower_2); // 1 float
EEPROM_WRITE_VAR(i, delta_radius_trim_tower_3); // 1 float
#elif ENABLED(Z_DUAL_ENDSTOPS)
EEPROM_WRITE_VAR(i, z_endstop_adj); // 1 float
 
EEPROM_READ_VAR(i, delta_diagonal_rod_trim_tower_2); // 1 float
EEPROM_READ_VAR(i, delta_diagonal_rod_trim_tower_3); // 1 float
00370: EEPROM_READ_VAR(i, delta_radius_trim_tower_1); // 1 float
00371: EEPROM_READ_VAR(i, delta_radius_trim_tower_2); // 1 float
EEPROM_READ_VAR(i, delta_radius_trim_tower_3); // 1 floa
recalc_delta_settings(delta_radius, delta_diagonal_rod);
#elif ENABLED(Z_DUAL_ENDSTOPS)
 
delta_diagonal_rod_trim_tower_2 = DELTA_DIAGONAL_ROD_TRIM_TOWER_2;
delta_diagonal_rod_trim_tower_3 = DELTA_DIAGONAL_ROD_TRIM_TOWER_3;
delta_radius_trim_tower_1 = DELTA_RADIUS_TRIM_TOWER_1;
delta_radius_trim_tower_2 = DELTA_RADIUS_TRIM_TOWER_2;
delta_radius_trim_tower_3 = DELTA_RADIUS_TRIM_TOWER_3;

recalc_delta_settings(delta_radius, delta_diagonal_rod);
 
SERIAL_ECHOPAIR(" B", delta_diagonal_rod_trim_tower_2);
00787: SERIAL_ECHOPAIR(" C", delta_diagonal_rod_trim_tower_3);
00788: SERIAL_ECHOPAIR(" I", delta_radius_trim_tower_1 );
SERIAL_ECHOPAIR(" J", delta_radius_trim_tower_2 );
SERIAL_ECHOPAIR(" K", delta_radius_trim_tower_3 );
SERIAL_EOL;
#elif ENABLED(Z_DUAL_ENDSTOPS)
~~~~
After doing this, you can dynamically adjust your Delta Radius Trim on all three towers and save (and restore) the settings from EEPROM.

Roxy-3D, 2016-07-11:
> @Roxy-3D that is because the measure grid is square and not circular, and the bed might be plan but isn't in X/Y pane (means the mounting points are not all on the same level), and with a [...]

You would benefit from the G27 Physical Bed Leveling tool.   It probes (and reports) the values along lines to the towers and in circular paths around the bed.  It helps you see very clearly where the the issues are.   I'll try to get that running again with the new probe code soon.

kdupke, 2016-07-11:
@Roxy-3D : thank you. Do you have a diff style patch? "diff -c originalfile updatedfile" That would be much easier to apply. Or is this in some git repository so I can create my own diff?

Roxy-3D, 2016-07-11:
Soon...  (About 2 weeks???)  I'll have it all up on GitHub as a branch off of RC-6 RCBugFix.  (Or maybe RC-7 RCBugFix).   But until then, you'll have to Cut & Paste.

kdupke, 2016-07-11:
On 07/12/2016 12:04 AM, Roxy-3D wrote:
> Soon... (About 2 weeks???) I'll have it all up on GitHub as a branch off
> of RC-6 RCBugFix. (Or maybe RC-7 RCBugFix). But until then, you'll have
> to Cut & Paste.

OK.

regards,
kai

jbrazio, 2016-08-01:
UBL is now available at:
https://github.com/MarlinFirmware/Marlin/tree/devel-ubl

closed 2016-08-01 by jbrazio

Blue-Marlin, 2016-08-01:
@jbrazio 
Please explain how to comment on 'branches'.

jbrazio, 2016-08-01:
Comment on branches ?