MarlinFirmware/Marlin #4261: [improvement] [delta] [Bedleveling] [grid] [G29] Make probe_feedrate configurable

kdupke, 2016-07-10:
Version 1.1.0-RC6

In file Marlin_main at line 1555:
// move down slowly until you find the bed
      feedrate = homing_feedrate[Z_AXIS] / 4;

would be nice to have this configurable as
/ move down slowly until you find the bed
      feedrate = PROBE_FEEDRATE;

My homing feed rate results in a believed too high probe feed rate, which result in harder than needed pushes on the build surface.

Would be nice to have this hard codable or as a runtime variable or parameter to G29, same as the travel speed is adjustable already today.

regards,
kai

Blue-Marlin, 2016-07-10:
> feedrate = homing_feedrate[Z_AXIS] / 4;

is a foul compromise. Because the the probing is not made with a double touch for deltas.
For the first touch `homing_feedrate[Z_AXIS] ` should be ok. The second touch should be made with `homing_feedrate[Z_AXIS] / HOMING_BUMP_DIVISOR[Z_AXIS]`.

If the home-bump-speed should not be better a direct speed than calculated with the divisor is worth an other discussion.

kdupke, 2016-07-11:
@Blue-Marlin : feel free to open another improvement request for double bump. 

I'm not so much for setting a divisor if the two values are not really related.

The homing speed is not related to the probe speed, so having the probe speed relative to it seems to be wrong.

Homing has not to be that accurate (because after it it is homed), homing has is own second bump already, and finally the mechanics are different (it does not break the end stop switches, but the probe could break the glass).

That said, having a divisor for the second bump sounds reasonable, having probe speed related to homing speed not.

regards,
kai

thinkyhead, 2016-07-13:
Looks like we should have this soon - separate speeds for the first (fast) and second (slow) probes.

AnHardt, 2016-12-10:
Done.

closed 2016-12-11 by thinkyhead