MarlinFirmware/Marlin #4262: Z Probe / Home configuration

RDickens, 2016-07-10:
I am having problems setting my Configuration.h file to use a servo arm level probe at the same time as using a Z Min switch. I have a couple of different hot ends and they are different lengths and was hoping to just have top tweak my Z Min limit switch position to easily swap between hot ends.

I'm trying to get it to home all the axis using the fixed limit switches with the servo arm retracted. Then when asked to level the bed to then swing the servo arm into position and do the standard probing routing, using the probe data to level the bed but not destroying the Z Min setting.  

I have a separate pins for the servo arm switch and the Z Min switch.
I'm running a Prusa i2 with only 1 extruder
RAMPS 1.4.

Can anyone make any suggestions

Blue-Marlin, 2016-07-10:
Try
```
//
#define Z_MIN_PROBE_ENDSTOP

// Enable Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN to use the Z_MIN_PIN for your Z_MIN_PROBE.
// The Z_MIN_PIN will then be used for both Z-homing and probing.
//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

// To use a probe you must enable one of the two options above!

// This option disables the use of the Z_MIN_PROBE_PIN
// To enable the Z probe pin but disable its use, uncomment the line below. This only affects a
// Z probe switch if you have a separate Z min endstop also and have activated Z_MIN_PROBE_ENDSTOP above.
// If you're using the Z MIN endstop connector for your Z probe, this has no effect.
//#define DISABLE_Z_MIN_PROBE_ENDSTOP
```
and in Marlin_main.cpp (RCBugFix)
```diff
@@ -2360,11 +2360,11 @@ static void homeaxis(AxisEnum axis) {
       (axis == X_AXIS) ? x_home_dir(active_extruder) :
     #endif
     home_dir(axis);
 
   // Homing Z towards the bed? Deploy the Z probe or endstop.
-  #if HAS_BED_PROBE
+  #if HAS_BED_PROBE && DISABLED(Z_MIN_PROBE_ENDSTOP)
     if (axis == Z_AXIS && axis_home_dir < 0) {
       #if ENABLED(DEBUG_LEVELING_FEATURE)
         if (DEBUGGING(LEVELING)) SERIAL_ECHOPGM("> ");
       #endif
       if (DEPLOY_PROBE()) return;
@@ -2461,11 +2461,11 @@ static void homeaxis(AxisEnum axis) {
   endstops.hit_on_purpose(); // clear endstop hit flags
   axis_known_position[axis] = true;
   axis_homed[axis] = true;
 
   // Put away the Z probe
-  #if HAS_BED_PROBE
+  #if HAS_BED_PROBE && DISABLED(Z_MIN_PROBE_ENDSTOP)
     if (axis == Z_AXIS && axis_home_dir < 0) {
       #if ENABLED(DEBUG_LEVELING_FEATURE)
         if (DEBUGGING(LEVELING)) SERIAL_ECHOPGM("> ");
       #endif
       if (STOW_PROBE()) return;
```

RDickens, 2016-07-11:
Thanks that worked a treat :)  
I did have to tweak my Z_PROBE_OFFSET_FROM_EXTRUDER setting to match my Z Min switch location but apart from that its great.

Now I'm using the RCBugFix version my RepRapDiscount FULL GRAPHIC Smart Controller seems a little sluggish is this just down to the fact its not a full release version of the firmware?

thinkyhead, 2016-07-12:
> move axis usually responds much faster

It now waits for 1/4 second after you stop touching the encoder before it starts to move, rather than moving immediately. We might want to change this for the smaller move distances. It's meant to prevent accumulating a bunch of moves until you've honed your destination.

thinkyhead, 2016-07-12:
> buzzer tones don't seem to be adjusting correctly

It might be slightly better today? We tweaked the buzzer code a little bit yesterday. The beeper on my graphical display has never sounded very good (a lot of "noise" in the tones), but do you find it worse with `RCBugFix`?

RDickens, 2016-07-12:
Ahh that explains the delay on moving the axis, I would suggest you scrap the delay on the 0.1mm moves as you should never accidentally go crazy with that one but it would be nice not to have a delay when stepping the extruder or tweaking the axis.

I cant tell any difference with the buzzer but as you say it never sounded grate ion the first place..

thinkyhead, 2016-07-13:
@RDickens Done! #4285

closed 2016-07-18 by jbrazio

PaulStew67, 2017-05-27:
I'm trying to do the same, home Z with one endstop and level the bed with the servo endstop.  Im using a Ramps 1.4 board with only a single Z Min connection, how do you wire this so that I can use either endstop?  Thanks

Bob-the-Kuhn, 2017-06-03:
Please try [bugfix-1.1.x]( https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x).  There's been a fix installed for similar issues.