MarlinFirmware/Marlin #4267: Issue dual printing (RCBugFix)

Alex9779, 2016-07-11:
In addition to #4266 I have the following issue when trying to dual print.
I am trying to print this file:
[20mm-box.zip](https://github.com/MarlinFirmware/Marlin/files/356804/20mm-box.zip)
Which looks like this:
[image]
and the result is this:
[image]
The first tool prints ok, the second tool does strange moves. Actually there is filament only on two spots but it is on the left side too. This cannot be seens because that additional move extrudes filament and then the head moves the same way back but the extruder is also running reversed with extrusion speed. So on the left side there is no filament in the nozzle anymore to be extruded because it has been "retracted" before too much...

thinkyhead, 2016-07-15:
I've added more logging for `gcode_T` in https://github.com/thinkyhead/Marlin/tree/rc_debug_gcode_t so please test!

Alex9779, 2016-07-18:
Like I commented in #4266 I will test with latest RCBugFix, if I have issues I will activate debugging and report...

Alex9779, 2016-07-18:
I did this test again and except the strange things regarding homing with a `T1` I mentioned in #4266 selected this seemed ok now...

closed 2016-07-18 by Alex9779