MarlinFirmware/Marlin #4288: when I use cura to send gcode,3d printer did not work.

lin6328, 2016-07-13:
I want to use marlin firmware to do sla 3d printer.SO I modify the temperature to cheat the cura.But when I kick the botton "printing", it did not work. I want to know why?

thinkyhead, 2016-07-13:
It is one of the great mysteries. Many have asked, but few have received the answer.

paulusjacobus, 2016-07-13:
if you just need marlin for doing the z axis lifting then there are easier
options. I use grbl with creativeworks which does a great job. You even
don't need a mega but can use an Arduino Uno with just one stepper.

lin6328, 2016-07-13:
@paulusjacobus that is DLP 3D printer, not SLA 3D printer. I also need x axis and y axis.The different is that I do not use motor. But GRBL also can do this.

paulusjacobus, 2016-07-13:
Ahh you are drive galvano meters and a laser. Good news is that Grbl can
drive upto 4 axis and drive the laser with a pwm signal and switch it
on/off. A cnc shield cost abt 7 Dollars on ebay.

lin6328, 2016-07-14:
@paulusjacobus that is right. Maybe one day , you will see my SLA 3D-printer on kickstarter.ahh

closed 2016-07-18 by jbrazio