MarlinFirmware/Marlin #4290: [Question] : does Marlin support to pause immediately

kloknibor, 2016-07-13:
Hi guys!

I'm thinking about implementing marlin in an resin printer. But this resin printer has an switch which detect if the hood is present. As soon as this hood is present there should be an error printed and movement stopped(So as an interrupt and not waiting till the movement is done) and preferably it should be paused but if I can show this error my host software can pause it too. 

Is this easy to implement in Marlin as an beginner in C? I've wrote a C program before but not in combination with Marlin and I have experience programming PLC's .

Thanks!

Robin

thinkyhead, 2016-07-13:
We recently added the ability to do a quick-stop with `stepper_quickstop()`. Using this function, if the machine is moving very fast steps may be lost, but it works.

As for detecting the hood switch, you can wire an extra opto endstop, and enable the options which keep endstops active during printing. Adding code to respond to the extra endstop is pretty straightforward. Take a look at functions like `gcode_M119` which reports the current endstop status, and follow the trail of symbols to the `endstops.cpp` file where the endstops are managed.

kloknibor, 2016-07-13:
That sounds exaclty like what I need! The swotch is present already, just a simple switch ;)! But this is great! I will try to integrate it :)

I really appreciate your answer! Thank you!

closed 2016-07-13 by kloknibor