MarlinFirmware/Marlin #4293: Possibility to list all G and M codes

thopiekar, 2016-07-13:
While thinking about an "remote control" implementation in Cura, I just found out that this is missing in Marlin:

* http://reprap.org/wiki/G-code#G:_List_all_G-codes
* http://reprap.org/wiki/G-code#M:_List_all_M-codes

It would be very helpful to have this, as we could ask the printer for supported commands and then provide buttons to trigger filament change for example. Of course you could add this information manually, but asking the printer makes more sense then updating the printers information on every firmware upgrade.

Roxy-3D, 2016-07-13:
I tried to edit it a few commands to the list.  It wouldn't let me.   Specifically, I was going to add M100, G27 and G26.   It is supposed to be a Wiki, but even when I create an account, it won't let me do anything.

jbrazio, 2016-07-13:
It's a good catch but this will be a PROGMEM eater..

Blue-Marlin, 2016-07-13:
It would nor be machine readable anyway.

We could react with a link.
https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/Marlin_main.cpp#L100
That's always the the most up to date info we have (besides the code itself - and even for that they are on the right source then.)

Blue-Marlin, 2016-07-13:
Ups. Some lines above we need some updates.

jbrazio, 2016-07-13:
What do you mean @Blue-Marlin ?

Blue-Marlin, 2016-07-14:
```
 * Look here for descriptions of G-codes:
 *  - http://linuxcnc.org/handbook/gcode/g-code.html // 404
 *  - http://objects.reprap.org/wiki/Mendel_User_Manual:_RepRapGCodes //404
 *
 * Help us document these G-codes online:
 *  - https://github.com/MarlinFirmware/Marlin/wiki/G-Code-in-Marlin // needs update
 *  - http://reprap.org/wiki/G-code
```

Roxy-3D, 2016-07-14:
The RepRap link is BS.   Please go try to edit the Wiki.   I wasted 45 minutes today trying to get it updated.   You sign up for an account, and it tells you to go sign up for an account.  And, I already had an account at RepRep to monitor the Marlin Firmware area.

Obviously, that link is never going to be current and correct.   I'm grabbing a couple new GCodes, but there is NO REASON for me to pay any attention to any of the information there.   Who knows???  Maybe they will catch up and get my new GCodes documented there.   But I'm not going to trust their information as being correct.  They can't even figure out how to make a Wiki work right.

AnHardt, 2016-07-14:
Ähhh - @Roxy-3D 
Login in end editing at the RepRap wiki was no problem for me. I edited the 
http://reprap.org/wiki/G-code#G:_List_all_G-codes
http://reprap.org/wiki/G-code#M:_List_all_M-codes
articles yesterday to malin = no.

For me the wiki is very slow - but working.

bobc, 2016-07-14:
@Roxy-3D RepRap admin had to restrict access due to massive spam problems, please see topic http://forums.reprap.org/read.php?33,632743

jbrazio, 2016-07-18:
Closing due to being currently offtopic, initial topic question was answered.
Please re-open this issue if still valid.

closed 2016-07-18 by jbrazio