MarlinFirmware/Marlin #4295: E in mm3 always On - RC6

epoxi, 2016-07-13:
Hi!

Last modification to my firmware was to add the M600 change filament menu option.
After that, I started to notice that the amount of extruded material was really poor. Looking at the configuration (LCD) found that the extrusion in mm3 was On. I couldn't change it and remains always On.

Attached are my config files.

Marlin 1.1.0-RC6 on Rumba board.

[Marlin-config.zip](https://github.com/MarlinFirmware/Marlin/files/362115/Marlin-config.zip)

Regards.-

epoxi, 2016-07-13:
Ok. Restoring "failsafe" dafaults and saving them corrected the problem.

Thanks in advance.

closed 2016-07-14 by jbrazio

thinkyhead, 2016-07-14:
@epoxi We've patched the EEPROM to be more robust, so hopefully this won't happen in 1.1.0.