MarlinFirmware/Marlin #4313: Bug V1.1.0 RC6 when selected "disable steppers"

raoul596, 2016-07-15:
When you disable your stepper motors the software still "thinks" the motors are in the place where he left them. Not sure if you will get the same problem when the printer disables the stepper motors because of the stepper drivers timeout function.

Blue-Marlin, 2016-07-15:
You should be warned by the blinking 'X' <-> ' ' coordinate letters on the display.

raoul596, 2016-07-15:
Yeah, that's right, but it shouldn't allow me to do the auto bed level procedure before homing again. isn't it?

jbrazio, 2016-07-15:
I do agree no ABL/MBL should be allowed if the machine is not homed.
@thinkyhead any thoughts ?

avidtech, 2016-07-15:
I have always inserted G28 before G29 in my starting G codes. Technically, your procedure should work fine if the axes are not moved after a stepper timeout and the machine remains on.

Blue-Marlin, 2016-07-15:
@jbrazio 
> I do agree no ABL/MBL should be allowed if the machine is not homed.

You did not understand the problem. We already require homed axis before G29.
What @raoul596 means is requiering known_position, what is switched off when the steppers shut down. Depending on the static load on the axis we may lose 0, 1 or a infinite number of full steps.
If the axis are not moved by hand, usually the number of steps lost is 0 or 1 - a number where we can live with on most systems. Only when the axis moves by itself when the stepper(s) are off we have a real problem - but in this case i expect not to shut down the stepper(s).

Is is god as it is now.

Ether don't switch off the steppers,
or home before G29,
or live with a few lost steps (and be warned about that by a blinking display).
or ignore the problem at all by setting `DISABLE_REDUCED_ACCURACY_WARNING`
or replace `axis_homed` with axis_known_position for example in `axis_unhomed_error()` and everywhere else you like.

closed 2016-07-15 by raoul596

thinkyhead, 2016-07-16:
> I do agree no ABL/MBL should be allowed if the machine is not homed.

True. But "unknown position" is less serious. A few lost steps is not problematic in most instances. If you move your carriages by hand and then issue `G29` before homing, you learn quickly not to do that. I suppose we could offer ever more tighter controls, even up to the point of requiring a `G28` after the steppers go to sleep, before movement commands or `G29` will work at all. (So the firmware may remain blameless!) But then Marlin gets the terrible reputation of being too much of a "nanny" firmware. So, who knows? Perhaps in future we do add an option to require re-homing every time steppers sleep.