MarlinFirmware/Marlin #4315: Incorrect Z position after G29 Auto Bed Leveling

funbobby2001, 2016-07-15:
I'm setting up auto bed leveling/compensation for my first time and I could not get the correct Z height using Marlin 1.0.2. It seemed to be ignoring my probe z offset. I am using the Z min endstop on the board as my home and probe.

I just figured out that commenting the following line in Marlin_main.cpp fixes the issue:

//current_position[Z_AXIS] = zprobe_zoffset; // in the lsq we reach here after raising the extruder due to the loop structure

avidtech, 2016-07-15:
I also experienced this in the past, I didn't change anything in the source, though. 
I set the parameters M218 (and other M code parameters), after reading from memory, then storing to eeprom with M500. This problem (and a few others) went away for me after the manual steps.

TimlEU, 2016-07-15:
im having the same problem

Blue-Marlin, 2016-07-15:
Please help us to help you.

What do you see?
What do you expect?
Steps to reproduce?
Configs?

1.0.2 will not be patched any more. Try RCBugFix.

Roxy-3D, 2016-07-16:
@jbrazio  Did you see this?     It might be worth checking on your machine and seeing if the problem gets resolved.

jbrazio, 2016-07-16:
The closest reference I can see is [this one](https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/Marlin_main.cpp#L1533).
This does not seem to be the exact same error, because my z-offset is not ignored, I had to increase it comparing with the previous fw version and with no physical changes on the machine. Nevertheless I'm keeping an eye on this issue.

thinkyhead, 2016-07-16:
@jbrazio What was the difference, again, between your proper nozzle-to-probe offset and the one that you are required now to use? Did you already post your G28 / G29 debug log output and probe configs on another issue?

jbrazio, 2016-07-16:
I didn't open a specific issue for it we just discussed it on Hangouts.
The offset I used was -0.96 and now it's something like -1.50 in order to have the PLA squash into the table. I will attach a debug log.

jbrazio, 2016-07-16:
<details>
<summary>G28/G29 log</summary>
```
Connecting...
echo:Marlin 1.1.0-RCBugFix
echo: Last Updated: 2016-04-27 12:00 | Author: @jbrazio
Compiled: Jul 15 2016
echo: Free Memory: 2927  PlannerBufferBytes: 1296
echo:V24 stored settings retrieved (398 bytes)
echo:Steps per unit:
echo:  M92 X160.00 Y160.00 Z8000.00 E203.00
echo:Maximum feedrates (mm/s):
echo:  M203 X250.00 Y250.00 Z2.00 E200.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X1000 Y1000 Z20 E1000
echo:Accelerations: P=printing, R=retract and T=travel
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X10.00 Z0.40 E2.00
echo:Home offset (mm)
echo:  M206 X0.00 Y0.00 Z0.00
echo:Material heatup parameters:
echo:  M145 S0 H210 B70 F0
echo:  M145 S1 H240 B110 F0
echo:PID settings:
echo:  M301 P19.18 I1.36 D67.42 C100.00 L20
echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0
echo:Z-Probe Offset (mm):
echo:  M851 Z-1.50
echo:SD card ok
echo:SD card ok
Printer is now online.
>>> M113 S60
SENDING:M113 S60
>>> M111 S32
SENDING:M111 S32
echo:DEBUG:LEVELING
>>> G28
SENDING:G28
>>> gcode_G28
current_position=(0.00, 0.00, 0.00) : setup_for_endstop_or_probe_move
> endstops.enable(true)
Raise Z (before homing) to 5.00
>>> homeaxis(1)
current_position=(0.00, 0.00, 5.00) : sync_plan_position
current_position=(0.00, 0.00, 5.00) : sync_plan_position
current_position=(0.00, 0.00, 5.00) : > TRIGGER ENDSTOP
>>> set_axis_is_at_home(1)
For 89 axis:
 home_offset = 0.00
 position_shift = 0.00
 sw_endstop_min = 0.00
 sw_endstop_max = 297.00
> home_offset[89] = 0.00
current_position=(0.00, 0.00, 5.00) :
<<< set_axis_is_at_home(1)
current_position=(0.00, 0.00, 5.00) : sync_plan_position
current_position=(0.00, 0.00, 5.00) : > AFTER set_axis_is_at_home
<<< homeaxis(1)
current_position=(0.00, 0.00, 5.00) : > homeY
>>> homeaxis(0)
current_position=(0.00, 0.00, 5.00) : sync_plan_position
current_position=(0.00, 0.00, 5.00) : sync_plan_position
current_position=(0.00, 0.00, 5.00) : > TRIGGER ENDSTOP
>>> set_axis_is_at_home(0)
For 88 axis:
 home_offset = 0.00
 position_shift = 0.00
[... 109 lines cut ...]
current_position=(142.00, 267.00, 3.50) : set_probe_deployed
deploy: 0do_probe_raise(5.00)
do_blocking_move_to(142.00, 267.00, 6.50)
do_blocking_move_to(142.00, 267.00, 6.50)
current_position=(142.00, 267.00, 6.50) : clean_up_after_endstop_or_probe_move
current_position=(142.00, 267.00, 6.50) : > probing complete
Eqn coefficients: a: 0.00249252 b: 0.00124743 d: 1.02251160
uncorrected_position=(142.00, 267.00, 6.50) : >>> set_bed_level_equation_lsq
current_position=(142.00, 267.00, 6.50) : >>> set_bed_level_equation_lsq
corrected_position=(142.02, 267.01, 5.81) : <<< set_bed_level_equation_lsq
current_position=(142.02, 267.01, 5.81) : sync_plan_position
Bed Level Correction Matrix:
+0.999997 +0.000000 +0.002493
-0.000003 +0.999999 +0.001247
-0.002493 -0.001247 +0.999996
> BEFORE apply_rotation_xyz > stepper_z = 6.50 ... z_tmp  = 5.81
> AFTER apply_rotation_xyz > z_tmp  = 6.60
current_position=(142.02, 267.01, 5.92) : sync_plan_position
current_position=(142.02, 267.01, 5.92) : > corrected Z in G29
<<< gcode_G29
>>> M114
SENDING:M114
X:142.02 Y:267.01 Z:5.92 E:0.00 Count X: 22720 Y:42720 Z:52828
```
</details>

thinkyhead, 2016-07-17:
So, in other words, it is not raising up enough, based on the probe offset. Examining the log……

thinkyhead, 2016-07-19:
@jbrazio Do your position issues seem to be fixed with the latest code, updating the movement in `G28`?

jbrazio, 2016-07-19:
I just arrived home, will now pull latest RCBugfix and update the firmware.

jbrazio, 2016-07-23:
Let's track all G28/G29 on the same issue: #4325

closed 2016-07-23 by jbrazio