MarlinFirmware/Marlin #4316: k8200 with reprap discount full graphics controller

k8200Dave, 2016-07-15:
Hi first post new user the issue I am having is that if I use old firmware my sd card works fine but if I use any new version from 2015 on, The sd card will let me see the files but I cannot print them just returns to the main info screen when connected through pronterface reports sd card ok no errors reported if I send a M21 from pronterface the menu on lcd chages from no sd to print from sd and than all works fine I am using at the moment latest RC BUG fix still same fault

thinkyhead, 2016-07-16:
Do you know what SPI speed setting is being used? Try experimenting with the options:
```cpp
//
// SD CARD: SPI SPEED
//
// Uncomment ONE of the following items to use a slower SPI transfer
// speed. This is usually required if you're getting volume init errors.
//
//#define SPI_SPEED SPI_HALF_SPEED
//#define SPI_SPEED SPI_QUARTER_SPEED
//#define SPI_SPEED SPI_EIGHTH_SPEED
```
Also, compare pins in the old and new firmware. `SD_DETECT` and `SDSS` in particular.

k8200Dave, 2016-07-17:
Thanks for reply I did check SPI_SPEED but cannot remember which version so will try again will also check the pins are correct busy with work so may be a few days but will be back

k8200Dave, 2016-07-17:
sorry hit wrong button

thinkyhead, 2016-07-17:
> sorry hit wrong button

Command-return or control-return guarantees the "submit" button in most textfields these days.

k8200Dave, 2016-07-18:
as far as I can make out SD_DETECT_PIN 49
SDSS pin 25 been a long time since I done any C Code not at home so cannot test SPI_SPEED

k8200Dave, 2016-07-18:
Have tried SPI_SIXTEENTH_SPEED and all other SPI Speeds but  still not working is there some way to send m21 in the code as a workaround

thinkyhead, 2016-07-19:
We've made some adjustments to the SD code in recent days. Can you try the `RCBugFix` branch and see if it addresses the problem?

Since you can send M21 and get the SD card to read, I'm assuming your settings are fine. But it sounds like the SD card is possibly not getting properly initialized for reading.

As part of trying to diagnose SD card issues, I have to ask you to back up your SD card and do a low-level format on it, then put your files back onto the card. By doing a low-level format we can eliminate the possibility of any data corruption. And a low-level format has been known to fix SD reading issues for some users.

To further diagnose the issue, make sure your Pronterface is connected, and then when you use the LCD controller to (try to) start a print, Marlin should output a message like:
```
enqueuing "M23 topcat~1.gco"
Now fresh file: topcat~1.gco
File opened: topcat~1.gco Size: 12362
File selected
```
If it doesn't say any of that, it might say:
```
open failed, File: topcat~1.gco
```

k8200Dave, 2016-07-21:
Thanks thinkyhead busy next couple of days so will write back when I have tested new code

k8200Dave, 2016-07-26:
Just a quick update tried the latest bug fix  still has same problem with sd card, I got a different 2gb sd card formatted FAT 32 and put 1 file on the bender head from thingiverse  this is the output from pronterface when I was using LCD to print
Marlin 1.1.0-RC6
echo: Last Updated: 2016-04-24 12:00 | Author: (none, default config)
Compiled: Jul 26 2016
echo: Free Memory: 2732  PlannerBufferBytes: 1232
echo:Hardcoded Default Settings Loaded
echo:Steps per unit:
echo:  M92 X64.25 Y64.25 Z3200.00 E600.00
echo:Maximum feedrates (mm/s):
echo:  M203 X500.00 Y500.00 Z5.00 E500.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X3000 Y3000 Z100 E10000
echo:Accelerations: P=printing, R=retract and T=travel
echo:  M204 P1000.00 R1000.00 T1000.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
echo:Home offset (mm):
echo:  M206 X0.00 Y0.00 Z0.00
echo:Material heatup parameters:
echo:  M145 S0 H180 B70 F0
echo:  M145 S1 H240 B110 F0
echo:PID settings:
echo:  M301 P24.29 I1.58 D93.51 C100.00 L20
echo:  M304 P341.88 I25.32 D1153.89
echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0
echo:Z-Probe Offset (mm):
echo:  M851 Z0.00
echo:SD card ok
echo:enqueueing "M23 bender~1.gco"
echo:enqueueing "M24"

and it just sat there doing nothing for 5 mins so I reset if I send m21 from pronterface can print from sd card fine,  also the screen timing bug was back so I changed the #if F_CPU == 2000000 to #if F_CPU == 16000000  screen ok now (ultralcd_st7920_u8glib_rrd.h)

hope this helps

thinkyhead, 2016-07-27:
@k8200Dave Is your SD Card lock tab in the LOCK position? Try unlocking it.

k8200Dave, 2016-07-28:
Hi Thinkyhead yes both cards are write enabled was one of first things I checked is there away to do m21 from LCD old version use to have init card on menu thanks again for help

thinkyhead, 2016-07-28:
@k8200Dave If you look at `ultralcd.cpp` and find a command like `G28` it should be obvious how to add a menu item that does `M21`. Meanwhile, I will put together a branch with added debugging so we can try to solve this issue.

thinkyhead, 2016-07-28:
Incidentally, what is your `MOTHERBOARD` setting? Probably we need to more properly adjust the `CPU_ST7920_DELAY_n` values in `ultralcd_st7920_u8glib_rrd.h` for your board.

@Blue-Marlin would you concur that some of these delays might still need adjustment?

thinkyhead, 2016-07-28:
Did you ever do a "low level format" on any of your SD cards?

Blue-Marlin, 2016-07-29:
@thinkyhead 
That's absolutely possible. I remember that someone refused to cooperate - and we have not had feedback from all boards. Some boards seem to have additional capacity/inductivities what could delay the signals. Some users may be mad enough to let their programmers on the ISP-ports. But if any screen was transferred without errors it's very unlikely the following will fail. Resets by timing errors are extremely unlikely because there is __no__ feedback by the displays.

k8200Dave, 2016-07-30:
Hi Thinkyhead yes did low level format of 1 card with hdllf 4.40 so when I have time will check out suggestions above again as always thanks for all your help

k8200Dave, 2016-07-31:
Motherboard setting BOARD_K8200

k8200Dave, 2016-07-31:
Which is number 78 I think

thinkyhead, 2016-08-01:
In the file `ultralcd_st7920_u8glib_rrd.h` you will find the following block:
```cpp
#elif (MOTHERBOARD == BOARD_3DRAG) || (MOTHERBOARD == BOARD_K8200) || (MOTHERBOARD == BOARD_K8400)
  #define CPU_ST7920_DELAY_1 DELAY_0_NOP
  #define CPU_ST7920_DELAY_2 DELAY_3_NOP
  #define CPU_ST7920_DELAY_3 DELAY_0_NOP
```
These delays are fundamental to the bit-by-bit communication to the display. Try editing the delay values here in various ways and see if you can find a combination that makes a difference.

On another note, I recently discovered that my character-based LCD controller has the wrong pin connected for `SD_DETECT`. So in my case, it causes any _locked_ SD card to appear as though it had been removed. As with your issue, `M21` would allow me to use the card.

While trying to track down the problem with my LCD, I added logging to the sd card bit to track down what was going on. From that I was able to come up with a workaround. You might try my branch where I added this workaround to see if it also works on your setup. https://github.com/thinkyhead/Marlin/tree/rc_allow_locked_sdcard

k8200Dave, 2016-08-04:
Just uploaded code from your branch lcd working correct not messed around with settings no time at the moment, how do I see logging data is there a command to see it or enable it in code will get back to u as soon as I can work going crazy at the moment thanks dave

thinkyhead, 2016-08-06:
@k8200Dave For logging of the SD Detected bit, I would need to add it back into the code. If you find that my branch more-or-less solves the SD Card Detected issue, then you may have a similar issue to my own. Anyway, I will add the logging back in and let you know when the branch has been updated.

thinkyhead, 2016-08-07:
I added logging of the SD Detect Pin in my branch. You will need to issue `M111 S2` to enable logging for this item.

Roxy-3D, 2016-08-07:
> I added logging of the SD Detect Pin in my branch. 

Just as an FYI, we probably need to expand the size of marlin_debug_flags.  I've already grabbed the last few bits in the byte.  I suggest we make it an uint16_t.    
~~~~
inline void gcode_M111() {
  marlin_debug_flags = code_seen('S') ? code_value_ushort() : (uint8_t) DEBUG_NONE;

  const static char str_debug_1[] PROGMEM = MSG_DEBUG_ECHO;
  const static char str_debug_2[] PROGMEM = MSG_DEBUG_INFO;
  const static char str_debug_4[] PROGMEM = MSG_DEBUG_ERRORS;
  const static char str_debug_8[] PROGMEM = MSG_DEBUG_DRYRUN;
  const static char str_debug_16[] PROGMEM = MSG_DEBUG_COMMUNICATION;
  #if ENABLED(DEBUG_LEVELING_FEATURE)
    const static char str_debug_32[] PROGMEM = MSG_DEBUG_LEVELING;
    const static char str_debug_64[] PROGMEM = MSG_DEBUG_MESH_SEGMENTS;
    const static char str_debug_128[] PROGMEM = MSG_DEBUG_MESH_ADJUST;
  #endif
~~~~

thinkyhead, 2016-08-07:
The debug leveling stuff was added more for developers than for users, and it has helped in the development of that feature. The hope is that it will be able to go away, and not clutter up the bits. I would recommend that we don't keep adding more debug flags in this manner.

k8200Dave, 2016-08-10:
Thanks thinkyhead will try as soon as I can

boelle, 2016-08-14:
@k8200Dave had a chance to try it?

k8200Dave, 2016-08-18:
Hi not had chance to try yet put old version back on as I needed to use printer to print some parts will try soon thanks for all your help

boelle, 2016-08-26:
@k8200Dave had a new time / chance to test it?

k8200Dave, 2016-08-29:
Sorry not yet but will when I get time I still check forum

boelle, 2016-09-03:
@k8200Dave i guess yoou are still busy?

boelle, 2016-09-10:
@thinkyhead @k8200Dave 

given that got a K8200 tossed at me for free i wonder how i can help with this one? still rebuilding it as its a pile of junk as build pr velleman instructions

one of the biggest upgrades is the Z axis... i have done this one from thingiverse http://www.thingiverse.com/thing:1685126

but i also need to figure the bed as you simply cant bed it flat.... either the bed frame from a prusa i3 and a mk2a heat plate.... or just a glass sheet on top 

but a full graph display is on  the shopping list... just not sure when it will happen as funds are always limited

SwiftNick, 2016-09-15:
I had the same problem on the K8400, the SD card listed fine but when you select a file to print, nothing happens.

The card detect on REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER uses the SS signal which is Arduino Pin 53 (Physical pin 19 on the ATMEGA2560 chip)

Try adding the following SD_DETECT entry to the end of your k8200.h file
```
#undef  SD_DETECT_PIN  
#define SD_DETECT_PIN  53
```

thinkyhead, 2016-09-16:
Oh, SS! Well, actually there is an SDSS pin, ya know. In the pins_3DRAG.h file you will find this line which specifically mentions pin 53:
```cpp
//
// SD Card
//
#undef SDSS
#define SDSS               25//53
```
So, under which circumstances should it be 25, and when should it be 53?

thinkyhead, 2016-09-16:
Try this patch instead. Edit the file `pins_3DRAG.h` and replace the lines shown above with:
```cpp
//
// SD Card
//
#if MB(3DRAG)
  #undef SDSS
  #define SDSS             25//53
#endif
```

SwiftNick, 2016-09-16:
On the 3DRAG/k8200  and 3DualDrag/K8400  the SS Pin is Arduino Pin 53 (ATMEGA2560 pin 19),
so the underlying SS pin definition should be 53.

But which variable is the SDCard code using for card detection?
SD_DETECT_PIN, SDSS, SS_PIN, SS?
One of these gets overridden based on display type, and I think that's causing the issue, as the #ifdef tree isn't complete in 1.10

SwiftNick, 2016-09-16:
Somewhere in the code, SDSS , SD_DETECT_PIN & SS are getting confused.
The definition for SDSS is correct as pin 25 (PA3).  
But this is not the SS signal that is on pin 53.

From the table below for the full graphic Smart controller display, the SD card chip select is connected to SDSS, the SD card detect is connected to SS.

I think somewhere SDSS has been used instead of SS or SD_DETECT.
It's a bit of a mess!!

[http://www.open-electronics.org/full-graphic-smart-controller-display-for-3drag-3dprinter/](url)

PA3 is Arduino pin 25 (chip pin 75)
SS is Arduino pin 53 (chip pin 19)
EXP1 & EXP2 are the connectors on the display board.

[image]

k8200Dave, 2016-09-16:
Thanks for info guys I am using magu version and all is fine with SD card will be busy at work till end of month so should have time to mess around having all the pins in one file with #ifdef is hard to follow sometimes but thanks to all who make marlin the best

k8200Dave, 2016-09-16:
@boelle

Nice upgrade on z axis

thinkyhead, 2016-09-17:
> But which variable is the SDCard code using for card detection?
Somewhere in the code, SDSS , SD_DETECT_PIN & SS are getting confused.

Take a look at `CardReader::initsd`. Does this make sense from your understanding of these pins?
```cpp
void CardReader::initsd() {
  cardOK = false;
  if (root.isOpen()) root.close();

  #ifndef SPI_SPEED
    #define SPI_SPEED SPI_FULL_SPEED
  #endif

  if (!card.init(SPI_SPEED,SDSS)
    #if defined(LCD_SDSS) && (LCD_SDSS != SDSS)
      && !card.init(SPI_SPEED, LCD_SDSS)
    #endif
  ) {
    //if (!card.init(SPI_HALF_SPEED,SDSS))
    SERIAL_ECHO_START;
    SERIAL_ECHOLNPGM(MSG_SD_INIT_FAIL);
  }
  else if (!volume.init(&card)) {
    SERIAL_ERROR_START;
    SERIAL_ERRORLNPGM(MSG_SD_VOL_INIT_FAIL);
  }
  else if (!root.openRoot(&volume)) {
    SERIAL_ERROR_START;
    SERIAL_ERRORLNPGM(MSG_SD_OPENROOT_FAIL);
  }
  else {
    cardOK = true;
    SERIAL_ECHO_START;
    SERIAL_ECHOLNPGM(MSG_SD_CARD_OK);
  }
  workDir = root;
  curDir = &root;
}
```
And then the `SD_DETECT_PIN` is used dynamically to see when the state changes:
```
#if PIN_EXISTS(SD_DETECT)
  #if ENABLED(SD_DETECT_INVERTED)
    #define IS_SD_INSERTED (READ(SD_DETECT_PIN) != 0)
  #else
    #define IS_SD_INSERTED (READ(SD_DETECT_PIN) == 0)
  #endif
#else
  //No card detect line? Assume the card is inserted.
  #define IS_SD_INSERTED true
#endif
```

SwiftNick, 2016-09-18:
I've looked at the cardreader code for Marlin 1.0.2 and 1.10 RC7 and the initsd function is pretty much the same. As the SD card has worked in previous versions and the code hasn't changed it should still be ok. As the code uses LCD_SDSS I defined that in k8400.h.  It didn't work. 
The SD card menu is present and I can list the gcode files, but can't execute them. So I think that code is fine as it is. 

What should happen, is when there is no SD card detection signal, the code assumes the card is present.

In Cardreader.h 
Marlin 1.0.2 there is a test for the SDDETECT definition (-1 is disabled)
```
#if (SDCARDDETECT > -1)
# ifdef SDCARDDETECTINVERTED 
#  define IS_SD_INSERTED (READ(SDCARDDETECT)!=0)
# else
#  define IS_SD_INSERTED (READ(SDCARDDETECT)==0)
# endif //SDCARDTETECTINVERTED
#else
//If we don't have a card detect line, aways asume the card is inserted
# define IS_SD_INSERTED true
#endif
```
 in 1.10 RC7 there is 
```
#if PIN_EXISTS(SD_DETECT)
  #if ENABLED(SD_DETECT_INVERTED)
    #define IS_SD_INSERTED (READ(SD_DETECT_PIN) != 0)
  #else
    #define IS_SD_INSERTED (READ(SD_DETECT_PIN) == 0)
  #endif
#else
  //No card detect line? Assume the card is inserted.
  #define IS_SD_INSERTED true
#endif
```
I can't find any definition of SD_DETECT in the code...  A bit strange!

I have examined the schematics for the 3DRAG(K8200) and the 3DUALDRAG(K8400),
and the displays in the SMART_CONTROLLER range that support SD cards and they all share the same physical connection to the ATMEGA.  Arduino Pin 53 is used for SD card detection (insert/removal).

As the K8200 & K8400 boards are derived from the 3DRAG, the SD_DETECT_PIN definition should go at the top of  pins_3DRAG.h just under the SDSS defines. Line 51
```
#undef SDSS
#define SDSS               25//53

#undef SD_DETECT_PIN
#define SD_DETECT_PIN 53
```
That way it will be redefined by other display controllers should they need it.

I think some work needs to be done when there is no SD detect signal as currently with no signal the SD card can be browsed, but not printed from.

thinkyhead, 2016-09-18:
If an SD card reader has a proper `SD_DETECT_PIN` and that pin is defined, then the code uses that pin to determine the presence or absence of a card.

If the pin is not defined, or defined as -1, the code assumes the card is present.

If the card is slow or screwy and cannot be read soon after insertion, the code assumes there is no card, and you have to use `M21` to "wake it up."

Some controllers accidentally connect the SD Writeable pin to the SD detect, and so if a card's tab is in the LOCK position, the card is detected for a moment, and then the detect pin goes LOW after a moment. So you have to use `M21` to wake it up in that case also.

thinkyhead, 2016-09-18:
> As the K8200 & K8400 boards are derived from the 3DRAG, the `SD_DETECT_PIN` definition should go at the top of pins_3DRAG.h just under the SDSS defines. Line 51

I guess this is just one of those unfortunate choices made just to be different. Do you have a PR ready to go, or should I just throw one together now?

closed 2016-09-19 by thinkyhead

thinkyhead, 2016-09-19:
Patched directly in `RCBugFix`. Who needs PRs anyway?