MarlinFirmware/Marlin #4317: Duplicate Printing with Dual Extruder on Single X Carriage

jstefanop, 2016-07-15:
I know marlin has Dual X Carriage support, but would be nice if there was a way to set both extruders to active on a dual setup (two extruders attached to same X Carriage). This way you can print two small parts at the same time (that are less wide than the nozzle offsets of course). 

This should be a rather simple addition if its not already possible with a commination of M commands (new M command that sets both extruders to active, and sends extrusion commands to both stepper drivers?)

thinkyhead, 2016-07-16:
Marlin has only a single E axis at this time, so it's not possible to run two extruders with separate objects.

thinkyhead, 2016-07-16:
On second thought, this is an impossible idea. The Y axis (often the moving bed) is shared.

jstefanop, 2016-07-16:
@thinkyhead I mean two identical objects at the same time. Printing two different objects would obviously be impossible like you mentioned. The whole point is to utilize the second extruder on small parts to essentially build double the number of parts at the same time. 

Firmware wise, all this would require is to send the same E commands to the second extruder at the same time. Currently the active extruder is set via T0, T1 etc. It would be great if there was an M command that would set BOTH extruders active, and the second extruder simply feed off the same signal as the first. This would involve the exact same method that dual Z steppers work(i.e. the same Z commands are sent to both Z steppers at the same time for printers that have two steppers to drive the Z axis).

brainscan, 2016-07-16:
Just split the signal like when using two z motors.

jbrazio, 2016-07-16:
Indeed, for "duplication" it could be a matter of running two steppers from the same driver.
In the end the critical path is mostly slicer object placement "magic" rather than Marlin's printing.

jstefanop, 2016-07-16:
Well double Z motors are driven by two separate drivers on the board, the same singles are just sent to both stepper drivers at the same time in the firmware. Two E steppers are also plugged into two stepper drivers, just need the same behavior in the firmware, with an M command to disable/enable, since it would not make sense to keep flashing different firmware just to enable or disable this feature.

brainscan, 2016-07-17:
You can also run two steppers on one driver.

jstefanop, 2016-07-17:
How? With a Y cable? Wouldn't that exceed the current limits of the driver? Either way if that was possible its still not as clean as a firmware addition. Since you would still need to unplug/replug wires for normal operation.

brainscan, 2016-07-17:
Have you never seen ramps board? It's actually a pretty easy thing you can set up without firmware change.

oysteinkrog, 2016-07-17:
This is already supported through the dual x carriage config option I think.

thinkyhead, 2016-07-17:
> This is already supported

Correct. This is already supported, which is why the request confused me.

jstefanop, 2016-07-18:
I think there is still confusion in what Im talking about...I don't have a dual x carriage...i just have two extruders on a single carriage, that I want the same extrusion commands to be sent to both at the same time. 

The whole point is to have an M option so you wouldn't need to flash firmware each time anyone would want this enabled. Either way, how exacty would sending the same signal to extruder #2 work via whats already supported?

oysteinkrog, 2016-07-18:
I see, that might be a little bit useful, but it's also very easily handled by the slicer?

jstefanop, 2016-07-18:
The only thing that can be done via slicer/gcode currently is switch which extruder is currently active, so that wouldn't help since you can't set both extruders active at the same time with T0, T1 commands.

jstefanop, 2016-07-18:
Here is another reference to this...this feature is actually part of Sailfish firmware for Makerbot clones and is called "Ditto Printing"

https://forum.simplify3d.com/viewtopic.php?f=8&t=1617

brainscan, 2016-07-18:
How would you use the same X carriage for normal printing? You say it would be as simple as an m code to switch but that would mean permanently cutting down the available print area, or am I missing something. People have done this for a long time with milling machines just piggybacking the signals where needed.

jstefanop, 2016-07-18:
This is strictly for mass producing small identical parts, it essentially cuts down the time in half. You place down a single part on the slicer, and make sure the next placement on the X axis is offset by the two nozzle separation distance + some padding. 

So as far the the slicer is concerned there are only half the amount of parts placed on the bed, but because you duplicate the same extruder signal to extruder #2, you build two of the same parts at the same time. 

I print multiple copies of the same small part all the time, but the second extruder usually sits idle (unless its printing support material or two color stuff). This would vastly increase the utility of dual extruder setups, and its a pretty common feature request if you do a google search. 

Firmware wise can anyone with a dual extruder setup (NOT Dual X carriage) confirm whether enabling dual X carriage and then sending M605 S2 will do what I'm asking?

@thinkyhead is that what you mean by its supported?

thinkyhead, 2016-07-20:
> whether enabling dual X carriage and then sending `M605 S2` will do what I'm asking?
```
Mode 2: Duplication mode. The firmware will transparently make the second x-carriage and extruder copy all actions of the first x-carriage. This allows the printer to print 2 arbitrary items at once. (2nd extruder x offset and temp offset are set using: M605 S2 [Xnnn] [Rmmm])
```
This is supported for a printer that has two carriages on the same X axis, so they can each move independently in X, while Y and Z movement affects them both. I don't know of a machine that has multiple nozzles on an axis other than X.

jstefanop, 2016-07-20:
@thinkyhead nozzles are 60mm apart, so its enough distance to double print many parts (the only limiting factor is 60mm X width...Y can be as long as the build plate and this will work). 

Im assuming implementing this would be rather easy codebase wise? However the firmware handles double Z motors, it can apply the same logic to double E motors and an M toggle to enable/disable the feature?

Also why wouldn't the Duplication mode with M605 S2 not work? Obviously the printer does not have a second carriage so any commands to the second X motor will be ignored, but the second extruder should work no?

jbrazio, 2016-07-20:
KISS.. Connect a Y cable to from E0 to both extruders.

thinkyhead, 2016-07-20:
> why wouldn't the duplication mode with M605 S2 not work?

It's just not supported for anything but Dual X Carriage. If you want the second extruder to synchronize with the first in other situations, it would have to be coded as a new feature. It would be a simple feature to implement. But how many users have their nozzles 6cm apart? Not too many. And Marlin will not help you if you try to print two objects larger than your setup can accommodate. You would have to tell your slicer not to start printing the object in the middle of the bed.

jstefanop, 2016-07-20:
@thinkyhead right I understand its not supported, but in theory if I enable Dual X Carriage and send that command, wont the second extruder activate and copy the commands for the first? Is that how M605 S2 is currently coded?

Since I don't have a second X carriage any other commands going to the second X motor will just be ignored? Or will the firmware crash since I have a RAMBO board which does not have a second X motor plug?

In terms of the feature request, its pretty common...there is actually another similar request for marlin (#2783), and a google search will bring up people asking for this feature in multiple places. 

Im sure most people with dual extruder setups don't realize this is even possible, and would use the feature if implemented. If its rather simple to implement I think it would be a nice addition to marlin and add more utility for people with dual extruders.

thinkyhead, 2016-07-20:
You won't even have an `M605` command unless you enable Dual X Carriage. And you also won't have any code that checks for dual extruder mode. I would need to be implemented as an extension to the feature.

jbrazio, 2016-07-20:
**KISS.. Connect a Y cable to from E0 to both extruders.**

jstefanop, 2016-07-20:
@jbrazio yes I know there are hardware hacks to do what I'm asking, but its inconvenient to need to take the printer apart and connect/disconnect Y cables every time I want to switch from Duplicate printing mode back to normal dual extruder for printing in dual color/support material. A simple M command to enable this would be awesome.

thinkyhead, 2016-07-20:
@jstefanop #4362 demonstrates how the implementation would have to be.

jstefanop, 2016-07-20:
@thinkyhead awesome! Ill test this out.

SlickNickeL, 2018-07-12:
REPETIER has a "Ditto" setting by which 4 extruders can be simultaneously used to print 4 prints, on one X carriage. The benefits of this setting, is the ability to mass produce and manufacture many parts at one time.

Can Marlin consider expanding the current configuration to at least go to 3 extruders, from the current 2 allowed extruders? This would be a great benefit to any company who wants to manufacture single-run continuous filament productions. Strength, speed, appearance and accuracy are greatly improved by keeping one print to one head from start to finish. Size is limited to the distance of each extruder based on the X carriage spacing, which isn't a problem for small pieces. Since most 3D printers made to perform such tasks, have one purpose and have identically matched hotends/motors, the data to all extruders can be cloned without a significant amount of problems in the prints by simply assigning/cloning data from EO to E1, E2 etc.  Problems arise when motors are wired in parallel from a single stepper.

photodude, 2019-12-31:
For those looking, Ditto printing was done in the Shaqfoo fork of Marlin 1.0.0
https://github.com/ShaqFoo/Marlin_Firmware_Deluxe_With_Ditto_Print-Auto_Leveling

Note: it's crazy old Marlin 1.0.0 and Dual X Carriage support has been stripped out completely to implement Ditto printing. (the stepper.h file would be a mess to implement both the Dual X Carriage support and the ditto printing on a single X carriage... but everything else would be fairly straight forward)

SlickNickeL, 2019-12-31:
It only allows 1 additional copy  for a total of 2. I asked for additional copies for 3 or more print heads.
My standard manufacturing printer prints 3 identical pieces at one time like (STACKER) does, except we use REPETIER  to do this since Marlin didn’t have the additional extruder selection. I do believe Marlin is a better program but at this point in our business, we have to stay the course.
Paul

photodude, 2020-01-01:
@SlickNickeL True. But if someone wanted to extend the concept that ShaqFoo put forward to more nozzles it might not be too hard. going from 2 nozzles ditto printing to 3 nozzles is likely relatively easy, but going from 2 nozzles to N number of nozzles [2, 3, 4, 5, 6] would likely be a more difficult task.

Just my quick impression of the ShaqFoo changes I think the ditto printing is mostly controlled by the `stepper.h` setup for ditto printing. The stepper.h `defines` look like they tell each nozzle extruder to do what the E0 extruder is doing. The rest of the changes look like the user setup for the LCD interface to activate ditto printing. Not totally sure if that's how it works, this is just my initial impressions. 
https://github.com/ShaqFoo/Marlin_Firmware_Deluxe_With_Ditto_Print-Auto_Leveling/blob/master/Marlin/stepper.h

SlickNickeL, 2020-01-01:
Supposedly, Marlin went ahead and implemented my proposal, I just haven’t had the time to get it into a machine. 
I haven’t seen any users commenting, so I don’t know about the usability.
We currently have 22 machines that use 3 print jets at one time. We are using Repetier and it’s working ok, but any attempt at understanding advanced bed leveling functions is limited due to the language barrier with tech support.
I would love to hear of some success of the Extended Ditto printing before putting it on a machine.
Marlin’s auto level and documentation are superior to Repetier. Not to say that auto-level is even possible on a three head printer due to (3 points over/X=mm span to tram) is even doable at this point. We know of the caveats that are involved, and we are willing to engineer a solution, it’s just that Repetier doesn’t speak English well at all, and Marlin doesn’t have a selectable 3 head Ditto Mode that we can trust to use in production at this moment. Maybe someone can provide some feedback?

AnHardt, 2020-01-02:
An other weak concept.
For printing with more than one nozzle (at the same time) and automatic bed leveling, there is no working concept.

On a bumpy, not only tilt bed, it's obvious. One of the nozzles could be on a hill and the other could be over a valley. There is no way both of the nozzles can be a the right distance at the same time.
For a tilt only bed it's not that obvious. To have the same bed-nozzle distance for all of the nozzles, the tilt of the plane thru the nozzle tips has to be the same as the the tilt of the bed. Because we can't automatically determine  the bed-nozzle-offset for all nozzles we can't find out how the tilt of the bed should look like. There is only one exception - when the bed-nozzle-offset is exactly the same for all nozzles and the bed is exactly horizontal. But here leveling the bed is the simple part and leveling the nozzles the time consuming one.
For cases where the horizontal distance of the nozzles varies (mirror mode) the needed tilt angle of the bed changes with the distance (if not zero).

Automatic gantry leveling does only solve half of the problem (if the nozzles are collinear - so not for a kraken-hotend) if at all. If the reference is the frame, it depends on the squareness of the frame. If the reference is the bed it does nothing what couldn't be done by bedleveling. In no case we get the really important information about the different heigth-offsets of the nozzles.

InsanityAutomation, 2020-01-02:
> For those looking, Ditto printing was done in the Shaqfoo fork of Marlin 1.0.0
> https://github.com/ShaqFoo/Marlin_Firmware_Deluxe_With_Ditto_Print-Auto_Leveling
>
> Note: it's crazy old Marlin 1.0.0 and [...]

The current branch supports ditto printing with up to 6 extruders. @thinkyhead and myself finished that before MRRF last year....

// Add a Duplicate option for well-separated conjoined nozzles
//#define MULTI_NOZZLE_DUPLICATION

InsanityAutomation, 2020-01-02:
@AnHardt Correct, with 3/4 point gantry leveling you can get a repeatable plane but still need an exceptionally flat surface (MIC6 plate) and nozzles set very carefully. I wish more manufacturers used jack screws on hotend carriages with fine thread bolts.... Thats part of why I like the Slice copperhead design, it allows some very simple adjustment.

SlickNickeL, 2020-01-02:
All of what you are saying is correct. 
With over two years of development, our conclusion to leveling is to use the far left nozzle at the far left back side of the build plate with a piece of paper. Then, far right back corner using the far right nozzle and finally the third point, front middle, using the middle nozzle. When all three nozzles grab the paper with the same amount of force, the bed is level.
Anything else is a waste of time.
If the Z min endstop is set correctly, there is no danger of bed crashes, (unless) there is a bulge in the bed.  

Development: What we have been trying to do is use an induction probe to reset the Z height, with a single location reading that will establish the current positions reading as Z=first layer.  
We can’t use auto level for the whole bed due to the afore mentioned problems of 3 heads vs an unleveled bed plane. BUT! We can go to the center of the middle part that is to be printed, and set that point as Z=first layer. Then when that part/process is finished, the head goes to a different part of the build plate and starts a completely deferent part/process and starts with a G=28,29,31 goes to the center of that part and probes a single reading, to reestablish the current Z=first layer. This probe would reset the Z for the new part position. Our bed and X gantry bars are high enough for the parts we are printing that there is no problems with collisions. This was all figured out during the setup of the print file.

Conclusion: what this does is reset the Z at its new printing location for all three heads to offset environmental temperature differences throughout the printing day. Our first layer must be perfect, and temperature variances are the biggest problem.

Problem: We have spoke with Repetier about this, but they can’t clearly explain the process of any of the G codes in clear concise English, in order for us to try this. The problem arises with the way Repetier uses Zmin and Zmax to level the bed. Add in their crazy head offset calculations, and all opposing reference number that are backwards, and all that can be concluded is, (Let’s not due any of this until we understand exactly what is going to happen first) Repetier can’t explain the current condition of any of the G code, so we refuse to attempt this. By doing so, we risk a bed crash, and just one such crash would cause hours of replacing nozzles. 
Marlin doesn’t have the Zmin/Zmax problem with autolevel as far as I can remember. This is the solution to multipart/multiprocess full bed manufacturing. If anyone from Marlin has this answer, we have an open machine currently ready for production that we can try, but only if we can understand the process and are able to enter in a set of starting numbers that are logical, and that will work.

Repetier uses the term “Rotation” for traming/leveling. With a crappy translation and crappy explanations in un-usable instruction sets, the process simply isn’t understandable, to simply apply a solution safely to any G code for auto leveling.

SlickNickeL, 2020-01-02:
Ok, so I need to take a look at this. What about a G code that will do what I just previously posted? let’s say G28 home all, then 1: G1 X40 Y40 ; moves induction head probe to center of part for printing  2: G29? 28? 30? 31? ; probes current location and resets Z for first layer. ( For Repetier, the head would have to be high enough above bed to NOT trigger probe, if so, an error occurs. So there would be a Z10 inserted before the first G1/0)
Fast forward to second part/piece sequence in file 86743: G1 X 230 Y145 ; moves induction head probe to second location, center part, center print head 86744: G29? or 31?; probes bed and resets new locations Z for first layer.

SlickNickeL, 2020-01-02:
Ok!
 I see that Marlin has went to 2.0? And this Ditto printing is in this release?

I have a machine that we will try this on starting today. It is probe ready and a complete 3 head fixed carriage, with all three parts printing at the same time. 

I have allocated 40 production hours to this. Should only take 2 at the most. 

Not sure about the auto-level 

If someone could tell me about Marlins G code for resetting Z in between prints, with parameters, that would be great, if not, it my take us some time to figure it out. Worried about a probe setting that bypasses Zmin safety and head crash. REPETIER DOES DO THIS AND WILL CRASH A BED IN A HEARTBEAT. It overrides Zmin on some of the configurations. We will need to figure this out before proceeding.

photodude, 2020-01-02:
> The current branch supports ditto printing with up to 6 extruders. @thinkyhead and myself finished that before MRRF last year...
> 
> // Add a Duplicate option for well-separated conjoined nozzles
> //#define MULTI_NOZZLE_DUPLICATION

@InsanityAutomation Awesome, thanks for the information. Hopefully, I can get my printer's manufacturer to officially drop the crazy old ShaqFoo fork and move to Marlin 2.0.

I tried to look up the MULTI_NOZZLE_DUPLICATION documentation but didn't find anything on the marlin website search
http://marlinfw.org/meta/search/?q=MULTI_NOZZLE_DUPLICATION

the marlin website does have the old DUAL_NOZZLE_DUPLICATION_MODE documentation for M605 which it looks like the MULTI_NOZZLE_DUPLICATION replaced http://marlinfw.org/docs/gcode/M605.html

It would be nice if there was better documentation for this feature. Still awesome that it's there.

SlickNickeL, 2020-01-02:
It’s under Configuration_adv.h as far as enabling. 

You may want want to separately PID each hot end. It’s under config.h #define PID_PARAMS_PER_HOTEND. Since we have two cooling fans facing in and center head is constrained, the temps for each hot end are really off by quite a bit for PID.

SlickNickeL, 2020-01-08:
We have finished the testing.

We were not able to engage E2(Third Extruder) using M605 S2 P7 :Multi Nozzle Duplication on (012)

We tried P15 as well with no results.

We could not get The third Extruder to be addressed under any setting.

This was how the old Marlin worked. It could make one copy but that was it. Same is true now, from our testing. 

M605 S2 P7 only engages E0 and E1 and it should be 0,1,2

What are we missing?

For clarity of use of ease, the ability to stop or exclude an extruder at this point should NOT be considered. 

First state of M605 S2 should be E0 and E1 for a total of two prints. Second state of M605 S2 should be E0,E1,E2 for a total of three prints.
Third state E0,E1,E2,E3 for a total of four prints, and so on.
All cloning of code should always emanate from E0

SlickNickeL, 2020-07-03:
What a shame.

Repetier has had this issue solved for many years now, and Marlin refuses to address this milestone.  
There is no way Marlin can be competitive in the future without addressing its capabilities to multi-print Duplication/Ditto.
This single action will keep Marlin out of the commercial duplication business.
Marlin has now committed its resources to be just a firmware for hobbyists and frivolous one-off trinket production machines. 

Prove me wrong!

Sadly, I opened this thread quite some time ago in hopes that a solution would have been found by now.  We are now committed to our current situation that has its limitations. 
Marlin actually has all the precursors for a very fine production run, multi part, auto-level manufacturing firmware. Repetier does not, and will not have multi head auto-level capabilities due to its inability to address the Zmax/ M28-32 conundrum. This is not a viable and efficient way to tram a bed, not to mention the wear and tear per cycle.

Who can I throw some money at, to solve this problem?  All current players who were originally involved in delaying or ignoring this issue will be exempt from this offer.

thinkyhead, 2020-07-03:
@SlickNickeL — There might be some talented people who have time and space to help you over at Discord and at these other forums. Please try to remember that everyone who works on this project is just a hobbyist who works on the project for fun in their spare time. If you cannot be diplomatic and patient with these people under the most difficult of times then no one will care to help you.

- Marlin Discord server. Join link: https://discord.gg/n5NJ59y
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)

Roxy-3D, 2020-07-03:
@SlickNickeL    

> Who can I throw some money at, to solve this problem?  All current players who were originally involved in delaying or ignoring this issue will be exempt from this offer.

You don't need to throw any money at this problem.    I'll do what ever it takes to get this (and other features you want) working.   But I don't want to code against a 'mythical machine'.    I will need a production quality machine to write the code and to test against.

SlickNickeL, 2020-07-03:
Mythical? Lol
How about magical?
This is a three head Bowden that prints out three separate and identical parts at one time. We have the wiring ready for Marlin auto-level but there is no sense in hooking it up until this problem is solved. By changing the carriage size we can add more heads and at different intervals. This is the same as a STACKER, we use the same Repetier firmware, but we have tweaked it a bit.
STACKER has some serious first layer mechanical adjustment considerations that pose a nightmare for a push-button-walkaway.
Currently we use Repetier with manual calibration. Works like a charm, EXCEPT temperature fluctuations throughout the day causes the first layer to be less than perfect at times. A simple auto level, when doing a second set of prints in the same run prior to laying down the first layer, would account for expansion and contraction due to temperature differences. Repetier can’t do this. It uses its zero from Zmax. The bed has to go all the way back to Zmax before probing. Marlin CAN do this if Ditto/Duplication is working correctly. Mind you, the mesh settings and specific areas for probing are not to be considered due to possible head crashing, but a single probe to reset zero on top of the New area to be printed does work manually.  Mesh is strictly a no-no when spreading the printheads over an expansive bed. Even with Alca5 aluminum, you are taking a chance if the mesh had an inaccuracy during probing. 

I will send some pictures if you would like. Apparently this server doesn’t allow pictures.

Roxy-3D, 2020-07-03:
> Mythical? Lol
> How about magical?

Mythical as in I don't have one to see and understand.   I don't have one to write code against and test that code for proper operation.   Perhaps that was a bad term to use.
I would like to see some pictures.

SlickNickeL, 2020-07-03:
Send me your email. I’m out of town until Thursday. 
I will need to hide some parts of the machine because they are proprietary for the part it makes.  We decided to skip the Patient on the part and keep the process secret.

Ditto/Duplication mode really is a well established process already in Repetier. 

See https://stacker3d.com/.

Roxy-3D, 2020-07-03:
Are you on Discord?   (I don't see you there.)   That would be a simple way to send you contact information.  
Alternatively we can exchange contact information on www.3dprintboard.com.    Or pretty much any way that works for you.

Roxy-3D, 2020-08-05:
@SlickNickeL     

Do you have everything you need?   
Would some firmware help benefit you?

SlickNickeL, 2020-08-06:
I originally set aside a machine for testing Marlin Duplicate printing mode.

We needed three prints to be printed on the same gantry at the same time keeping each print head separate and adjustable for temp, extrusion multiplier, feed rate through the slicer. Marlin can only do 2 prints at one time. I then submitted this request to Marlin for the change from 2 prints to 3 prints simultaneously. This was already being done by Repetier, and for several years now. Only problem with Repetier is, we were working on an auto level capability with our three head printer, and Repetier uses Zmax for zero to start autolevel. This is a big waste of time. Marlin’s auto level would be compatible with a three head system.

An effort was made to get this code working. The code was put in place, and I tested it. It didn’t work. Only two heads could be engaged at one time. This is all documented, and I reached out to various people to continue this, but everyone was busy.

I provided plenty of professional quality feedback but nothing more has progressed since then.

I’m giving lots of lip service, but Marlin has yet to provide a working model.

Unfortunately, my startup company has to move forward with Repetier at the helm. 

The ability to push a button, and 3 completed units consisting of two parts apiece, are perfectly printed 2 hours later. This was the goal, and we now have finished this phase and working towards marketing. I personally would love to see Marlin firmware performing this function, but we don’t have the time to sit and wait for someone to work on this project.

It will take an enormous amount of time and energy to get Marlin to the point of final production on our part. It’s a very complicated part with a very simplistic but practical application.  18 separate and unique slicer setting, takes “tweaking settings” to a whole new level. I can’t imagine doing this all over again for Marlin.

I currently know of a company that needs 1000, 2 piece units by mid September.  Ditto/Duplication mode would be the way to print these small pieces. I drew the part and sent them the sample. Our printer would work, but our ROI is so high on our invention, that we can’t spare the time or machines to do the run.  In order for them to bring this part in at a reasonable cost, they will need their own production units.  

Manufacturing small parts with 3 print heads is a viable business model. Our machines are proprietary to the part we make, do exist, and are working to a commercial level of reliance. We simply don’t sell them, or allow any of the 20+ original advancements in our printing technologies available to the public.  Saves on Patent applications. Without these advancements, the part we produce, couldn’t be printed. This way, we have the full confidence in our manufacturing ability, to protect the product we make, from general reproduction by the public. It simply wouldn’t be worth the time to attempt to reproduce, at the quality that we have achieved. This doesn’t mean that it won’t happen, but there will be caveats that will discourage most attempts at reproduction.

I’m out. Two years of beating a dead horse has gone on long enough. I’ll let someone else beat the dead horse if they want to.

SlickNickeL

InsanityAutomation, 2020-08-06:
I dont blame you. Unfortunately those of us familiar with this area of code are few in number and quite simply currently totally overwhelmed. I havnt touched a thing that wasnt compensated in over 6mo not for lack of interest but lack of time. Its the nature of OSS development. Ive got working being paid for by 2 manufacturers and a distributor at the moment and even that is going at a pace slower than theyd like due to commitments in automotive automation applications that put food on my table.

I do know the issue is valid though and will need to be addressed at some point. There is an issue with the macro expanding the bitwise value.

closed 2020-09-12 by github-actions[bot]

a7madovic, 2022-02-02:
> For those looking, Ditto printing was done in the Shaqfoo fork of Marlin 1.0.0 https://github.com/ShaqFoo/Marlin_Firmware_Deluxe_With_Ditto_Print-Auto_Leveling
>
> Note: it's crazy old Marlin 1.0.0 and [...]

Hi @photodude! Were you able to build the firmware using the files by Shaqfoo? I am trying to use his files but not sure how to include them in the marlin firmware. Thanks

a7madovic, 2022-02-02:
@InsanityAutomation @SlickNickeL @jstefanop I am relatively new to coding and editing the Marlin firmware and trying to use the duplicate printing command with 2 extruders that extrude simultaneously into a single nozzle. In addition, would like to vary the flowrate of one of the polymers relative to the other one both real time and fixed ratios. I tried using the MULTI_NOZZLE_DUPLICATION command in the configuration.adv file and that wasn't working for me. The firmware was not compiling successfully when I build it in visual studio. Has anyone tried the duplicate printing before with Marlin and had success with it?
I know some other hardware hacks but it's currently limited to only identical flowrates for both extruders unfortunately. I will appreciate your input. Thank you!

AnHardt, 2022-02-02:
For two extruders feeding the same nozzle use a MIXING extruder.