MarlinFirmware/Marlin #4318: Feature Request: Probe Averaging, extended version of M48.

GreatGrizzly, 2016-07-15:
I noticed that Marlin contains a  Z Probe repeatability test. However it looks like this is only used for debugging purposes. Why not use this for all probings?

For any Z homing move, probe the same spot an arbitrary number of times, average the result, and return that as the new 0.

Blue-Marlin, 2016-07-15:
> Why not use this for all probings?
Because grid-bed-leveling is already the averaging variate of 3-point levelling.

Double touch probing for deltas is in the queue. That should improve the results of a single probe.

thinkyhead, 2016-07-16:
Continually improving this area, especially at the moment there seems no end of the cleanup we are able to do…

jbrazio, 2016-07-19:
Closing due being marked as duplicate.

closed 2016-07-19 by jbrazio