MarlinFirmware/Marlin #4324: [Feature Request] Use RGB addressable LEDs to indicate print status

slovenian6474, 2016-07-16:
I think it'd be neat to use addressable LEDs, wire it into the Ramps/arduino and be able to program machine style status lights.  For example, they'll turn green when the print is done. Red indicating some kind of issue.  Maybe some preset sequences like slow roll through colors while printing or bright white while printing.

Roxy-3D, 2016-07-16:
You can already do that...    Just wire up the LED's and use M42 to turn them on and off.   You could do much of what you suggested up above by putting the appropriate M42 commands into your Slicer's Start Up and Shut Down GCode.

Incidentally...  I have 3 LED's wired up to help me with debugging some code I'm writing right now.   The code to do things to them is brute force and not shrunk down.   It is more important that it executes fast so it doesn't throw the timing off.  

~~~~
// These are some primitive debug routines that will get stripped out later when the code is solid
// I have some LED's wired up to my RAMPS v1.4 board that let me turn them on and off.  I use them
// for different things as I debugged the code.   I would turn them on at certain points in the code
// or I would light up more or less of them depending upon the value of a variable.   I some times
// used the white one to indicate a + or - value of the Mesh Point being used in the calculations.

// Pin 63  Red   LED
// Pin 65  Red   LED
// Pin 66  White LED
 
void status_LED( int pin, int action)
{
static int pin_63 = -1, pin_65 = -1, pin_66 = -1;

	switch (pin) {
	case 63:	if (pin_63== -1){		// if so, we need to initialize the mode to output
				pin_63 = 0x00;		// toggle the pin's status to simpile On/Off mode
    				pinMode(63, OUTPUT);
			}
			if (action==0) {
       				digitalWrite(63, 0);
				pin_63 = 0;	// remember the pin is turned off
				return;
			}
			if (action==1) {
       				digitalWrite(63, 255);
				pin_63 = 1;	// remember the pin is turned on
				return;
			}
			if (pin_63) { // toggle the pin's status
				pin_63 = 0;
				digitalWrite(63, 0);
			} else {
				pin_63 = 1;
				digitalWrite(63, 255);
			}
			break;

	case 65:	if (pin_65== -1){		// if so, we need to initialize the mode to output
				pin_65 = 0x00;		// toggle the pin's status to simpile On/Off mode
    				pinMode(65, OUTPUT);
			}
			if (action==0) {
       				digitalWrite(65, 0);
				pin_65 = 0;	// remember the pin is turned off
				return;
			}
			if (action==1) {
       				digitalWrite(65, 255);
				pin_65 = 1;	// remember the pin is turned on
				return;
			}
			if (pin_65) { // toggle the pin's status
				pin_65 = 0;
				digitalWrite(65, 0);
			} else {
				pin_65 = 1;
				digitalWrite(65, 255);
			}
			break;
	case 66:	if (pin_66== -1){		// if so, we need to initialize the mode to output
				pin_66 = 0x00;		// toggle the pin's status to simpile On/Off mode
    				pinMode(66, OUTPUT);
			}
			if (action==0) {
       				digitalWrite(66, 0);
				pin_66 = 0;	// remember the pin is turned off
				return;
			}
			if (action==1) {
       				digitalWrite(66, 255);
				pin_66 = 1;	// remember the pin is turned on
				return;
			}
			if (pin_66) { // toggle the pin's status
				pin_66 = 0;
				digitalWrite(66, 0);
			} else {
				pin_66 = 1;
				digitalWrite(66, 255);
			}
			break;

	default:	SERIAL_ECHOPAIR("?Unable to take action on LED pin: ", pin);
			SERIAL_ECHO(" \n");
			break;
	}
	return;
}
~~~~

thinkyhead, 2016-07-17:
Check out the `TEMP_STAT_LEDS` feature, which (maybe oddly) sets led colors based on whether any of the heaters are at (or targeted) over 55 degrees.

jbrazio, 2016-07-18:
We consider this topic as answered.
Please re-open this issue if still valid or if you still require more support from us.

closed 2016-07-18 by jbrazio