MarlinFirmware/Marlin #4334: string format via serial?

gbaumgart, 2016-07-17:
Hi,
I could manage to connect via custom serial-port client in Node.JS. However, when connecting I don't get really plain text but utf8 looking like text.

Could please anyone tell me what i have to do to get it as readable text. I was studying pronto-CLI and also your source code all day long but I can't get it converted. 

thats me serial config : 

var options = {
                baudrate: 25000,//is correct
                parity: 'none',//'none', 'even', 'mark', 'odd', 'space'
                rtscts: false,
                xon: false,
                xoff: false,
                xany: false,
                hupcl:true,
                rts: true,
                cts: false,
                dtr: true,
                dts: false,
                brk: false,
                databits: 8,
                stopbits: 1,
                buffersize: 256,
                parser: com.parsers.raw
};
here a sample: \r�E\u0010\u0017w\u001e[�L�O|���\u001ek\u001ak,\u001b$rT' 

or as hex: 

 <Buffer 3a 24 0a 0f fb 59 0d d4 a3 50 78 c0 1a d9 be 61 1a 94 5f 72 70 94 04 52 1e 11 50 c3 d4>

any idea ? 
I was also looking for simple protocol specs on reprap.org but i am ending always in dead links or incomplete docs.

thanks a lot.

jbrazio, 2016-07-17:
Are you positive you're connecting to the Marlin serial port ?
We use plain ASCII on the serial port, try lowering the baud rate or disabling RTS or CTS.

birkett, 2016-07-17:
Baud rate is incorrect - needs to be 250000.

jbrazio, 2016-07-17:
@birkett he's got a comment `//is correct` for the baud rate :-)

gbaumgart, 2016-07-18:
hi,
so tried different port speeds and RTS/CTS settings but no. Btw, the printer connect successfully at 250000 with the printrun console on linux.  So now I am out of advice. 

i gonna try harder by debugging prconsole with PyCharm and if there is any interest, let you know. 

I can also post a minimal test package here with tries to connect to Marlin via node-js "serialport". There are actually some little apps out there doing the same but they are all abandoned, outdated or not working. 

thanks a lot again. close this here just. I understand its not really your problem. Btw, is there a special command terminator after "Mxx" or "Gxx" i have to send ? something like 0x0d or so ?

Blue-Marlin, 2016-07-18:
'\n' is required to end a line. (nothing will happen before seen that)
'\r' is optional.

jbrazio, 2016-07-18:
One problem you may find is the baud rate, some terminal programs may not support 250000.
I would start by testing at 115200 and keep it until the connection is stable.

gbaumgart, 2016-07-18:
Thanks Blue Marlin & jbrazio. Gonna re-install Marlin with that baud rate.

gbaumgart, 2016-07-18:
holy crap jbrazio, that worked!  Now i am back business and we can enable our user to build their own printer software with a few clicks: 

[image]

i guess, we just have to add the G-CODEs and M-Codes of Reprap but also from Marlin.

horray! 2 days fighting over such stupid thing. i gonna tell the author of nodejs/serialport.

closed 2016-07-18 by gbaumgart