MarlinFirmware/Marlin #4346: Support for G-Code (dot) subcommands

jbrazio, 2016-07-18:
Developers, how do you feel about supporting the G-Code standard of (dot) subcommands ?
Something like (ignore the stupid example):
- `G28`: Standard cartesian home function
- `G28.1`: Delta home function

This could be usefull when we have the same command with multiple meanings which cannot be controlled by a single argument switch.

Roxy-3D, 2016-07-18:
Are you asking if this is a good format for the documentation?   I think (with very little analysis) this would work well.

If you are asking for people's opinions on doing this in the firmware, I think my reply is in the opposite direction.

jbrazio, 2016-07-19:
No.. not for documentation.. let me get a better example from the [RS274/NGC standard](http://www.nist.gov/customcf/get_pdf.cfm?pub_id=823374):

> G92 offset coordinate systems and set parameters
> G92.1 cancel offset coordinate systems and set parameters to zero
> G92.2 cancel offset coordinate systems but do not reset parameters
> G92.3 apply parameters to offset coordinate systems

This allow us to "group" G-Code commands, for instance instead of having:
- `G75` Start print timer
- `G76` Pause print timer
- `G77` Stop print timer

We could have:
- `G75` Show current timer
- `G75.1` Start print timer
- `G75.2` Pause print timer
- `G75.3` Stop print timer

Roxy-3D, 2016-07-19:
Yes...  That would be good.  But really...   Those commands (G75, G76 and G77) could have just looked at a parameter to make it not necessary to grab 3 separate GCode's.

jbrazio, 2016-07-19:
In fact we use the S, P, E, F wrongly.. if we look at the spec F always means Feedrate, P always means Parameter, S always means Speed etc.

Maybe it's to late to change this.
Let's hope when we change to binary based protocol we do a better job.

closed 2016-07-19 by jbrazio

Roxy-3D, 2016-07-19:
Well...   The Slicer's only use a few commands to do all of their work.   So, keeping that standard GCode is very important.   But for all of the weird commands that a user is typing in to do something (or putting into their startup or ending GCode) I'm not so sure it matters if we conform to the specification.

With regard to the binary based protocol, wouldn't the information being sent just be a binary representation of it?   Or would there be further encoding of the commands done?

jbrazio, 2016-07-19:
Binary based is a bit more strict, in one hand we have to encode different native types (int, float, bool) , and on the other hand we need to leverage the format to allow bit-wise math so we can save space by using one byte with multiple meanings.

thinkyhead, 2016-07-20:
GCode for 3D printers is not an exact implementation of CNC-style GCode, but more just inspired by it. There are probably dozens of things that we do differently.

Roxy-3D, 2016-07-20:
> There are probably dozens of things that we do differently.

Yes.    My personal thinking is we have to be extremely careful to not do anything that disrupts the community.   For example, we can't change any of the commands that the Slicer's use to do their work.   But on new commands...   We don't even have to stick to one letter parameter names.    My vote is we start supporting full word parameters because people can remember those more easily.

But right now...    That is probably too controversial to mess with.