MarlinFirmware/Marlin #4366: Inverse kinematics support?

roykolak, 2016-07-21:
I'm wondering if marlin supports mapping inverse kinematics. Working on a project involving a ramps board and a dobot arm. Was able to get everything hooked up and moving, but of course the arm just flails wildly when I attempt to print anything. 

Any feedback here would be great!

thinkyhead, 2016-07-21:
Out of the box we support kinematics for Delta and SCARA machines. CoreXY, CoreXZ, and CoreYZ "kinematics" are also supported. If you want to add novel kinematics, it shouldn't be difficult. Search for the text "`ENABLED(DELTA)`" or "`ENABLED(SCARA)`" in the code, and you can see all the hooks for these custom kinematics. In addition to kinematic differences, they also require homing to the top, consideration for a round (or irregular) bed and "unreachable" points at the top, grid-based bed leveling, etc.

Depending on the design of your robot arm, SCARA kinematics may suit your purpose. It is designed for systems with two points of rotational motion that combine to produce XY motion.

roykolak, 2016-07-21:
Cool! will check that out, flashing ramps w/ the scara config seems straightforward enough.

I realize i typed 'robot' and not 'dobot'. Not sure if you are familiar with the [dobot](http://dobot.cc/). It doesn't have the same configuration as a scara arm. In your experience, have you ever seen anyone be successful w/ using a similiar style arm as the dobot w/ marlin?

thinkyhead, 2016-07-21:
@roykolak I haven't seen any robot-arm-specific uses of Marlin. But there is certainly some interest. For example, [this](https://groups.google.com/forum/#!topic/deltabot/y9jyD1MVbVo) and [this](https://www.youtube.com/watch?v=FoWEvUt-hpU).

The main difference between the basic SCARA and the **dobot** is that its arm is rotated 90 degrees, so the arm joints alone move in "Y" and Z, while it rotates to produce a curved "X". The XY kinematics for the SCARA can be roughly translated into the "Y" and Z movements for the dobot, while the "X" motion goes to both the rotation of the arm joints and the rotation of the base.

The rotation of the base is actually very simple to solve. You simply rotate to align to the current XY position relative to the center of rotation (CX, CY). After that you only need to extend the arm to achieve the distance-at-height for the given XYZ. This is where the sideways version of SCARA will apply. You simply convert the SCARA XY kinematics directly into "Y" and Z, where the "Y" is simply the radius from CX, CY to X, Y, and "Z" is the equivalent to the SCARA's X.

Of course, you simply use `atan2` to get the rotation angle for XY. I leave it to you to solve the other two joints based on the SCARA code…
```cpp
void calculate_delta(float cartesian[3]) {
  float SCARA_pos[2];
  static float SCARA_C2, SCARA_S2, SCARA_K1, SCARA_K2, SCARA_theta, SCARA_psi;

  SCARA_pos[X_AXIS] = cartesian[X_AXIS] * axis_scaling[X_AXIS] - SCARA_offset_x;  //Translate SCARA to standard X Y
  SCARA_pos[Y_AXIS] = cartesian[Y_AXIS] * axis_scaling[Y_AXIS] - SCARA_offset_y;  // With scaling factor.

  #if (Linkage_1 == Linkage_2)
    SCARA_C2 = ((sq(SCARA_pos[X_AXIS]) + sq(SCARA_pos[Y_AXIS])) / (2 * (float)L1_2)) - 1;
  #else
    SCARA_C2 = (sq(SCARA_pos[X_AXIS]) + sq(SCARA_pos[Y_AXIS]) - (float)L1_2 - (float)L2_2) / 45000;
  #endif

  SCARA_S2 = sqrt(1 - sq(SCARA_C2));

  SCARA_K1 = Linkage_1 + Linkage_2 * SCARA_C2;
  SCARA_K2 = Linkage_2 * SCARA_S2;

  SCARA_theta = (atan2(SCARA_pos[X_AXIS], SCARA_pos[Y_AXIS]) - atan2(SCARA_K1, SCARA_K2)) * -1;
  SCARA_psi = atan2(SCARA_S2, SCARA_C2);

  delta[X_AXIS] = SCARA_theta * SCARA_RAD2DEG;  // Multiply by 180/Pi  -  theta is support arm angle
  delta[Y_AXIS] = (SCARA_theta + SCARA_psi) * SCARA_RAD2DEG;  //       -  equal to sub arm angle (inverted motor)
  delta[Z_AXIS] = cartesian[Z_AXIS];
}
```

closed 2017-10-31 by thinkyhead

andresdominguez92, 2018-08-31:
Good does not help, I mount all this robotic robot model that is this
https://www.thingiverse.com/thing:1718984
it works and everything but I want to take advantage of the power of the RAMPS 1.4, I want to put a 3D printer head, they told me that it is possible, but that it does not bother me, because later I can solve it, what I want is for the arm can make figures or piece in 3D, so if someone recommends me what I can do, or what programs to use, if you can give me a hand, because they recommended this blog

thinkyhead, 2018-08-31:
@andresdominguez92 — This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. For best results getting help with configuration and troubleshooting, please use the following resources:
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube

andresdominguez92, 2018-08-31:
So what the fuck did they recommend me, this place since nowhere does it help me with the problem that I have that is, only to make a robot arm print 3D, as well as the robot Dobot, which shit less aml that I come to ask for help or an advice

thinkyhead, 2018-08-31:
Sorry. This is a workplace, not a community forum. Please read:

https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question

AnHardt, 2018-08-31:
That [RobotArm](https://www.thingiverse.com/thing:1718984) is a toy. It will never be able to reliably 3d-print anything.
Currently we do not know anyone who has a qlue about robot arms and is around here.

The best advice i can give you is to try to, write a g-code preprocessor to rewrite the coordinates and maybe split up the moves, to get more 'straight' moves. Writing a preprocessor can be done much faster for you than a newcomer can learn how to do it in Marlin.

Building at first a machine and then realising, it can't do your job is pure madness.  
The better order of doing things is to first think about what you want to do, and after that to select a machine/project what can do that hard and software wise.

You are invited to come back with a PR or at least concrete questions about Marlin.

The--Captain, 2020-01-10:
If you're still looking for an example of someone who has already done something very similar with Marlin (although they likely mangled it a bit), you might want to check out the source for uArm Swift Pro firmware here on github (https://github.com/uArm-Developer/SwiftProForArduino).  The DEFAULT_AXIS_STEPS_PER_UNIT looks potentially interesting...