MarlinFirmware/Marlin #4367: homing problem

alifadl09, 2016-07-21:
I was using an old marlin firmware with auto bed leveling enabled and every think worked great, but after i updated marlin after it runs G29 the nozzle raise then start printing from the air.
I use proximity sensor for auto bed leveling, my start Gcode command is:
G28
G29
G92 Z0.2
it turns out that i forgot to set (z min end stop) to true ,so i set it true but another problem appears ,when i hit z homing button it home to the opposite direction(not until it hit end stop) and auto leveling also .

[marlin_with_autobed_leveling.zip](https://github.com/MarlinFirmware/Marlin/files/376514/marlin_with_autobed_leveling.zip)

﻿

brainscan, 2016-07-21:
Your probe offset is positive but should be negative if using proximity sensor. Also why the  G92 Z0.2? I think that was a work around on old Marlin that's no longer needed.

jbrazio, 2016-07-23:
Possible duplicate of #4325

closed 2016-07-23 by jbrazio