MarlinFirmware/Marlin #4375: Assembler output helper script

thinkyhead, 2016-07-22:
Here's a handy script I wrote to do a quick dump of the assembler code from Marlin. Adapt for your system. If we come up with a "universal" script that works on Linux and Windows, we might include it as an additional helper.
```bash
#!/usr/bin/env bash
#
# marlindump
#
# Dump and view Marlin's object output in Assembler
#

OBJDUMP="/Volumes/Media/Applications/DIY/Arduino 1.6.7/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-objdump"
TEMPFIND="/var/folders/*/*/T/*.tmp"
DEST="/Users/thinkyhead/Desktop/scratch"

MARNAME=Marlin.ino
ELFNAME=$MARNAME.elf
HEXNAME=$MARNAME.hex

MARLIN_ELF=$(find $TEMPFIND -name $ELFNAME)
MARLIN_HEX=$(find $TEMPFIND -name $HEXNAME)

if [[ -z $MARLIN_ELF ]]; then
  echo "`basename $0`: No 'Marlin.ino.elf' found." 1>&2 ; exit 1
fi

SIZE=`stat -f%z "$MARLIN_HEX"`
DATE=`ls -la "$MARLIN_ELF" | awk '{ print $6 " " $7 " " $8 }'`

echo "Dumping build from $DATE ($SIZE)"

mkdir -p "$DEST"

"$OBJDUMP" -S "$MARLIN_ELF" >"$DEST/marlin.a" && subl "$DEST/marlin.a"
```

Roxy-3D, 2016-07-22:
I'm on a Window's system.  I've never been able to find any temporary files for Marlin when I've looked for them.   It kind of seems like Arduino is deleting all the temporary files.   That might be a good question to ask over on the Arduino forums.

jbrazio, 2016-07-22:
No it's quite easy do change the folder, if you have arduino in your PATH just do the following:
```
arduino --pref build.path=C:\tmp\arduino
```

Otherwise just chdir into the folder where you have it and run that command from a cmd window.

thinkyhead, 2016-07-22:
@Roxy-3D For temporary files on Windows, poke through `\Documents and Settings\<USER>\Local Settings\Temp`.

Blue-Marlin, 2016-07-22:
C:\Users\username\AppData\Local\Temp\build*.tmp (win7)

thinkyhead, 2016-07-22:
A: and B: are your floppy drives, Windows users. Floppy drives!

Blue-Marlin, 2016-07-22:
No floppy drives. But reserved letters.

jbrazio, 2016-07-22:
Don't make fun of it, MacOS just got diagonal window resizing a couple of years ago.. ;-P

Blue-Marlin, 2016-07-22:
C:\Program Files (x86)\Arduino\hardware\tools\avr\bin\avr-objdump.exe (win7)

thinkyhead, 2016-07-23:
> MacOS just got diagonal window resizing

Resizing from the edges, you mean. It always had resize from the corner. I believe they might have had to wait for a patent to expire, but correct me if I'm wrong.

jbrazio, 2016-07-23:
> Starting with OS X 10.7, Apple changed its method for resizing windows. In the Classic Mac OS and through the first six releases of OS X, you had to resize windows using the tab at the bottom-right of the window; however, with OS X Lion Apple changed this behavior to give resizing controls along all edges of application and document windows.

> What this means is in most cases, instead of having to move the left and top sides of a window to a desired location and then resize the rest accordingly, you can simply resize from the left, from the top, or any corner of these edges to move your window around.

thinkyhead, 2016-07-23:
Now… watch our comments disappear!

Roxy-3D, 2016-07-23:
> Now… watch our comments disappear!

How could our comments disappear?  Very few people have the privilege level to make that happen?

thinkyhead, 2016-07-23:
(offtopic comments deleted, cheers!)

petrzjunior, 2016-07-23:
Btw. Git on Windows installs Bash cmd. It was able to run the missingTraslations script (but only from direct input, not from file), but writing something in Windows cmd shouldn't be such problem.

closed 2018-02-26 by thinkyhead