MarlinFirmware/Marlin #4377: MINTEMP when uploaded different version

Kamran14, 2016-07-23:
So I've upgraded my printer and wanted to get the newest update as well. When I configured everything I uploaded it to my printer, only to find out my z-axis wouldn't budge at all. I tried the IRC for help and we couldn't find out why. So I decided to try my old firmware and see if it made a difference, surprisingly it did. My Z-axis started moving and I could finally home everything. BUT, a new problem has occurred!! I now get a 'Err:MINTEMP'. Now I just got a brand new hotend(E3D v6) and I made sure that all the wiring is correct. I'm 99.99% that its firmware based. My setup is EFF(will be EFB when I get heated bed and new mount for extruder).

Side Note: Along with the new Firmware, the text 'X', 'Y', and 'Z' would turn into '?' every other second and turn back. I could still move my X and Y, just not my Z.

thinkyhead, 2016-07-23:
Doe sit behave better if you do a `G28` first?

thinkyhead, 2016-07-23:
> Err:MINTEMP

Is this only with the newer code (RC6 or RCBugFix, for example), and not with older (1.0.2-1) code?

If it occurs always, check the thermistor resistance with a multimeter.

Blue-Marlin, 2016-07-23:
> Err:MINTEMP

Not defining a bed-thermistor may help - if you don't have one.
Have a look on the console output. There you can see if it's bed or extruder related.

Kamran14, 2016-07-24:
@thinkyhead  I have tried changing the acceleration and it didn't change. The `G28` command also did nothing to Z-axis.

I get the `MINTEMP` error with the old version, not RC6.

@Blue-Marlin how would I be able to see the console output? Something like pronterface? I defined my printer as an EFF, shouldn't the bed be disabled by default?

Blue-Marlin, 2016-07-24:
If the bed thermistor is defined but not connected you __will__ get a MINTEMP error!

Blue-Marlin, 2016-07-24:
If you have defined SAFE_HOMING you have to home X and Y before Z.

Kamran14, 2016-07-24:
@Blue-Marlin When looking at console, I get this: `
[ERROR] Error:1

: Extruder switched off. MINTEMP triggered !
Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)`

Should I be focusing on the older version or the new ones?

Blue-Marlin, 2016-07-24:
That is the extruder thermistor 1 - the second extruder!
Don't define any thermistors you did not connect! Or use Nr. 999 or 998 if available.

Kamran14, 2016-07-24:
I've only defined 1 extruder though... `#define EXTRUDERS 1`

Kamran14, 2016-07-24:
I replaced my thermistor with a fresh one and get the same error.

thinkyhead, 2016-07-24:
I believe what @Blue-Marlin is trying to say is, for some reason the error is related to temp sensor 1, not temp sensor 0. So perhaps you have set this to non-zero:
```cpp
#define TEMP_SENSOR_1 1
```

Kamran14, 2016-07-24:
So I took a break and when I came back and turned my printer on, the error was gone...
Now when I'm calibrating my x-axis, no matter what value I put for `DEFAULT_AXIS_STEPS_PER_UNIT`, it always moves 91mm.

thinkyhead, 2016-07-24:
> when I'm calibrating my x-axis, no matter what value I put for DEFAULT_AXIS_STEPS_PER_UNIT, it always moves 91mm

Even after `M502`?

Kamran14, 2016-07-24:
@thinkyhead After that I ran that command and told it to go 100mm, it went 99.73mm. So pretty close. Would I need to run that commands always?

thinkyhead, 2016-07-24:
It indicates that the values stored in the EEPROM are stale. After `M502` do `M500` to write the defaults to EEPROM, so they will be restored properly on startup.

Kamran14, 2016-07-24:
Okay, Thank you.
I've run into another problem(once again). My hotend heater cartridge isn't heating up at all... I checked wiring and it seems normal.

thinkyhead, 2016-07-25:
> I checked wiring and it seems normal.

Check the wires with a voltage meter.

Kamran14, 2016-07-25:
so I should test it while the board is on?

thinkyhead, 2016-07-25:
> so I should test it while the board is on?

If you want to see how much voltage the bed terminals are putting out, it would help.

The resistance on the heater cartridge you can test while the heater is disconnected.

Kamran14, 2016-07-26:
@thinkyhead I checked and it said 0. I talked to some people on IRC and they said I should get new pins and try that. I got new pins and another RAMPS board just incase. When I tried the pins, it gave the same result. So when I tried the board, you'd think it would be different, but it also gave same result....

thinkyhead, 2016-07-26:
@Kamran14 Please post your `Configuration.h`.

Kamran14, 2016-07-27:
@thinkyhead I pasted it on an 3rd party site. Is that okay? https://ideone.com/BjHooD

thinkyhead, 2016-07-28:
@Kamran14 Your settings seem ok to me. You might try the latest version of Marlin (1.1.0-RC7) and see if it works differently, or better. If it does work with the newer Marlin, then maybe we fixed some bug that is causing your issue. Unfortunately we can't support fixing bugs in older versions of Marlin, so we generally recommend upgrading as a first step.

closed 2017-10-31 by thinkyhead