MarlinFirmware/Marlin #4386: Issues in firmware?

krom7, 2016-07-23:
3D printer has been transformed into CNC. Firmware Marlin was partially changed:
The axes to change the values ​​of the minimum to maximum value . CNC work , but with the memory card hangs . By USB and Bluetooth works well , but if a long time to connect to the card - just freezes . Video:
https://www.youtube.com/watch?v=wQNvgGxlw0U
Possible bug in the firmware ?

brainscan, 2016-07-23:
That video is over a year old and says the problem is solved. Have you used the correct link?

https://www.youtube.com/watch?v=wQNvgGxlw0U

jbrazio, 2016-07-24:
We do not [yet] officially support any CNC implementations of Marlin.
The example video you show is from 2015, I advise to test with the latest [RCbugfix](https://github.com/MarlinFirmware/Marlin/tree/RCBugFix), but this means you'll have to port back any custom changes.

closed 2016-07-24 by jbrazio

thinkyhead, 2016-07-24:
@krom7 I see you just signed up today. Welcome to Github!

krom7, 2016-07-24:
Thanks for answering.
Roller author writes the problem is solved , but it is not.
After some time the problem arose again .
This may not be due to the firmware ?

thinkyhead, 2016-07-24:
It might be a bug in the firmware, but unfortunately we have moved forward and are not patching older versions of Marlin. So I suggest trying to upgrade to a newer version of Marlin as a stating-point and make the CNC modifications that were made to whatever version of Marlin that your machine's firmware is based on. To get started on moving to newer code, please join in the discussion at #3208. Perhaps one of the enthusiastic modders there can help point you in the right direction.