MarlinFirmware/Marlin #4388: Looking for testers to join the team

jbrazio, 2016-07-24:
We need your help, Marlin supports different kinematics and the developers do not have access to all of them. The idea is to create some groups of testers for the main kinematics which can test the new developments and help solving bug by testing PR.

Thank you for supporting Marlin by volunteering.

Kinematics | Group Name | Testers
--- | --- | ---
Cartesian | @MarlinFirmware/testers-cartesian-team |
CoreXY | @MarlinFirmware/testers-corexy-team |
CoreXZ | @MarlinFirmware/testers-corexy-team |
Delta | @MarlinFirmware/testers-delta-team |

thinkyhead, 2016-07-24:
We are planning an ongoing "testing lab" here in Portland. Since it will be done online and over a span of days, please feel free to join in. I will give more details once I have the itinerary and guidelines posted online.

tiagom62, 2016-07-24:
Put me down for the delta team.

Roxy-3D, 2016-07-24:
@tiagom62   What kind of Delta do you have?   How well is it calibrated?

tiagom62, 2016-07-24:
I have a Folger Tech Kossel, not the greatest calibration but its good enough for what i toy around with.

Roxy-3D, 2016-07-24:
> I have a Folger Tech Kossel, not the greatest calibration but its good enough for what i toy around with.

Cool!  I have a Geeetech Delta that totally sucks.   That is why I started designing the Unified Bed Leveling.  It was to fix all of the Delta calibration issues I had.   But I don't have the Delta side of the code done yet.  But I have a Delta Physical Bed Leveling command (G25) in the UBL branch that helped me a lot to get my low quality Geetech Delta able to print things 40 cm from the center of the bed.   

I need  to get my Delta back together again.  And then I'll get that code and the UBL code to work on Delta's.   The reason I asked "How well is it calibrated?" is because this code is brute force fix for badly calibrated machines.   (It should help high quality machines that are calibrated very well too, but the improvement will be much less.)   

Let me get the Delta side of the equation cleaned up (a couple of weeks) and it would be very helpful to see how much G25 and UBL helps you get full use of your Print Envelope.

molak34, 2016-07-24:
I have a folgertech i3 with some modifications and properly calibrated.I'm finishing a smartcore XY 

Put me for Cartesian and core XY

Roxy-3D, 2016-07-24:
> I have a folgertech i3 with some modifications and properly calibrated.

We especially need you on the UBL Test Team!     If you have an i3-2020,  The entire system was developed on your printer!

molak34, 2016-07-24:
>  If you have an i3-2020

I have the acrylic i3. Back when I bought it the i3-2020 wasn't for sale ..

brainscan, 2016-07-24:
You can add me to the Cartesian group, hoping to have a delta and coreXY very soon.

jbrazio, 2016-07-24:
@esenapaj don't you have a delta bot ? do you mind joining the tester group for deltas ?

superlou, 2016-07-25:
I have an FLSUN delta I'd be happy to help test with if someone has a procedure on how to do the testing.  It does not have an automatic bed leveling sensor, though I'm looking in to adding one.

jbrazio, 2016-07-25:
@superlou actually this will be on demand.

adamfilip, 2016-07-25:
Im willing to test with my CoreXY machine. should be up and running soon

prophet-id, 2016-07-25:
I can help on testing Delta.

MichaelBrock, 2016-07-26:
I would gladly help test for corexy.  I have a homebuilt SmartrapCore that is reasonably well dialed in.   I have some experience digging around in the code (and I code for a living although not in any C derivative).  Just as an fyi, the group names for CoreXY and CoreXZ in the OP are the same name.

jbrazio, 2016-07-26:
Good catch @MichaelBrock but there is a reason.. I don't know if we'll find CoreXZ people thus I haven't created yet the group inside the organization. (lazy me :-P)

Thank you @adamfilip @prophet-id and @MichaelBrock for joining in.

tiagom62, 2016-07-27:
@Roxy-3D Will be interesting to test. I use M665 and M666 generated from http://escher3d.com/pages/wizards/wizarddelta.php without it i cant print a object larger then 40mm without driving the hotend into the bed.

superlou, 2016-07-27:
I also use escher3d's wizard to compensate for errors in my low cost delta.  Is UBL expected to correct for tower angle errors and rod length errors, or just estimate the orientation of the build plate?

Roxy-3D, 2016-07-27:
I guess I need to get the UBL code running on Delta's, because I think it will cure your problem.   That is why I wrote it.  To fix my low quality Geeetech Delta!!!     But there is a lot of other good Delta stuff in the UBL branch.  Those things need to jump the fence and get back into the RCBugFix branch.   M665 and M666 are a good example.

Roxy-3D, 2016-07-27:
> I also use escher3d's wizard to compensate for errors in my low cost delta. Is UBL expected to correct for tower angle errors and rod length errors, or just estimate the orientation of the build plate?

I think it can.   You will want to run the Mesh at as high of resolution as you can.  But I think a high resolution Mesh will do just fine.   We need to get the Delta version of it up an running and we will find out!

otvald, 2016-07-28:
May I suggest that the participating rigs (I.e. rig type/name, controller, and short description) **and** configuration files are documented?

If the configuration files change, it would be nice to track those changes too.

This give a placeholder for comparing against our own individual setup, as well as serves as a reference point in case there are regressions errors introduces in the code.

otvald, 2016-07-28:
@Roxy-3D I would very much like to follow the development of your UBL, and participate with whatever I can bring to the table (at least some testing).

Is this thread the base for discussion/feedback or?

Roxy-3D, 2016-07-28:
> I would very much like to follow the development of your UBL, and participate with whatever I can bring to the table (at least some testing).

> Is this thread the base for discussion/feedback or?

Joao is trying to fill out the test teams with this thread.    So creating new issues is probably a better way to go.    But, as far as following the development and helping, just jump in, especially when you see a problem and have similar hardware.

jbrazio, 2016-07-28:
@otvald, would you like to join a testing team, if so please tell me your kinematics so I can add you to the correct team.
@Roxy-3D do you believe we require a temporary specific team for UBL ?

Roxy-3D, 2016-07-29:
>  do you believe we require a temporary specific team for UBL ?

@jbrazio   Not yet.   We need to just get a few systems up and in use.   I already know that the Full Graphics LCD Panels do not behave properly.  I have one on order from eBay and it should be here in a week.   I'll start trying to get those up and running as soon as I get the display.   And I haven't even bothered trying to get Delta's up and running.   Delta's are so ornery.  So, I need to get my low quality Geeetech Delta back together and running so I can start debugging the Delta version of the code.   That is where this code will really shine.   

So...    I think it might be best if we go slow.   I have several weeks of work to address the issues I know about.   Let's wait to get more people screaming "I found a bug!!!"   But with that said, I'll try to help anybody that is trying to bring up the code.

otvald, 2016-07-29:
@jbrazio When time permits, I would be happy to test on my setup.

I have a 'china' (Micromake clone) of a delta. Called K800, I believe.
Features:
- 'delta'
- RAMPS 1.3 
- Autolevel built into probe (when probe touches plate, continous pressure will cause a switch being triggered)
- 150mm diameter print size and 260+ mm print height.
- Diagonal rod length 220.61mm
- Currently no heated plate

jbrazio, 2016-07-29:
@otvald welcome to the delta team :-)

zenmetsu, 2016-08-12:
I have two deltas, a 3DPrinterCzar Rostock Mini Pro and a copy of the Delta-Six from openbuilds, so I can use one as a guinea pig.  

I also have a build in progress which is a 6-DOF Stewart Platform.  It could certainly be used as a 3D printer if you only use the 3 translational axes.  The kinematics are more or less identical to a delta in that case.  Not sure how the 3 rotational axes would benefit 3D printing, but I am building it for milling/laser engraving.

esenapaj, 2016-08-13:
@jbrazio 
I participate in the Marlin project from the user side on the basis of only its own interests and free will.
Sorry.

chasecrom, 2016-08-20:
I'll be making a CoreXY machine in the coming weeks, so put me down with the CoreXY group.

mjmeans, 2016-08-29:
I'm using an upsized delta Kossel Mini. Reprap'ed custom. It has 15mm black extrusion with 360mm sides and 750mm verticals. The bed is lifted 60mm up by using [this](http://www.thingiverse.com/thing:626279) both over and under the bottom vertexes and [this](http://www.thingiverse.com/thing:1305559) at the top vertexes. This significantly stabilizes the printer without needing to go to expense of 20mm extrusion. 250mm diameter cork + kapton heater + 3mm aluminum spreader + borosilicate glass as the bed. Printed recirculating linear bearings. My own modification of the deployable allen zprobe (3mm brass rod, positioned at the edge the effector and lowered). My own modification of bed mounting brackets. My own design of a deployable allen stow post fitting the appearance of frame braces. Mirrored mounted Plexiglas on two sides and 12V strip lighting up the tower. All printed parts are black ABS so the appearance is all black with mirrors and shiny aluminum bed. From the front I can monitor the print from all sides while it is printing. Arduino Mega + RAMPS 1.4. Using 288mm diagonal rods (which isn't long enough to get full use of 250mm diameter bed [not enough torque at the extreme edges] so I'll be replacing those if I ever need to actually print something full width). Using RC7 with a few hand picked fixes from RCBugFix.

closed 2017-10-31 by thinkyhead