MarlinFirmware/Marlin #4391: M500 Not saving Bed PID Settings

Ghost114179, 2016-07-24:
I noticed that after issuing M304 and the PID values (M304 Pxx.xx Ixx.xx Dxx.xx), the serial will report p:xx.xx i:xx.xx d:xx.xx with the correct values, but sending M500 does not save the values to EEPROM. After looking through the code for saving to EEPROM, it seems that the M500 command will not save the bed PID values, only the hot end PID values.

As a quick fix I added several lines of code to Config_StoreSettings() in ConfigurationStore.cpp:
EEPROM_WRITE_VAR(i,bedKp);
EEPROM_WRITE_VAR(i,bedKi);
EEPROM_WRITE_VAR(i,bedKd);

and the matching commands to Config_RetrieveSettings() in the same file

EEPROM_READ_VAR(i,bedKp);
EEPROM_READ_VAR(i,bedKi);
EEPROM_READ_VAR(i,bedKd);

I haven't looked into how the arduino uses EEPROM, but the functions for reading and writing seemed self explanatory, so I added this and it seemed to fix the issue.
Please let me know if there is something wrong with doing this that I am not aware of, and if possible, add the ability to save Bed PID Values to EEPROM with M500.

thinkyhead, 2016-07-24:
You mean, like [here](https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/configuration_store.cpp#L280) and [here](https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/configuration_store.cpp#L456)?

Ghost114179, 2016-07-24:
After looking at the linked code it appears I am using an older version that did not have those lines of code. This is what I have for the Config_RetrieveSettings function. I bolded the code I added; would this work or is it important that I use the newer version of Marlin (I was unable to find the version I am using anywhere in the code).
 I am hesitant to use the newer version as I had quite a few problems trying to set and tune Marlin to my printer, and I have just gotten it working really well (other than Bed PID settings not saving).

Note: I basically have the matching EEPROM_WRITE_VAR functions listed in the same way under Config_StoreSettings() next to the EEPROM_WRITE_VAR function for the hot end PID settings.

`#ifdef EEPROM_SETTINGS
void Config_RetrieveSettings()
{
    int i=EEPROM_OFFSET;
    char stored_ver[4];
    char ver[4]=EEPROM_VERSION;
    EEPROM_READ_VAR(i,stored_ver); //read stored version
    //  SERIAL_ECHOLN("Version: [" << ver << "] Stored version: [" << stored_ver << "]");
    if (strncmp(ver,stored_ver,3) == 0)
    {
        // version number match
        EEPROM_READ_VAR(i,axis_steps_per_unit);  
        EEPROM_READ_VAR(i,max_feedrate);  
        EEPROM_READ_VAR(i,max_acceleration_units_per_sq_second);
        
        // steps per sq second need to be updated to agree with the units per sq second (as they are what is used in the planner)
		reset_acceleration_rates();
        
        EEPROM_READ_VAR(i,acceleration);
        EEPROM_READ_VAR(i,retract_acceleration);
        EEPROM_READ_VAR(i,minimumfeedrate);
        EEPROM_READ_VAR(i,mintravelfeedrate);
        EEPROM_READ_VAR(i,minsegmenttime);
        EEPROM_READ_VAR(i,max_xy_jerk);
        EEPROM_READ_VAR(i,max_z_jerk);
        EEPROM_READ_VAR(i,max_e_jerk);
        EEPROM_READ_VAR(i,add_homeing);
        #ifdef DELTA
          EEPROM_READ_VAR(i,delta_radius);
	    EEPROM_READ_VAR(i,delta_diagonal_rod);
          EEPROM_READ_VAR(i,max_pos);
          EEPROM_READ_VAR(i,endstop_adj);
          EEPROM_READ_VAR(i,tower_adj);
          EEPROM_READ_VAR(i,z_probe_offset);
          // Update delta constants for updated delta_radius & tower_adj values
          set_delta_constants();
        #endif
        #ifndef ULTIPANEL
        int plaPreheatHotendTemp, plaPreheatHPBTemp, plaPreheatFanSpeed;
        int absPreheatHotendTemp, absPreheatHPBTemp, absPreheatFanSpeed;
        #endif
        EEPROM_READ_VAR(i,plaPreheatHotendTemp);
        EEPROM_READ_VAR(i,plaPreheatHPBTemp);
        EEPROM_READ_VAR(i,plaPreheatFanSpeed);
        EEPROM_READ_VAR(i,absPreheatHotendTemp);
        EEPROM_READ_VAR(i,absPreheatHPBTemp);
        EEPROM_READ_VAR(i,absPreheatFanSpeed);
        #ifndef PIDTEMP
        float Kp,Ki,Kd;
        #endif
        // do not need to scale PID values as the values in EEPROM are already scaled		
        EEPROM_READ_VAR(i,Kp);
        EEPROM_READ_VAR(i,Ki);
        EEPROM_READ_VAR(i,Kd);
        **EEPROM_READ_VAR(i,bedKp);**
        **EEPROM_READ_VAR(i,bedKi);**
        **EEPROM_READ_VAR(i,bedKd);**
        #ifndef DOGLCD
        int lcd_contrast;
        #endif
        EEPROM_READ_VAR(i,lcd_contrast);
    
		// Call updatePID (similar to when we have processed M301)
		updatePID();
        SERIAL_ECHO_START;
        SERIAL_ECHOLNPGM("Stored settings retrieved");
    }
    else
    {
        Config_ResetDefault();
    }
    Config_PrintSettings();
}
#endif`

thinkyhead, 2016-07-24:
That's pretty old, alright. The EEPROM has been patched for a long while. And recently we even added a checksum.

closed 2016-07-24 by thinkyhead