MarlinFirmware/Marlin #4392: Sanity check and extended documentation for Z_SAFE_HOMING needed

Alex9779, 2016-07-24:
We need a sanity check for `Z_SAFE_HOMING` and update comments would help too.
It should be emphasised in the comment that the position is calculated for the probe.
I realised that if you set the position to something the probe cannot reach no movement in any axis happens so compare to the bed leveling position where compilation fails if you set spots the probe cannot reach this should be done for `Z_SAFE_HOMING`too...

thinkyhead, 2016-07-24:
> emphasized in the comment that the position is calculated for the probe
```cpp
// - Position the Z probe in a defined XY point before Z Homing when homing all axes (G28).
```

closed 2016-07-25 by thinkyhead

Alex9779, 2016-07-25:
Ok now I need again a probe, I have no problem with that but I remember #4170 where I had to deactivate the probe because it bricked MBL...