MarlinFirmware/Marlin #4393: Feature request: relative Temperature changes

Dim3nsioneer, 2016-07-24:
Hi guys

I wonder what you think about having the possibility the change the hotend temperature relative to the current value by gcode, e.g. something like 'M104 R-5' to lower temperature by 5°C?
It could become handy for machines printing from sd card or similar where you might want to print an object again, but just a few degree hotter or cooler. Or with a different material. Without having to reslice the object. Also, if you change the temperature during print, following temperature changes would still work as they were intended to do.

Opinions?

If I missed a similar feature request (did a quick look through the code and the open issues), please point it out.

jbrazio, 2016-07-29:
This is kinda complicated, not to implement code wise but to handle real world usage situations..

Let's take the example, in your slicer you have a start G-Code block which tells the machine no heat the bed and the nozzle to a certain temperature, home the axis and so on. If you print one object using this G-Code at 200C and then you send to the printer `M104 R-5`, what will happen when the printer starts reading the G-Code ? It will set back the temperature to 200C because it's written on the G-Code.

You kinda can shift temperature nowadays by going into the menu and adjust the nozzle temperature while Marlin is printing, but once again, you can only do this after the start G-Code which sets the temperature to 200C.

Dim3nsioneer, 2016-07-29:
Thank you @jbrazio for your thoughts. I absolutely agree there has to be a absolute temperature defining command in the gcode for all machines which do not have their own material profiles stored.
I see the main use case in the fact one can use a gcode maybe sliced for PLA and which includes several temperature changes during print also for e.g. copolyester by adjusting the temperature with the menu possibilities you mentioned as soon as this first absolute temperature command is executed and having the consecutive temperature changes executed in a way they fit to the other material. Having absolute temperatures there would just lead to sudden underextrusion if the print is performed unattended.

boelle, 2019-02-18:
@Dim3nsioneer are you still missing this ?

Dim3nsioneer, 2019-02-19:
Nope, it was neglected long enough that I don't miss it anymore in the days of more professional 3d printers... ;-)

closed 2019-09-24 by shitcreek