MarlinFirmware/Marlin #4403: Unclear how to use mesh bed leveling

mgsb, 2016-07-25:
I have been trying to get Mesh Bed Leveling working since 1.1RC6. There have been a few issues opened that appear to have been fixed. It seems that MBL has evolved since some of the initial articles appeared describing it early this year (March). I am confused how to use it with RCBugFix latest. I grabbed RCBugFix today (7/24) and built it using more or less defaults for MBL. My confusion involves the Z-endstop and the settings in Configuration.h. 

I have a mechanical Z-endstop that I have traditionally used to set the initial Z distance of "0" after homing. This worked reasonably well for a flat and level print bed. My bed is not perfect so I want to try MBL to compensate. I got it to work once randomly and it was great. I could see the Z axis moving slightly as X and Y moved. 

Given these MBL settings in Configuration.h:

//===========================================================================
//============================ Mesh Bed Leveling ============================
//===========================================================================

#define MESH_BED_LEVELING    // Enable mesh bed leveling.

#if ENABLED(MESH_BED_LEVELING)
  #define MESH_INSET 10        // Mesh inset margin on print area
  #define MESH_NUM_X_POINTS 3  // Don't use more than 7 points per axis, implementation limited.
  #define MESH_NUM_Y_POINTS 3
  #define MESH_HOME_SEARCH_Z 4  // Z after Home, bed somewhere below but above 0.0.

  //#define MESH_G28_REST_ORIGIN // After homing all axes ('G28' or 'G28 XYZ') rest at origin [0,0,0]

  #define MANUAL_BED_LEVELING  // Add display menu option for bed leveling.

  #if ENABLED(MANUAL_BED_LEVELING)
    #define MBL_Z_STEP 0.025  // Step size while manually probing Z axis.
  #endif  // MANUAL_BED_LEVELING

#endif  // MESH_BED_LEVELING

Questions:

1) How should I set my Z-endstop for Home? Should it be set so that the nozzle is 4mm above the bed?

2) If the answer is 4mm above, how do I accommodate driving approximately 4mm below the Z endstop? It will just "compress" my Z endstop switch unmercifully.

Not complaining here - I think MBL will really be helpful. I am just wondering how to actually use it with my setup. Thanks for the neat feature.

Mark

thinkyhead, 2016-07-25:
1. Set your Z endstop normally. Get it as close to the ideal Z homing height as you can. Don't raise the Z endstop up, because you won't be able to move the Z axis lower without squishing it.
2. The setting `MESH_HOME_SEARCH_Z` merely indicates how much free space you are given to move while doing the mesh leveling procedure. 4mm is a safe value, but lower values might be fine also.

If you need the nozzle to raise up in-between points when doing the MBL procedure, try setting `MIN_Z_HEIGHT_FOR_HOMING` or `Z_RAISE_BETWEEN_PROBINGS` to a reasonable value, such as 5mm.

mgsb, 2016-07-25:
Thanks for the help. When I have the MESH_HOME_SEARCH_Z set to 4, I see the matrix (mesh) of values come out to be +/- 4.0 as reported by G29 S0. Is this expected? It seems like when I subsequently print with these values, the hotend is actually being driven +/-4.0 below the Z endstop. What should the value of Z offset by as reported by G29 S0? Example output below:

Send: G29 S0
Recv: State: 1
Recv: Num X,Y: 3,3
Recv: Z search height: 4
Recv: Z offset: 0.00000
Recv: Measured points:
Recv:   4.00000  4.12500  4.22500
Recv:   3.87500  4.10000  4.22500
Recv:   3.85000  3.95000  4.15000
Recv: X:72.79 Y:90.17 Z:19.30 E:35.10 Count X: 5837 Y:7230 Z:18670
Recv: ok

Roxy-3D, 2016-07-25:
If you are willing to be on the front end of things...   I'll give you very quick and timely support.    You might want to consider loading up the UBL (Unified Bed Leveling) code at:  https://github.com/MarlinFirmware/Marlin/tree/devel-ubl

It is based very heavily on @epatel 's Mesh Bed Leveling.   The very lowest levels of the code have not changed.   If you check out the ReadMe page when you go to that URL, you will have a better idea of whether you want to venture into the bleeding edge.   But on my i3 with a 20x4 display, it is rock solid and it compensates for very bad bed conditions.

adamfilip, 2016-07-25:
Im going to try this on my work in progress CoreXY machine.

Roxy-3D, 2016-07-25:
Great!   Just one caution...   The further you deviate from an Prusa i3 with a 20x4 display, the more likely we are to run into problems.   But with that said...  I'll do everything I can to make sure you have a good experience.

adamfilip, 2016-07-25:
Thanks.. Besides what fun is it.. if it all works properly :)
nothing like slamming the power button when the print  head tries to drive into the glass :)

thinkyhead, 2016-07-25:
> I see the matrix (mesh) of values come out to be +/- 4.0 as reported by G29 S0. Is this expected?

This is a side-effect of the design, but it is likely change in the near future. Either the `MESH_HOME_SEARCH_Z` will be subtracted from the mesh report, or the code will be altered so that `MESH_HOME_SEARCH_Z` isn't serving two somewhat separate purposes.

mgsb, 2016-07-25:
I would be interested in trying the new UBL code. I have a Makerfarm Pegasus which is an i3 derivative and have a ramps 1.4 with the 20x4 LCD. Reading the README, I guess I would need to have a Z probe? I have been doing manual probing but I could add a Z probe.

brainscan, 2016-07-25:
Mesh levelling is easier to set up using opto or Hall effect end stops as you can give yourself a lot of room.

Roxy-3D, 2016-07-25:
We should probably move the UBL discussions to a separate thread because we are going to confuse support questions being discussed in this thread.

> I would be interested in trying the new UBL code. I have a Makerfarm Pegasus which is an i3 derivative and have a ramps 1.4 with the 20x4 LCD. Reading the README, I guess I would need to have a Z probe? I have been doing manual probing but I could add a Z probe.

No.  A Z-Probe is not mandatory in the new UBL System.  It is nice to have a Z-Probe because that will let you acquire a lot of the bed data accurately and easily.   But it is not required.    You can just invalidate the Mesh with a G29 P0 and then you have two options on how to continue.  You can jump to the G26 Mesh Validation command and see where you are off, or you can manually probe (at least) parts of your bed with the nozzle using G29 P2 commands.

adamfilip, 2016-07-25:
Thought UBL was intended to be Automatic probing. to speed up the process.

Roxy-3D, 2016-07-25:
> Thought UBL was intended to be Automatic probing. to speed up the process.

It is!   But that doesn't mean a Z-Probe is mandatory.   And in fact, you don't need to have a Mesh defined to use the 3-Point (or soon Grid Based) leveling.   A G29 P0 will create a full Mesh that is all zero'ed out.   And then that Mesh can be tilted as appropriate.

closed 2017-10-31 by thinkyhead