MarlinFirmware/Marlin #4410: Unify pin mappings

thinkyhead, 2016-07-25:
One HAL to rule them all?

Assigning myself to manage the process of getting rid of the `Sd2PinMap.h` file dependencies. This is a topic to collect information to make that process go smoothly.

Roxy-3D, 2016-07-25:
> Use the fastio.h pin mappings where possible in the Sd2 code.

I know this is going to be a stupid question, but looking at the Sd2PinMap.h file, every value in there is known at compile time.   Why isn't it possible to move entirely over to the fastio.h pin mappings?  Why did you qualify the statement with a 'where possible' ?    What don't I understand?

> Take a look at https://github.com/mmarchetti/DirectIO as another alternative

I like the DirectIO alternative!!!

thinkyhead, 2016-07-25:
> Why isn't it possible to move entirely over to the fastio.h pin mappings? 

It comes down to the three SPI pins, apparently. I'm not sure what values end up where, whether the SPI pins specified in the `pins_BOARD.h` files need to override the SPI pins set in the Sd2 files, or the other way around, and so forth. But it's only that I haven't dived into it. I hope it will all make complete sense to me soon.

Roxy-3D, 2016-07-25:
OK...  That makes some sense.   I might have just gone through a similar thing because there are many dependencies in the different .cpp files because header files get included in different a different order depending on which ones get included first.

I wonder if we should consider having Marlin.h include all of the important .h files and always include Marlin.h first.  (It knows what to do because it always includes Configuration.h first.)   That might help to get things more deterministic in this situation.

thinkyhead, 2016-07-25:
> should consider having Marlin.h include all of the important .h files

That can be fine, but tends to mask errors where a header forgot to include its own dependencies. In general, each file should strive to include only the headers it needs, and `.h` files should be especially stingy about what they include. There are some good coding guidelines on the ordering of `#include` directives. (Basically, use them in an order that won't mask a forgotten `#include` elsewhere.)

Roxy-3D, 2016-07-25:
> (Basically, use them in an order that won't mask a forgotten #include elsewhere.)

Can you give me an example of how and where this can happen?   What you said doesn't make any sense to me.

Roxy-3D, 2016-07-25:
> each file should strive to include only the headers it needs, and .h files should be especially stingy about what they include.

And right now, the root cause of the problem I was fighting was the header files are including each other (and themselves) recursively.   That is why we have each header file do a first line of 
~~~~
#ifndef MARLIN_H
#define MARLIN_H
~~~~
That is done explicitly because none of the header files are following your 'best practices'.

AnHardt, 2016-07-25:
DirectIO could be an alternative if it would support AT90USB-processors an or Teensy - but i can't find that.
Removing `Sd2PinMap.h` should be __relative__ simple because the functions defined there are only used at a hand full places. 
All in all the definition of the SPI pins is a mess - if you try to follow the logic. Hardware-SPI, Software_SPI, used for display, SD-card, MAX-sensors, different SPI-ports wehter the SD-card is connected on board or at the display make it difficult to get an overview.

Only for the hardwar SPI this is obvious because the processors we use have only one hardware SPI-port. So finding out MISO, MOSI, SCLK is the easy task.

The variety of names for the chip select (Slave Select - SS) overwhelmed me.

Digging a bit deeper into software-SPI it becomes obvious we have at least two implementations. One in u8glib and one in SD2lib. 

The ideal faster than Arduino-IO library would be:
* fast
* supporting all processors/boards
* pin number could be a variable - not only a constant (even if slower than)
* supporting analogeWrite
but up to now i could not find that all together.

thinkyhead, 2016-07-25:
> Can you give me an example of how and where this can happen?

Let's say `myclass.h` forgot to include a dependency like `math.h` that the header needs for an inline function. Then in `myclass.cpp` (which also needs to use `math.h`) you have:
```cpp
#include "math.h"
#include "myclass.h"
```
This masks the fact that `myclass.h` forgot to include `math.h`. So the correct way to do include ordering is:
```cpp
#include "myclass.h"
#include "math.h"
```

thinkyhead, 2016-07-25:
> #ifndef MARLIN_H
#define MARLIN_H

These bracketing directives are good and normal practice in C and C++.

thinkyhead, 2016-07-25:
> consider making our header files not include any other header files

Header files should include the other headers that they need for the code and data types that they reference. They should not include header files that are only needed by their companion `.cpp` files.

Roxy-3D, 2016-07-25:
> This masks the fact that myclass.h forgot to include math.h. So the correct way to do include ordering is:

OK.  I think I ran into this issue.   I would finally get an 'out of scope' issue cleaned up, and magically a bunch of new errors happened.

> DirectIO could be an alternative if it would support AT90USB-processors an or Teensy 

The file here seems to be where the different processor types get defined: https://github.com/mmarchetti/DirectIO/blob/master/DirectIO/ports.h   Could we just generate the pin mappings for AT90USB and Teensy to resolve this issue?   

> Header files should include the other headers that they need for the code and data types that they reference. They should not include header files that are only needed by their companion .cpp files.

OK!  That helps a lot!   That makes good, solid sense.

AnHardt, 2016-07-25:
> Could we just generate the pin mappings for AT90USB and Teensy to resolve this issue?

Off cause we can. But can we test that in as reasonable time? We had to revisit every time critical section in Marlin and check if we are too slow - or too fast then. I don't even know if i know about all time critical sections.

thinkyhead, 2016-07-26:
@Roxy-3D I guess we can do this now and clean up the configs: https://github.com/thinkyhead/Marlin/blob/e7302bcfa2d45f8e7330f68731ad05007e2b4999/Marlin/MarlinConfig.h

Roxy-3D, 2016-07-26:
> @Roxy-3D I guess we can do this now and clean up the configs:

I presume we should do the same thing to devel-ubl also to keep them as closely synchronized as possible?   I just ran into some more header file problems today because of different options being turned off.

thinkyhead, 2016-07-26:
> do the same thing to devel-ubl

Well, `devel-ubl` is still behind `RCBugFix` and needs to be brought up-to-date with all the other changes first. Then this one would come along with those.

> header file problems

If you post the errors you're getting, I can make sure they are addressed in the current codebase also.

Roxy-3D, 2016-07-26:
OK...  I'll sit tight on making #include changes to devel-ubl.    The #include errors I dealt with today had to do with G26_Mesh_Validation_Tool.cpp    So, those issues are not a problem with the main code base.  
   
(but it was the nested, recursive includes that caused me to not see the problem until today when somebody tried to compile without LCD Panel support.)

thinkyhead, 2016-07-26:
Are the contents of all your `.h` files bracketed by `#ifndef`…`#endif`?

Roxy-3D, 2016-07-26:
Yes!    G29_Unified_Bed_Leveling.cpp:
~~~~
#ifndef G29_UNIFIED_BED_LEVELING_H
#define G29_UNIFIED_BED_LEVELING_H
#include "Marlin.h"
#if ENABLED(UNIFIED_BED_LEVELING_FEATURE)
~~~~
and the bed_leveling.h:
~~~~
#ifndef UNIFIED_BED_LEVELING_H
#define UNIFIED_BED_LEVELING_H
#include "Marlin.h"
#include "math.h"
#if ENABLED(UNIFIED_BED_LEVELING_FEATURE)
~~~~

thinkyhead, 2016-07-26:
So, what kind of error messages did you get?

Roxy-3D, 2016-07-26:
G26 was missing the definition for stepper motors, thermal stuff, and if I remember right the planner object.     I added explicit #includes:
~~~~
 #include "Marlin.h"
  #include "Configuration.h"
 +#include "planner.h"
 +#include "stepper.h"
 +#include "temperature.h"
  #include "G29_Unified_Bed_Leveling.h"
~~~~

thinkyhead, 2016-07-26:
If your `.cpp` file relies on something defined in "stepper.h" then it should have the include for it. Don't rely on "stepper.h" being included by something else you're including.

closed 2017-10-31 by thinkyhead