MarlinFirmware/Marlin #4413: Does not compile.

OrbitAVI, 2016-07-26:
OK please excuse my ignorance. I have been maintaining my own fork of Marlin Bug Fix for a while now, so I think I can navigate around it ok, but am no expert by any means and am pretty new to 3D printing as a whole. 

I cannot get this to compile with my printer params (basic stuff like PID settings, steps, etc..). I get an error. If I change nothing I also get a compile error. Printer is a i3.  Any help would be greatly appreciated and again sorry for being a noob at this.

   stepper.synchronize();
   ^
exit status 1
'Planner' does not name a type

OrbitAVI, 2016-07-26:
OK I'm dumb. With the example config (ROXY) I can compile, but with errors. I figured the base config.h file was OK to use.

Roxy-3D, 2016-07-26:
Can you post your Configuration.h file please?    I'll get that resolved ASAP.    But in the mean time, please use my Configuration.h file and cross your settings over to it.   It may be you have some #define's that conflict with each other.

OrbitAVI, 2016-07-26:
I was able to get it t compile using your example config and adding my settings however I did still get errors or warnings. I also got an error about only defining one type of probe. I only uncommented the fixed probe, but some of the servo options were conflicting, so I had to comment those out too. I'll post my config file later. Thanks for your help.

OrbitAVI, 2016-07-26:
I was able to compile now with only warnings. The thing that threw me off using your example config is that you moved the servo probe out of the Z probe section. I'm using a fixed mounted probe.

This config file is compiling for me with warnings (sketch\Marlin.h:51:37: warning: extra tokens at end of #ifdef directive [enabled by default]). 

 I'll load it up and try it out tonight. Can't wait!

`/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#ifndef CONFIGURATION_H
#define CONFIGURATION_H

/**
 *
 *  ***********************************
 *  **  ATTENTION TO ALL DEVELOPERS  **
 *  ***********************************
 *
 * You must increment this version number for every significant change such as,
 * but not limited to: ADD, DELETE RENAME OR REPURPOSE any directive/option.
 *
 * Note: Update also Version.h !
 */
#define CONFIGURATION_H_VERSION 010100

#include "boards.h"
#include "macros.h"

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
[... 1110 lines cut ...]
 * Note may require analog pins to be defined for different motherboards
 **********************************************************************/
// Uncomment below to enable
//#define FILAMENT_WIDTH_SENSOR

#define DEFAULT_NOMINAL_FILAMENT_DIA 3.00  //Enter the diameter (in mm) of the filament generally used (3.0 mm or 1.75 mm) - this is then used in the slicer software.  Used for sensor reading validation

#if ENABLED(FILAMENT_WIDTH_SENSOR)
  #define FILAMENT_SENSOR_EXTRUDER_NUM 0   //The number of the extruder that has the filament sensor (0,1,2)
  #define MEASUREMENT_DELAY_CM        14   //measurement delay in cm.  This is the distance from filament sensor to middle of barrel

  #define MEASURED_UPPER_LIMIT         3.30  //upper limit factor used for sensor reading validation in mm
  #define MEASURED_LOWER_LIMIT         1.90  //lower limit factor for sensor reading validation in mm
  #define MAX_MEASUREMENT_DELAY       20     //delay buffer size in bytes (1 byte = 1cm)- limits maximum measurement delay allowable (must be larger than MEASUREMENT_DELAY_CM  and lower number saves RAM)

  #define DEFAULT_MEASURED_FILAMENT_DIA  DEFAULT_NOMINAL_FILAMENT_DIA  //set measured to nominal initially

  //When using an LCD, uncomment the line below to display the Filament sensor data on the last line instead of status.  Status will appear for 5 sec.
  //#define FILAMENT_LCD_DISPLAY
#endif

#include "Configuration_adv.h"
#include "thermistortables.h"

#endif //CONFIGURATION_H`

Roxy-3D, 2016-07-26:
@OrbitAVI  Thank's for the response.    Currently you need to have both a LCD Panel and Z-Probe defined.   I've added the corresponding checks for that in the Sanity Checks file.   Long term, both of those constraints can go away.   But we have some re-organization of the code to do before we can relax those requirements.

OrbitAVI, 2016-07-26:
I have both defined in mine with a fixed probe and REPRAP_DISCOUNT_SMART_CONTROLLER or am I missing something? Again thank you very much.

Roxy-3D, 2016-07-26:
@OrbitAVI I cut and pasted your posted Configuration.h file into the build environment.  The compiler complained about some unrecognizable characters (picked up from the cut and paste) but everything compiled clean with the current source.    Can you try your settings against the current code?   I've added a few changes to the Sanity Checks and I've pulled in more #include files in a few of the .cpp files.   That might get you compiling now.

OrbitAVI, 2016-07-26:
I can compile with warnings using the latest. 

Warning:
`In file included from sketch\Bed_Leveling.cpp:23:0:

sketch\Marlin.h:51:37: warning: extra tokens at end of #ifdef directive [enabled by default]

 #ifdef UNIFIED_BED_LEVELING_FEATURE || ENABLED(M100_FREE_MEMORY_WATCHER)

                                     ^

In file included from sketch\G25.cpp:6:0:

sketch\Marlin.h:51:37: warning: extra tokens at end of #ifdef directive [enabled by default]

 #ifdef UNIFIED_BED_LEVELING_FEATURE || ENABLED(M100_FREE_MEMORY_WATCHER)

                                     ^

In file included from sketch\G26_Mesh_Validation_Tool.cpp:5:0:

sketch\Marlin.h:51:37: warning: extra tokens at end of #ifdef directive [enabled by default]

 #ifdef UNIFIED_BED_LEVELING_FEATURE || ENABLED(M100_FREE_MEMORY_WATCHER)

                                     ^

In file included from sketch\G29_Unified_Bed_Leveling.cpp:1:0:

sketch\Marlin.h:51:37: warning: extra tokens at end of #ifdef directive [enabled by default]

 #ifdef UNIFIED_BED_LEVELING_FEATURE || ENABLED(M100_FREE_MEMORY_WATCHER)

                                     ^

In file included from sketch\M100_Free_Mem_Chk.cpp:43:0:

sketch\Marlin.h:51:37: warning: extra tokens at end of #ifdef directive [enabled by default]

 #ifdef UNIFIED_BED_LEVELING_FEATURE || ENABLED(M100_FREE_MEMORY_WATCHER)

                                     ^

In file included from sketch\thermistortables.h:26:0,

                 from sketch\Configuration.h:1206,

                 from C:\Users\zz7njq\Downloads\Marlin-devel-ubl (1)\Marlin-devel-ubl\Marlin\Marlin.ino:37:

sketch\Marlin.h:51:37: warning: extra tokens at end of #ifdef directive [enabled by default]

 #ifdef UNIFIED_BED_LEVELING_FEATURE || ENABLED(M100_FREE_MEMORY_WATCHER)

                                     ^

In file included from sketch\MarlinSerial.cpp:31:0:

sketch\Marlin.h:51:37: warning: extra tokens at end of #ifdef directive [enabled by default]

 #ifdef UNIFIED_BED_LEVELING_FEATURE || ENABLED(M100_FREE_MEMORY_WATCHER)

                                     ^

In file included from sketch\Marlin_main.cpp:35:0:

sketch\Marlin.h:51:37: warning: extra tokens at end of #ifdef directive [enabled by default]

 #ifdef UNIFIED_BED_LEVELING_FEATURE || ENABLED(M100_FREE_MEMORY_WATCHER)

                                     ^
[... 207 lines cut ...]
                                     ^

In file included from sketch\ultralcd.h:26:0,

                 from sketch\ultralcd.cpp:23:

sketch\Marlin.h:51:37: warning: extra tokens at end of #ifdef directive [enabled by default]

 #ifdef UNIFIED_BED_LEVELING_FEATURE || ENABLED(M100_FREE_MEMORY_WATCHER)

                                     ^

In file included from sketch\vector_3.cpp:42:0:

sketch\Marlin.h:51:37: warning: extra tokens at end of #ifdef directive [enabled by default]

 #ifdef UNIFIED_BED_LEVELING_FEATURE || ENABLED(M100_FREE_MEMORY_WATCHER)

                                     ^

In file included from sketch\watchdog.cpp:23:0:

sketch\Marlin.h:51:37: warning: extra tokens at end of #ifdef directive [enabled by default]

 #ifdef UNIFIED_BED_LEVELING_FEATURE || ENABLED(M100_FREE_MEMORY_WATCHER)`

Roxy-3D, 2016-07-26:
@OrbitAVI Can you try the very latest?   I think it is currently a compiler dependency we are fighting.  I made a change that might fix it.

OrbitAVI, 2016-07-26:
Same error if SD support is enabled. Otherwise no issues. 
'
exit status 1
'cancel_heatup' was not declared in this scope
'

Roxy-3D, 2016-07-26:
@OrbitAVI 
OK...   Let me take a quick look at that.    Also, it might be helpful for you to be using Arduino v1.6.8 instead of v1.6.9     But I've got v1.6.9 loaded now so I can more easily duplicate your problems.   Let me see what I can find out about SD Memory Card support.

I'll report back soon.

OrbitAVI, 2016-07-26:
Same error in v1.6.8.

Thank you

Roxy-3D, 2016-07-26:
I think I might have it fixed in both v1.6.8 and v1.6.9    

I can compile clean with SD Support turned on now.

OrbitAVI, 2016-07-26:
Compiled with the latest + SD Support. The latest code gives the warnings again, but everything is all good. Running phase 1 now.

Roxy-3D, 2016-07-26:
:)    When you get a G26 to run successfully....   Please post a picture for people to see!    And if you have any questions, or things that don't quite make sense to you....  Please don't hesitate to just ask!   You have already struggled enough.  Let's make the rest of the process easy and fun for you!!!!

OrbitAVI, 2016-07-27:
Thanks for your help. I do have some questions. Should I close this and start a new thread?

Roxy-3D, 2016-07-27:
Either way...   But it might be best to start a new issue with a title that matches your question.   Hopefully, you can get a G26 to print a test pattern.   If so, you will be able to calibrate your Mesh very quickly now.

OrbitAVI, 2016-07-27:
Thanks again for the awesome support! Closing this issue as I am able to compile and flash.

closed 2016-07-27 by OrbitAVI